espressif/esp-idf #13887: ESP32S3 BLE Extended Advertising Does Not Work

letusgitcommit, 2024-05-31:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

# Setup
Windows 11
Powershell 7.4.2
esp-idfv5.2.1 (fresh install retrieved via git as detailed on the release page)
esp32s3 devkit c1
Python 3.11.3
Ran export.ps1 to populate appropriate environment

# Process
1. cd into the examples folder into the nimble bleprph folder.
2. Run idf.py set-target esp32s3
3. Run menuconfig and enable ble 5 feature extended advertising and go to example configuration and enable extended advertising.
4. idf.py build flash monitor.
5. Search for BLE devices.
6. Cannot find the device advertising.

# Log Output
```
Executing action: monitor
Serial port COM8
Connecting....
Detecting chip type... ESP32-S3
Running idf_monitor in directory ~\esp\esp-idf-v5.2.1\examples\bluetooth\nimble\bleprph
Executing "~\.espressif\python_env\idf5.2_py3.11_env\Scripts\python.exe ~\esp\esp-idf-v5.2.1\tools/idf_monitor.py -p COM8 -b 115200 --toolchain-prefix xtensa-esp32s3-elf- --target esp32s3 --revision 0 ~\esp\esp-idf-v5.2.1\examples\bluetooth\nimble\bleprph\build\bleprph.elf --force-color -m '~\.espressif\python_env\idf5.2_py3.11_env\Scripts\python.exe' '~\esp\esp-idf-v5.2.1\tools\idf.py'"...
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM8 instead...
--- esp-idf-monitor 1.4.0 on \\.\COM8 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clockESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x178c
load:0x403c9700,len:0x4
load:0x403c9704,len:0xcbc
load:0x403cc700,len:0x2da0
entry 0x403c9914
I (27) boot: ESP-IDF v5.2.1 2nd stage bootloader
I (27) boot: compile time May 31 2024 14:15:57
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.2
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : DIO
I (43) boot.esp32s3: SPI Flash Size : 2MB
I (48) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  2 factory          factory app      00 00 00010000 00100000
I (87) boot: End of partition table
I (91) esp_image: segment 0: paddr=00010020 vaddr=3c060020 size=1d0b8h (118968) map
I (121) esp_image: segment 1: paddr=0002d0e0 vaddr=3fc98600 size=02f38h ( 12088) load
I (123) esp_image: segment 2: paddr=00030020 vaddr=42000020 size=5bbe8h (375784) map
I (194) esp_image: segment 3: paddr=0008bc10 vaddr=3fc9b538 size=011d0h (  4560) load
I (195) esp_image: segment 4: paddr=0008cde8 vaddr=40374000 size=14540h ( 83264) load
I (226) boot: Loaded app from partition at offset 0x10000
[... 262 lines cut ...]
CONFIG_BT_CTRL_SCAN_DUPL_TYPE=0
CONFIG_BT_CTRL_SCAN_DUPL_CACHE_SIZE=100
CONFIG_BT_CTRL_DUPL_SCAN_CACHE_REFRESH_PERIOD=0
# CONFIG_BT_CTRL_BLE_MESH_SCAN_DUPL_EN is not set
# CONFIG_BT_CTRL_COEX_PHY_CODED_TX_RX_TLIM_EN is not set
CONFIG_BT_CTRL_COEX_PHY_CODED_TX_RX_TLIM_DIS=y
CONFIG_BT_CTRL_COEX_PHY_CODED_TX_RX_TLIM_EFF=0

#
# MODEM SLEEP Options
#
# CONFIG_BT_CTRL_MODEM_SLEEP is not set
# end of MODEM SLEEP Options

CONFIG_BT_CTRL_SLEEP_MODE_EFF=0
CONFIG_BT_CTRL_SLEEP_CLOCK_EFF=0
CONFIG_BT_CTRL_HCI_TL_EFF=1
# CONFIG_BT_CTRL_AGC_RECORRECT_EN is not set
# CONFIG_BT_CTRL_SCAN_BACKOFF_UPPERLIMITMAX is not set
# CONFIG_BT_BLE_ADV_DATA_LENGTH_ZERO_AUX is not set
# end of Controller Options
# end of Bluetooth
```

What am I doing wrong here?

rahult-github, 2024-06-01:
Hi @letusgitcommit ,

> I (448) NimBLE: Device Address:
I (448) NimBLE: 48:27:e2:f0:22:ba
I (448) NimBLE:
I (448) NimBLE: GAP procedure initiated: extended advertise; instance=0
I (448) uart: queue free spaces: 8
I (458) main_task: Returned from app_main()

Logs indicate that extended advertising has been started. 

1. What is the remote device (then central device) being used ? 
2. Does the remote device find other devices which are doing extended advertising ? 
3. Can you try changing remote to say blecent example of s3 and check if it works or not.

letusgitcommit, 2024-06-03:
Turns out this was a me issue. Thanks for the response @rahult-github!

I was using the bleak client Python package for Windows 11, but I must be using it wrong (or it doesn't support extended advertising).

Steps to fix:
1. Build, flash, and monitor the `bleprph` example with appropriate menuconfig settings (extended advertising, etc.).

closed 2024-06-03 by letusgitcommit

rahult-github, 2024-06-04:
Hi @letusgitcommit ,

Ok . Thanks for confirming.

> (even if you don't happen to see the exact info). 

Default log level is at INFO, which prints the operation information only. If more detailed information is needed ( like what was the exact data content that was trasmitted / received ), you can enable the Debug logging mode and check .