espressif/esp-idf #13888: Example code 5.2.1 idf - heap_task_tracking example: main task name gets corrupted and longer as time goes on.

SCgenie, 2024-05-31:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

 "main" task name gets corrupted and longer as time goes on. I see the above issue when I run the heap_task_tracking example project. Is this a risk ? CAP_32BIT returns 0 always, Any suggestions? Thanks all

Task: Pre-Scheduler allocs -> CAP_8BIT: 3068 CAP_32BIT: 20
Task: ipc0 -> CAP_8BIT: 9584 CAP_32BIT: 0
Task: main -> CAP_8BIT: 3648 CAP_32BIT: 0
Task: ipc1 -> CAP_8BIT: 16 CAP_32BIT: 0
Task: example_task -> CAP_8BIT: 220 CAP_32BIT: 0

Task: Pre-Scheduler allocs -> CAP_8BIT: 3068 CAP_32BIT: 20
Task: ipc0 -> CAP_8BIT: 9584 CAP_32BIT: 0
Task: ���? -> CAP_8BIT: 3648 CAP_32BIT: 0
Task: ipc1 -> CAP_8BIT: 16 CAP_32BIT: 0
Task: example_task -> CAP_8BIT: 40 CAP_32BIT: 0

filzek, 2024-06-04:
> Hi! On the master it works fine. There was an issue #13467 that was fixed recently. This fix is already in v5.2.2.
> 
> > CAP_32BIT returns 0 always
> 
> Seems it is correct behavior. @SoucheSouche Could you check this? Thanks.

what is the git difference to the Release/v5.2 to the v5.2.2 it was not supose to have all commits to the same?

KonstantinKondrashov, 2024-06-04:
> what is the git difference to the Release/v5.2 to the v5.2.2 it was not supose to have all commits to the same?

There is a doc that describes IDF release branches - https://docs.espressif.com/projects/esp-idf/en/stable/esp32/versions.html#checking-the-current-version.

Yes, for sure, you can use the [release/v5.2](https://github.com/espressif/esp-idf/tree/release/v5.2) branch, it includes changes v5.2.2.
(release/v5.2 branch includes more commits than v5.2.2, the new commits will go to the next release v5.2.3 in the feature).

KonstantinKondrashov, 2024-06-04:
@SCgenie 

>  CAP_32BIT returns 0 always

Yes, it is the correct output. The freertos allocates the stack for tasks with the following caps:
`#define FREERTOS_SMP_MALLOC_CAPS    (MALLOC_CAP_INTERNAL|MALLOC_CAP_8BIT)`.

See the difference CAP_8BIT and CAP_32BIT -> https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/mem_alloc.html?highlight=cap_32bit#bit-accessible-memory

SoucheSouche, 2024-06-12:
Hi @SCgenie,
I will close the ticket since it seems that your concern was answered. Fill free to reopen if you have any other questions related to your problem.

closed 2024-06-12 by SoucheSouche