espressif/esp-idf #13891: IDF 5.1.4: Rare LoadProhibited panic when connecting to a WiFi

kriegste, 2024-06-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I recently experienced a LoadProhibited panic during WiFi connect (to an access point using WPA2, not WPA3). I tried to reproduce the error, but did not succeed... I am mentioning this here nonetheless. Time will tell if it is useful.

ESP-IDF 5.1.4

> Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
> 
> Core  0 register dump:
> PC      : 0x40114135  PS      : 0x00060f30  A0      : 0x00000000  A1      : 0x3ffddce0
> A2      : 0x3ffb7804  A3      : 0x00000001  A4      : 0x00000000  A5      : 0x3ffd3184
> A6      : 0x00000002  A7      : 0x00000000  A8      : 0x80114135  A9      : 0x3ffddcc0
> A10     : 0x3ffd3184  A11     : 0x3ffddd0c  A12     : 0xffffffff  A13     : 0x3ffddda8
> A14     : 0x80000001  A15     : 0xb33fffff  SAR     : 0x0000001b  EXCCAUSE: 0x0000001c
> EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff
> 
> 
> Backtrace: 0x40114132:0x3ffddce0

The backtrace points to this line:
https://github.com/espressif/esp-idf/blob/release/v5.1/components/wpa_supplicant/esp_supplicant/src/esp_wpa3.c#L505

EXCVADDR is 0, so evt was a NULL pointer, I guess.

Shreyas0-7, 2024-06-04:
Hey @kriegste is possible for you to provide console logs for the crash?
Also can you please mention the steps which caused the crash

kriegste, 2024-06-05:
I'm afraid this is all the info from the log I have.

The crash happened right after a restart when the device (ESP32) tried to connect (for the first time after the restart). It is in a location with a very weak signal (RSSI -83 dBm or worse). So it loses connection regularly and reconnects, but never crashes. There are no other crashes either. The access point uses WPA2, not WPA3. Maybe a data packet was corrupted and got misinterpreted WPA3 is enabled in sdkconfig, though.

AxelLin, 2024-06-08:
> I'm afraid this is all the info from the log I have.
> 
> The access point uses WPA2, not WPA3. Maybe a data packet was corrupted and got misinterpreted? WPA3 is enabled in sdkconfig, though.

Seems the authmode is WIFI_AUTH_WPA2_WPA3_PSK, so esp_wpa3_hostap_task() is executed.

kriegste, 2024-06-08:
ESP32 is in STA mode. The access point is a router (set to WPA2 PSK CCMP).

Shreyas0-7, 2024-06-10:
Hey @kriegste there might be some confusion with the configuration because the backtrace indicates there is some problem while deinitialising softAP  which will never take place if esp is in STA mode

kriegste, 2024-06-10:
That could be a hot lead! My app waits a few seconds and opens a softAP when the initial connect to the router fails. I closes it in case the connection to the router succeeds. That could have happened, but I did not keep the rest of the log...

Shreyas0-7, 2024-06-10:
Hey @kriegste can you please check your app, it shouldn't start softAP.

kriegste, 2024-06-10:
The app starts the softAP only as a fail-safe, if the STA connection fails.

Now I modified it to start/stop the softAP twice every minute and see what happens.

kapilkedawat, 2024-10-14:
@kriegste do you still face this issue?

kriegste, 2024-10-14:
No, I have never seen this problem again. Also, I am using IDF 5.3 now.