espressif/esp-idf #13892: gptimer has peculiar behavior/interaction with gpio usage

mbratch, 2024-06-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

My platform is an ESP32 Pico Kit v1 which has an ESP32 Pico D4 micro.
I am using ESP-IDF v5.2.1.

I have a fairly simple program that uses two gptimers to generate a Wiegand data stream given some data and number of bits to be transmitted. This uses two synchronized output GPIO pins for data0 and data1.  The program uses a recurring interval timer to determine when to generate a pulse, and a one-shot pulse timer which turns the pulse off at the correct time. The timers are initialized, configured and enabled up front and as simply started or stopped as needed to generate a data stream.

This program works and appears to provide very good timing. Key timing parameters for the data are the pulse width and the interval width (the distance between the start of pulses). 

Typically, I'm only seeing an an accuracy in the pulse width within a couple of microseconds which is pretty good. The odd thing I'm seeing with this implementation though is that all the pulses and intervals look great except the very first one. Typically, the first pulse is delayed (relative to the next) so the interval is shortened a little. Not by a huge amount but sometimes about 10us. Secondly, the pulse is compressed. So if I set the desired pulse width to 20us (minimum permitted)., all the pulses except the first are around 20-22us. But the first is often 5-10us. Using larger pulse widths, the first pulse is till "distorted" (compressed by 5-15us and delayed) relative to the remaining pulses

What I'm scratching my head over is why this problem is particular to the first pulse. There's really nothing special about it as far as the code is concerned, other than it being the pulse generated on the first alarm call-back of the interval timer after it is started.

I added an example of my program. My sdkconfig for GPTIMERS is as follows
```
CONFIG_GPTIMER_ISR_HANDLER_IN_IRAM=y
CONFIG_GPTIMER_CTRL_FUNC_IN_IRAM=y
CONFIG_GPTIMER_ISR_IRAM_SAFE=y
# CONFIG_GPTIMER_SUPPRESS_DEPRECATE_WARN is not set
# CONFIG_GPTIMER_ENABLE_DEBUG_LOG is not set
# end of GPTimer Configuration
```
I've tried a few different settings for the config, but it didn't make much difference.

Would it help if I used the timers in a separate CPU core, or would that matter? Looking for some explanation and some suggestions. Thanks!

suda-morris, 2024-06-03:
Hi @mbratch Thanks for this interesting example!

I think this issue was caused by the traditional "cache miss", because by default, the "gpio_set_level" function is placed in the flash text section. If you enable the `CONFIG_GPIO_CTRL_FUNC_IN_IRAM`, the weird behavior will disappear.

mbratch, 2024-06-03:
Hi @suda-morris thank you very much! I was not aware of that setting.
It definitely improved a lot with the setting.
In the extreme cases, checking the shortest allowed pulse of 20us and interval of 200us, the first pulse is delayed by about 12us, and is slightly shortened by an amount not worth concern. So the distance from beginning of first pulse to second is 188us, and then the pulses are at the programmed 200us interval.

filzek, 2024-06-04:
I had similar problem with the gptimer as well, but we know that the problem are related to the gptimer initate state, you need to iniate the timer and let it lend about 200us at least before any action on the gptimer, if you do the timer will simple move the sketch of the sides, sonner or later, so, it you stop and enable the timer the problem will not happen, the deal is with
ESP_ERROR_CHECK(gptimer_new_timer(&timerConfig, timerHandle));

somehow after this, you need 200us timer to run perfect, what you should do is move the creation to another side, and keep it, so, the timer is always created and doesnt need to be reacreated.

We had a lot of trouble to fix a Zero Crossing based in GPTimer and still some problems yet to solve. The HW timer in the SDK 4.4 was the best in perfection, the GPTimer has serious issues for perfect window hit yet.

mbratch, 2024-06-04:
Hello @filzek . Thank you for the ideas.

> you need 200us timer to run perfect, what you should do is move the creation to another side, and keep it, so, the timer is always created and doesnt need to be reacreated.

My code currently only creates and enables the timers one time at boot. So when I want to use them, I am not recreating them or even re-enabling them each time. When I run a sequence of pulses, I only start or stop the timers.

filzek, 2024-06-05:
Use direct GPIO manipulation on call like
disable gpio
GPIO.out_w1tc = (1 << dataPin1);

enable gpio
GPIO.out_w1ts = (1 << dataPin1);

this shall save some Us from code running.

maybe if you call  gptimer_stop(intervalTimerH); prior to gptimer_set_raw_count(intervalTimerH, 0); could be better run in the GPTIMER routime.

closed 2024-07-17 by espressif-bot