espressif/esp-idf #13896: Advertising Specific Services and Characteristics in BLE using ESP-IDF for ESP32

AhmedSalem636, 2024-06-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,

In my project. I am using ESP32 to communicate with a mobile-App over BLE.
we are using two Services. Each has its own characteristics. So, we were thinking if it is possible to advertise only one of them without the other while still being able to interact with the hidden one knowing its Service and characteristics UUIDs.

here are the code snappets that are related to the services creation and advertising.

static const struct ble_gatt_svc_def Mygatt_svr_svcs[] = {
{
    /*** Service: Security test. */
		.type= BLE_GATT_SVC_TYPE_PRIMARY,
		.uuid = BLE_UUID128_DECLARE(Service_UUID),
    .characteristics = (struct ble_gatt_chr_def[]) {  {
        /*** Characteristic: Static value. */
        .uuid = BLE_UUID128_DECLARE(WRITE_UUID),
        .access_cb = MyHandler,
        .flags = BLE_GATT_CHR_F_READ|
                 BLE_GATT_CHR_F_WRITE,
    }, {
        0, /* No more characteristics in this service. */
    } },
},

{
    // service: OTA Service
    .type = BLE_GATT_SVC_TYPE_PRIMARY,
    .uuid = &gatt_svr_svc_ota_uuid.u,
    .characteristics =
        (struct ble_gatt_chr_def[]){
            {
                // characteristic: OTA control
                .uuid = &gatt_svr_chr_ota_control_uuid.u,
                .access_cb = gatt_svr_chr_ota_control_cb,
                .flags = BLE_GATT_CHR_F_READ | BLE_GATT_CHR_F_WRITE |
                         BLE_GATT_CHR_F_NOTIFY,
                .val_handle = &ota_control_val_handle,
            },
            {
                // characteristic: OTA data
                .uuid = &gatt_svr_chr_ota_data_uuid.u,
                .access_cb = gatt_svr_chr_ota_data_cb,
                .flags = BLE_GATT_CHR_F_WRITE,
                .val_handle = &ota_data_val_handle,
            },
            {
                0,
            }},
},

{
    0, /* No more services. */
},
};

#if CONFIG_EXAMPLE_EXTENDED_ADV
/**
 * Enables advertising with the following parameters:
 *     o General discoverable mode.
 *     o Undirected connectable mode.
 */
static void
ext_bleprph_advertise(void)
{
    struct ble_gap_ext_adv_params params;
    struct os_mbuf *data;
    uint8_t instance = 0;
    int rc;

    /* First check if any instance is already active */
    if(ble_gap_ext_adv_active(instance)) {
        return;
    }

    /* use defaults for non-set params */
    memset (&params, 0, sizeof(params));

    /* enable connectable advertising */
    params.connectable = 1;
    params.itvl_min=3500;
    params.itvl_max=3500;
	params.latency = 0x10; //number of skippable connection events
	params.timeout = 0x64; // x 6.25ms, time before peripherial will assume connection is dropped.
    /* advertise using random addr */
    params.own_addr_type = BLE_OWN_ADDR_PUBLIC;

    params.primary_phy = BLE_HCI_LE_PHY_1M;
    params.secondary_phy = BLE_HCI_LE_PHY_2M;
    //params.tx_power = 127;
    params.sid = 1;

    params.itvl_min = BLE_GAP_ADV_FAST_INTERVAL1_MIN;
    params.itvl_max = BLE_GAP_ADV_FAST_INTERVAL1_MIN;

    /* configure instance 0 */
    rc = ble_gap_ext_adv_configure(instance, &params, NULL,
                                   bleprph_gap_event, NULL);
    assert (rc == 0);

    /* in this case only scan response is allowed */

    /* get mbuf for scan rsp data */
    data = os_msys_get_pkthdr(sizeof(ext_adv_pattern_1), 0);
    assert(data);

    /* fill mbuf with scan rsp data */
    rc = os_mbuf_append(data, ext_adv_pattern_1, sizeof(ext_adv_pattern_1));
    assert(rc == 0);

    rc = ble_gap_ext_adv_set_data(instance, data);
    assert (rc == 0);

    /* start advertising */
    rc = ble_gap_ext_adv_start(instance, 0, 0);
    assert (rc == 0);
}
#else
/**
 * Enables advertising with the following parameters:
 *     o General discoverable mode.
 *     o Undirected connectable mode.
 */
static void
bleprph_advertise(void)
{
    struct ble_gap_adv_params adv_params;
    struct ble_hs_adv_fields fields;
    const char *name;
    int rc;

    /**
     *  Set the advertisement data included in our advertisements:
     *     o Flags (indicates advertisement type and other general info).
     *     o Advertising tx power.
     *     o Device name.
     *     o 16-bit service UUIDs (alert notifications).
     */

    memset(&fields, 0, sizeof fields);

    /* Advertise two flags:
     *     o Discoverability in forthcoming advertisement (general)
     *     o BLE-only (BR/EDR unsupported).
     */
    fields.flags = BLE_HS_ADV_F_DISC_GEN |
                   BLE_HS_ADV_F_BREDR_UNSUP;

    /* Indicate that the TX power level field should be included; have the
     * stack fill this value automatically.  This is done by assigning the
     * special value BLE_HS_ADV_TX_PWR_LVL_AUTO.
     */
    fields.tx_pwr_lvl_is_present = 1;
    fields.tx_pwr_lvl = ESP_PWR_LVL_P9;

    name = ble_svc_gap_device_name();
    fields.name = (uint8_t *)name;
    fields.name_len = strlen(name);
    fields.name_is_complete = 1;

    fields.uuids16 = (ble_uuid16_t[]) {
        BLE_UUID16_INIT(Service_UUID)
    };
    fields.num_uuids16 = 1;
    fields.uuids16_is_complete = 1;

    rc = ble_gap_adv_set_fields(&fields);
    if (rc != 0) {
        MODLOG_DFLT(ERROR, "error setting advertisement data; rc=%d\n", rc);
        return;
    }

    /* Begin advertising. */
    memset(&adv_params, 0, sizeof adv_params);
    adv_params.conn_mode = BLE_GAP_CONN_MODE_UND;
    adv_params.disc_mode = BLE_GAP_DISC_MODE_GEN;
    adv_params.itvl_min=3500;
    adv_params.itvl_max=3500;
//    adv_params.conn_latency = 0x00; //number of skippable connection events
//    adv_params.supervision_timeout = 0xA0; // x 6.25ms, time before peripherial will assume connection is dropped.
    rc = ble_gap_adv_start(own_addr_type, NULL, BLE_HS_FOREVER,
                           &adv_params, bleprph_gap_event, NULL);
    if (rc != 0) {
        MODLOG_DFLT(ERROR, "error enabling advertisement; rc=%d\n", rc);
        return;
    }
}
#endif

AhmedSalem636, 2024-07-04:
anyone here can help?
Thanks

rahult-github, 2024-07-05:
Hi @AhmedSalem636 , 

> So, we were thinking if it is possible to advertise only one of them without the other while still being able to interact with the hidden one knowing its Service and characteristics UUIDs.

Do you imply that only one service would be registered in the gatt server database ?  i.e., when some central device attempts to perform gatt discovery, at that time only one service would show up ? 

If yes, then there is no other way the central would be aware of any other service . gatt service discovery is the only way a device gets information about services present on remote side.

rahult-github, 2024-07-24:
Closing issue due to no updates. Please feel free to reopen in case of further updates.

closed 2024-07-24 by rahult-github