espressif/esp-idf #13899: ESP IDF, Arduino as a component, doesn't compile

meakashrao, 2024-06-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.1.2-666-ga5e3e0b90e

### Espressif SoC revision.

ESP32-WROOM-32E-N8

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP-WROOM-32

### Power Supply used.

USB

### What is the expected behavior?

I expected that when using Arduino as a component, the Arduino core to compile, I had use one similar some 3 moths ago and it did compile. 

I had to make some changes, 

1) The Arduino Core I downloaded on 6th Feb 2024 did not have this line so I erased it after it threw the error. 
[image]

2) Same goes with this, earlier version does not have this and I had to comment this out. 
[image]

I also tried IDF Version 5.2.1, but it has some other error while downloading mdns as an component, so came back to idf version 5.1.2

### What is the actual behavior?

Even after making those above changes, there are some errors in .h files of the Arduino cores.

[IDF Bug.zip](https://github.com/user-attachments/files/15533415/IDF.Bug.zip)

### Steps to reproduce.

1. IDF Version 5.1.2, download Arduino as a Component
2. Create a component called as OTA
3. Use the below Files 

Its a .c file, for uploading here made .txt file
[OTA.txt](https://github.com/user-attachments/files/15533468/OTA.txt)
[CMakeLists.txt](https://github.com/user-attachments/files/15533426/CMakeLists.txt)

### Debug Logs.

```plain
-- Found Git: C:/Users/Akash/esp/Tools_Directory/tools/idf-git/2.39.2/cmd/git.exe (found version "2.39.2.windows.1") 
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
[... 273 lines cut ...]
D:/UP/GitHubVSArea/ESP_IDF/ESP32FW-IDF-RTOS/DelishUP-RTOS/components/arduino/cores/esp32/IPAddress.h:38:1: error: 
unknown type name 'class'
   38 | class IPAddress : public Printable {
      | ^~~~~
D:/UP/GitHubVSArea/ESP_IDF/ESP32FW-IDF-RTOS/DelishUP-RTOS/components/arduino/cores/esp32/IPAddress.h:38:17: error:
 expected '=', ',', ';', 'asm' or '__attribute__' before ':' token
   38 | class IPAddress : public Printable {
      |                 ^
D:/UP/GitHubVSArea/ESP_IDF/ESP32FW-IDF-RTOS/DelishUP-RTOS/components/arduino/cores/esp32/IPAddress.h:136:14: error: unknown type name 'IPAddress'
  136 | extern const IPAddress IN6ADDR_ANY;
      |              ^~~~~~~~~
D:/UP/GitHubVSArea/ESP_IDF/ESP32FW-IDF-RTOS/DelishUP-RTOS/components/arduino/cores/esp32/IPAddress.h:137:14: error: unknown type name 'IPAddress'
  137 | extern const IPAddress INADDR_NONE;
      |              ^~~~~~~~~
In file included from D:/UP/GitHubVSArea/ESP_IDF/ESP32FW-IDF-RTOS/DelishUP-RTOS/components/arduino/libraries/WiFi/src/WiFiSTA.h:29,
                 from D:/UP/GitHubVSArea/ESP_IDF/ESP32FW-IDF-RTOS/DelishUP-RTOS/components/arduino/libraries/WiFi/src/WiFi.h:33:
D:/UP/GitHubVSArea/ESP_IDF/ESP32FW-IDF-RTOS/DelishUP-RTOS/components/arduino/libraries/WiFi/src/WiFiGeneric.h:30:10: fatal error: functional: No such file or directory
   30 | #include <functional>
      |          ^~~~~~~~~~~~
compilation terminated.
[1635/1643] Building CXX object esp-idf/arduino/CMakeFiles/__idf_arduino.dir/libraries/WiFi/src/STA.cpp.obj       
ninja: build stopped: subcommand failed.

 *  The terminal process "C:\WINDOWS\System32\cmd.exe /d /c ninja " terminated with exit code: 1.
```

igrr, 2024-06-03:
```
In file included from D:/UP/GitHubVSArea/ESP_IDF/ESP32FW-IDF-RTOS/DelishUP-RTOS/components/arduino/cores/esp32/Print.h:28,
                 from D:/UP/GitHubVSArea/ESP_IDF/ESP32FW-IDF-RTOS/DelishUP-RTOS/components/arduino/libraries/WiFi/src/WiFi.h:29,
                 from D:/UP/GitHubVSArea/ESP_IDF/ESP32FW-IDF-RTOS/DelishUP-RTOS/components/OTA/OTA.c:24:
D:/UP/GitHubVSArea/ESP_IDF/ESP32FW-IDF-RTOS/DelishUP-RTOS/components/arduino/cores/esp32/Printable.h:25:1: error: 
unknown type name 'class'
```

From the log, the error occurs because you are including a C++ header file (WiFi.h) from a C source file (DelishUP-RTOS/components/OTA/OTA.c).

igrr, 2024-06-03:
Seems like `mdns` dependency in arduino-esp32 has to be declared as `public` because `mdns.h` is included from a public header file.

https://github.com/espressif/arduino-esp32/blob/8b4c1303f41f9a2b03ac73b5c37b3bd02b62c5dd/idf_component.yml#L47

https://github.com/espressif/arduino-esp32/blob/8b4c1303f41f9a2b03ac73b5c37b3bd02b62c5dd/libraries/ESPmDNS/src/ESPmDNS.h#L45

@meakashrao You can work around the issue by adding `espressif__mdns` to the list of requirements of your `OTA` component (in DelishUP-RTOS/components/OTA/CMakeLists.txt).

meakashrao, 2024-06-03:
> @meakashrao You can work around the issue by adding `espressif__mdns` to the list of requirements of your `OTA` component (in DelishUP-RTOS/components/OTA/CMakeLists.txt).

Thank you, this helps. 

What about these two issues
[image]
[image]

Deleting and commenting, helped me solve them, is this the right thing to do, doesn't look like it asking for clarity