espressif/esp-idf #13903: building an exmaple without the entire checkout, posible?

ulao, 2024-06-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

not sure

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

I was hoping to just ask a general question, not trying to log a bug or anything. Hope its ok to ask here. 

I use an esp32 on a wroom32 board that runs a blink LED test just fine. What I want to do is make a USB host and wanted to start with the USB Host Library Example example. 

I added the main code and the class_driver.c file but I have no usb/usb_host.h files is it not possible to built without the entire checkout  ?

 

### What is the actual behavior?

none

### Steps to reproduce.

none

### Build or installation Logs.

```plain
none
```

### More Information.

none

igrr, 2024-06-04:
Hi @ulao, you do need to get the entire IDF repository to build the example. Please follow the getting started guide for instructions.

To reduce the amount of data downloaded, if you're interested in that, you can do a shallow Git clone using a `--depth` argument.

ulao, 2024-06-08:
Hey iggr, thx for this, I'm trying to make an example run in platform.io. I was able to find and put all the includes in the right place to make it compile but now I'm hit with the linker part.

I'm not sure how that works in platform.io guessing I need to find what one calls "libs" and put them in the lib folder but do you happen to know what the linker is looking for and where the files live in this repo?

example:
Linking .pio\build\upesy_wroom\firmware.elf
C:/Users/Administrator/.platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: .pio\build\upesy_wroom\src\usb\host\usb_host_lib\main\class_driver.c.o:(.literal.action_open_dev+0x1c): undefined reference to `usb_host_device_open'

tore-espressif, 2024-06-12:
Hello @ulao , you said that you work with ESP32. ESP32 does not have the USB-OTG peripheral, so this won't work. The `usb` component is excluded from build process, so the linker cannot find it.

You need ESP32-S2 or ESP32-S3 to run USB examples

ulao, 2024-06-12:
eh I since learned that. I ordered a -s3 and will give it another go once I get it. thx

tore-espressif, 2024-06-12:
Great, I will close this issue then.

closed 2024-06-12 by tore-espressif