espressif/esp-idf #13904: esp32-s3 how to transmit large data with  triggers on both edges

WeiTianlia, 2024-06-04:
I want to make a dot matrix screen using AG5958 as the selector and mbi5268 as the data device. I want to use SPI to simulate sending data, but I found that SPI does not have the ability to support "double edge triggered" in the application. However, mbi5268 only supports this triggering mode. I don't know if anyone has a way to use SPI to send matching data to mbi5268,ths
[image]

wanckl, 2024-06-06:
@WeiTianlia 
Hi, FMY, GPSPI2 on esp32s3 is actually support DDR mode in hardware,    
But haven't supported in IDF yet     

maybe they can support later as an device flag is enough~   
for now you can have a try to config registers as bellow after `spi_bus_add_device`   
```c
GPSPI2.misc.clk_data_dtr_en = 1;
GPSPI2.misc.data_dtr_en = 1;
```

then you'll find your transactions become DDR mode!!

WeiTianlia, 2024-06-26:
> @WeiTianlia Hi, FMY, GPSPI2 on esp32s3 is actually support DDR mode in hardware, But haven't supported in IDF yet
> 
> maybe they can support later as an device flag is enough~ for now you can have a try to config registers as bellow after `spi_bus_add_device`
> 
> ```c
> #include "soc/spi_struct.h"
> ...
> GPSPI2.misc.clk_data_dtr_en = 1;
> GPSPI2.misc.data_dtr_en = 1;
> ```
> 
> then you'll find your transactions become DDR mode!!

that is right,had Verified
data[]={0b10101010,0b10101010,0b10101010,0b10101010},actal spi,4 cycle
[image]

closed 2024-06-26 by WeiTianlia