espressif/esp-idf #13906: Assertion of corrupted heap when SPI RAM is enabled for LWIP/WLAN on ESP32S3

eriksl, 2024-06-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.1

### Espressif SoC revision.

ESP32-S3, v0.1

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Not relevant here

### Power Supply used.

USB

### What is the expected behavior?

I expected to free up some internal RAM. It appears WIFI and LWIP are very memory demanding.

### What is the actual behavior?

System aborts, heap corruption, backtrace points to somewhere in LWIP, in the TCP timer handler. But the real origin is probably somewhere else.

### Steps to reproduce.

1. Enable use PSRAM for LWIP/WLAN in the menuconfig
2. Build
3. Flash
4. Wait a few seconds

AxelLin, 2024-06-04:
@eriksl
Could you try if you still hit this issue with v5.2.2 release?
I have used the same setting on esp32s3, but I didn't hit this issue.
If you still have this issue, you probably need to provide some simplified code to reproduce it.
Also please share the sdkconfig and the log when you hit the issue.

eriksl, 2024-06-04:
Apparently there is no v5.2.2 (yet):

```
gaia erik:~/.espressif/esp-idf $ git remote -v
origin	https://github.com/espressif/esp-idf.git (fetch)
origin	https://github.com/espressif/esp-idf.git (push)
gaia erik:~/.espressif/esp-idf $ git fetch -a 
gaia erik:~/.espressif/esp-idf $ git tag | fgrep 5.2.2
gaia erik:~/.espressif/esp-idf $ git tag | fgrep 5.2.1
v5.2.1
```

eriksl, 2024-06-04:
This is the option, BTW:

> Name: SPIRAM_TRY_ALLOCATE_WIFI_LWIP
> Prompt: Try to allocate memories of WiFi and LWIP in SPIRAM firstly. If failed, allocate internal memory
> Type: bool
> Value: n
> 
> Help:
> 
>   Try to allocate memories of WiFi and LWIP in SPIRAM firstly. If failed, try to allocate internal
>   memory then.
> 
> Direct dependencies (=y):
>      SPIRAM_USE_CAPS_ALLOC(=y) || SPIRAM_USE_MALLOC(=n)  (=y)
>   && SPIRAM(=y)
>   && !APP_BUILD_TYPE_PURE_RAM_APP(=n)  (=y)
> 
> Default:
>   - n

eriksl, 2024-06-04:
And this is the abort message. I do have maximum heap checking enabled, may that's the difference.

```
CORRUPT HEAP: Bad tail at 0x3c10e5bc. Expected 0xbaad5678 got 0x3c10e5dc

assert failed: multi_heap_free multi_heap_poisoning.c:279 (head != NULL)

Backtrace: 0x403799ef:0x3fcbe8d0 0x40385e2d:0x3fcbe900 0x4038ee96:0x3fcbe930 0x4038d873:0x3fcbea50 0x4037a372:0x3fcbea80 0x4038ef2d:0x3fcbeab0 0x42045e3f:0x3fcbeae0 0x42045e8a:0x3fcbeb10 0x42045f7a:0x3fcbeb40 0x4204c2dd:0x3fcbeb70 0x42045894:0x3fcbeba0
0x403799ef: panic_abort at /home/erik/.espressif/esp-idf/components/esp_system/panic.c:472
0x40385e2d: esp_system_abort at /home/erik/.espressif/esp-idf/components/esp_system/port/esp_system_chip.c:93
0x4038ee96: __assert_func at /home/erik/.espressif/esp-idf/components/newlib/assert.c:81
0x4038d873: multi_heap_free at /home/erik/.espressif/esp-idf/components/heap/multi_heap_poisoning.c:279 (discriminator 1)
0x4037a372: heap_caps_free at /home/erik/.espressif/esp-idf/components/heap/heap_caps.c:393
0x4038ef2d: free at /home/erik/.espressif/esp-idf/components/newlib/heap.c:39
0x42045e3f: mem_free at /home/erik/.espressif/esp-idf/components/lwip/lwip/src/core/mem.c:236
0x42045e8a: do_memp_free_pool at /home/erik/.espressif/esp-idf/components/lwip/lwip/src/core/memp.c:383
0x42045f7a: memp_free at /home/erik/.espressif/esp-idf/components/lwip/lwip/src/core/memp.c:440
0x4204c2dd: sys_check_timeouts at /home/erik/.espressif/esp-idf/components/lwip/lwip/src/core/timeouts.c:401
0x42045894: tcpip_timeouts_mbox_fetch at /home/erik/.espressif/esp-idf/components/lwip/lwip/src/api/tcpip.c:98
 (inlined by) tcpip_thread at /home/erik/.espressif/esp-idf/components/lwip/lwip/src/api/tcpip.c:142
```

eriksl, 2024-06-04:
Upgrading to v5.2.2 does seem to help (although I cannot find anything about this in the Changelog). Fingers crossed...

eriksl, 2024-06-05:
When it all settles a bit, I will push the project to github and notify. At the moment it's still all a bit messy (but not related to this issue).

eriksl, 2024-08-01:
Haven't seen the issue lately.

closed 2024-08-01 by eriksl