espressif/esp-idf #13909: Error writing HMAC NVS encryption key in efuse

alerighi, 2024-06-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.1

### Espressif SoC revision.

esp32c6

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

custom board

### Power Supply used.

USB

### What is the expected behavior?

When enabling NVS encryption by using HMAC scheme, with the following code, the write to efuse fails, if also secure boot is enabled on the board. 

```c
int partition_init(const char *name)
{
    nvs_sec_cfg_t securityCfg = {0};
    esp_err_t error = nvs_flash_read_security_cfg_v2(nvs_flash_get_default_security_scheme(), &securityCfg);
    if (error != ESP_OK)
    {
        ESP_LOGW(TAG, "Failed to read NVS security cfg: %d. Generate new keys", error);
        error = nvs_flash_generate_keys_v2(nvs_flash_get_default_security_scheme(), &securityCfg);
        if (error != ESP_OK)
        {
            ESP_LOGE(TAG, "failed to generate NVS encr-keys: [0x%02X] (%s)", error, esp_err_to_name(error));

            return -1;
        }
    }

    error = nvs_flash_secure_init_partition(name, &securityCfg);
    if ((error == ESP_ERR_NVS_NO_FREE_PAGES) || (error == ESP_ERR_NVS_NEW_VERSION_FOUND))
    {
        /* this is suggested from ESP documentation */
        ESP_LOGW(TAG, "partition %s was truncated, erasing...", name);
        error = nvs_flash_erase_partition(name);
        if (error != ESP_OK)
        {
            ESP_LOGE(TAG, "partition erase error: %d", error);
        }

        error = nvs_flash_secure_init_partition(name, &securityCfg);
        if (error != ESP_OK)
        {
            ESP_LOGE(TAG, "partition secure init error: %d", error);
[... 301 lines cut ...]
I (2030) gpio: GPIO[5]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (2040) RE: io: initializing i2c
I (2040) gpio: GPIO[15]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 0| Pulldown: 0| Intr:0 
I (2050) gpio: GPIO[3]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 0| Pulldown: 0| Intr:0 
I (2060) hal_flash: init flash partition factory-data
E (2070) nvs_sec_provider: Could not find HMAC key in configured eFuse block!
W (2070) hal_flash: Failed to read NVS security cfg: 61441. Generate new keys
I (2080) efuse: Batch mode of writing fields is enabled
I (2090) efuse: BURN BLOCK9
I (2090) efuse: BURN BLOCK9 - OK (write block == read block)
I (2100) efuse: BURN BLOCK0
E (2100) efuse: BURN BLOCK0 - ERROR (written bits != read bits)
W (2110) efuse: BLOCK0: next retry to fix an error [1/3]...
I (2110) efuse: BURN BLOCK0
E (2120) efuse: BURN BLOCK0 - ERROR (written bits != read bits)
W (2120) efuse: BLOCK0: next retry to fix an error [2/3]...
I (2130) efuse: BURN BLOCK0
E (2140) efuse: BURN BLOCK0 - ERROR (written bits != read bits)
W (2140) efuse: BLOCK0: next retry to fix an error [3/3]...
E (2150) efuse: Written data are incorrect
I (2150) efuse: Batch mode. Prepared fields are committed
E (2160) nvs_sec_provider: Failed to write the HMAC key to eFuse: [0xFFFFFFFF] (ESP_FAIL)
E (2170) hal_flash: failed to generate NVS encr-keys: [0xF003] (ERROR)
E (2170) RE: main: error open factory partition: 4096. Operations will fail!
```

KonstantinKondrashov, 2024-06-04:
Hi @alerighi!
BLOCK0 has the RD_DIS field (read protection field).
This log below says that the RD_DIS is write-protected. You can use `SECURE_BOOT_V2_ALLOW_EFUSE_RD_DIS=`**NO** to prevent it (by default it is set). SECURE_BOOT_V2_ALLOW_EFUSE_RD_DIS can be set later when HMAC is set as well.
>> I (1013) secure_boot: Prevent read disabling of additional efuses...

Here is a Kconfig help:
```
SECURE_BOOT_V2_ALLOW_EFUSE_RD_DIS
            bool "Allow additional read protecting of efuses"
            help
                If not set (default, recommended), on first boot the bootloader will burn the WR_DIS (of) RD_DIS
                efuse when Secure Boot is enabled. This prevents any more efuses from being read protected.
                ....
```

When HMAC KEY is burned then the following fields are set:
- HMAC KEY (BLOCK4 ... BLOCK9) [ok]
  - read protection for key (BLOCK0) [for your case NO, RD_DIS is write protected and it can not be changed].
  - write protection for key (BLOCK0) [ok]
- key_purpose ESP_EFUSE_KEY_PURPOSE_HMAC_... (BLOCK0) [ok]
  - write protection for key_purpose (BLOCK0) [ok]

HMAC keys should be read-protected (for security) otherwise you can read it out with software or espefuse.py. When read-protected is set HMAC keys are all 0s, only hardware can access to the actual data.

(Optionally you can change the order and deploy the HMAC key before the first boot with enabling SB).

alerighi, 2024-06-04:
Thanks for the answer. So, I understand that I need to enable `SECURE_BOOT_V2_ALLOW_EFUSE_RD_DIS` in the bootloader configuration, even if this is a potentially insecure feature (as said in the docs). 

The I guess I need, after enabling HMAC encryption, to burn the efuse that enables the write protection, correct? Is there a function in the IDF that I can use to make this operation?

KonstantinKondrashov, 2024-06-04:
Here is an example - https://github.com/espressif/esp-idf/blob/master/components/bootloader_support/src/esp32c3/secure_boot_secure_features.c#L63
just call `err = esp_efuse_write_field_bit(ESP_EFUSE_WR_DIS_RD_DIS);`, works for all chips.

alerighi, 2024-06-05:
Thanks @KonstantinKondrashov!

It's all clear to me, I can close the issue!

closed 2024-06-05 by alerighi