espressif/esp-idf #13913: Generic set state error

lebrislo, 2024-06-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.1

### Espressif SoC revision.

ESP-32-MINI

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP-32-MINI

### Power Supply used.

USB

### What is the expected behavior?

I have a mesh client that can send ESP_BLE_MESH_MODEL_OP_GEN_POWER_LEVEL_SET messages

```
Error_e mesh_client_sendGenericPowerLevelSet(uint16_t meshAddr, bool ack, uint16_t powerLevel)
{
    Error_e result = E_FAILURE;
    esp_err_t err = ESP_FAIL;

    esp_ble_mesh_client_common_param_t common = {0};
    esp_ble_mesh_generic_client_set_state_t set = {0};

    mesh_client_setCommonMsg(&common, meshAddr,
                             (ack) ? ESP_BLE_MESH_MODEL_OP_GEN_POWER_LEVEL_SET
                                   : ESP_BLE_MESH_MODEL_OP_GEN_POWER_LEVEL_SET_UNACK);
    common.model = g_powerLevelClient->model;

    set.power_level_set.power = powerLevel;

    err = esp_ble_mesh_generic_client_set_state(&common, &set);
    if (err != ESP_OK)
    {
        LOG_ERROR("Failed to send generic power level set");
        goto error;
    }

    result = E_SUCCESS;
error:
    return result;
}
```

When I call this function with the power level value set to 0xFFFF, I'm supposed to received on my other device the same value 0xFFFF. But I receive 0x00FF. (When I send 0xABCD, I receive 0x00AB)

### What is the actual behavior?

I search into the esp-idf to find out what's my problem. And after few research I find out. 

In the file : esp-idf/components/bt/esp_ble_mesh/models/client/generic_client.c at the line 787 : https://github.com/espressif/esp-idf/blob/release/v5.3/components/bt/esp_ble_mesh/models/client/generic_client.c#L787

You are casting a void* value into a struct bt_mesh_gen_power_level_set* set. the set struct has compilation padding, this means that the compilator will always align structure members to "natural" address boundaries - say, int members would have offsets, which are mod(4) == 0 on 32-bit platform.

So you are not casting the right values at the right address. the struct bt_mesh_gen_power_level_set have a total size of 6 bytes before padding but after it's 8 bytes. And so, the copied values are skewed.

To fix this you can change the struct at line https://github.com/espressif/esp-idf/blob/release/v5.3/components/bt/esp_ble_mesh/models/client/include/mesh/generic_client.h#L206

with this :
```
#pragma pack(push, 1)
struct bt_mesh_gen_power_level_set {
    bool     op_en;      /* Indicate whether optional parameters included */
    uint16_t power;      /* Target value of Generic Power Actual state    */
    uint8_t  tid;        /* Transaction Identifier                        */
    uint8_t  trans_time; /* Time to complete state transition (optional)  */
    uint8_t  delay;      /* Indicate message execution delay (C.1)        */
};
#pragma pack(pop)
```

And the code works fine now !

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### More Information.

I didn't try other models messages like this one but maybe the issue is not only on power level

forx157, 2024-10-17:
Hi, @lebrislo 

Thanks for the detailed bug report, it made reproducing this straightforward. Much appreciated.

forx157, 2024-10-17:
And I cannot reproduce the problem you described. The forced conversion of a `void*` pointer to a `gen_power_set` pointer is actually fine, because the content in the `void*` pointer also comes from a structure that adheres to the rules of memory alignment. Unless you have changed the memory alignment method of `set.power_level_set`.

closed 2025-04-07 by Alvin1Zhang