espressif/esp-idf #13915: W (233198) BLE_MESH: Ran out of retransmit attempts - Unable to get composition data

kavimanoj06, 2024-06-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.2.1

### Espressif SoC revision.

ESP32 WROOM-32 v3.0

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32-WROOM-32D

### Power Supply used.

USB

### What is the expected behavior?

When using ESP-BLE-MESH/on-off-server example in esp32-wroom-32d, the nRF mesh app should be able to get the Composition Data successfully 

### What is the actual behavior?

[image]
[image]

### Steps to reproduce.

1. download and install espressif-IDE 
2. import example project (onoff server model)
3. build and flash into esp32 wroom-32 kit
4. try to provision with the nRF mesh app
...

### Debug Logs.

```plain
I (31) boot: ESP-IDF v5.2.1 2nd stage bootloader
I (31) boot: compile time May 31 2024 19:14:16
I (31) boot: Multicore bootloader
I (35) boot: chip revision: v3.0
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 2MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00177000
I (92) boot: End of partition table
I (96) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=360e4h (221412) map
I (180) esp_image: segment 1: paddr=0004610c vaddr=3ffbdb60 size=05414h ( 21524) load
[... 155 lines cut ...]
I (230678) BLE_MESH: Resending 0/2, cred 0x00
I (230678) BLE_MESH: Resending 1/2, cred 0x00
I (230678) BLE_MESH: Resending 2/2, cred 0x00
I (231338) BLE_MESH: Attempts: 1
I (231338) BLE_MESH: Resending 0/2, cred 0x00
I (231348) BLE_MESH: Resending 1/2, cred 0x00
I (231348) BLE_MESH: Resending 2/2, cred 0x00
I (231608) BLE_MESH: Attempts: 1
I (231608) BLE_MESH: Resending 0/2, cred 0x00
I (231608) BLE_MESH: Resending 1/2, cred 0x00
I (231608) BLE_MESH: Resending 2/2, cred 0x00
I (232268) BLE_MESH: Attempts: 0
I (232268) BLE_MESH: Resending 0/2, cred 0x00
I (232278) BLE_MESH: Resending 1/2, cred 0x00
I (232278) BLE_MESH: Resending 2/2, cred 0x00
I (232538) BLE_MESH: Attempts: 0
I (232538) BLE_MESH: Resending 0/2, cred 0x00
I (232538) BLE_MESH: Resending 1/2, cred 0x00
I (232538) BLE_MESH: Resending 2/2, cred 0x00
W (233198) BLE_MESH: Ran out of retransmit attempts
```

### More Information.

i have also tried with the espressif EspBLEMesh app and with ESP32-wroom-32 kit

forx157, 2024-06-17:
hi, @kavimanoj06,
Sorry for the long delay in getting back to you and many thanks for being patient.
Please use the latest idf 5.2 version, the problem has been fixed.

kavimanoj06, 2024-06-17:
Thanks👍, it worked for me.

closed 2024-06-17 by kavimanoj06