espressif/esp-idf #13917: Maximize BLE Reception

diplfranzhoepfinger, 2024-06-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello, 

i use the Example 
https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/ble_eddystone
to receive BLE Tire Sensors, 
and it works quite well. 

BUT: 
when we are e.g. at Exhibitions or congresses, where a LOT of BLE Packets are in the Air, 
the ESP32 gets "blind" to the Sensor Values. 

I guess, the ESP32 receives the Packages, and puts them into an FIFO, and this overflows because of the Many Packets. 
can i extend this Fifo ? 

or i am wrong, and it could be something else ? 

Thanks, 
Franz

diplfranzhoepfinger, 2024-06-07:
Yes. My Question is inaccurate. I first need to know if we must do active or passive scan for our Sensors. Then we can go further.

diplfranzhoepfinger, 2024-06-11:
https://github.com/espressif/esp-idf/blob/8760e6d2a7e19913bc40675dd71f374bcd51b0ae/examples/bluetooth/bluedroid/ble/ble_eddystone/main/esp_eddystone_demo.c#L39

here we have BLE_SCAN_TYPE_ACTIVE. 
with PASSIVE we do not get any Sensors. 

could it be that PASSIVE is broken ? the Manufacturer says that the Sensors use passive Scan ? 

i set 

```
    .scan_interval          = 0x180,
    .scan_window            = 0x180,
```

is this good ? is a longer time better or worse ? 

i set the Filter to only allow the 4 Sensors i have (1 for every Tire on the Vehicle) 

still on Exhibitions we have low reception rates. 

what can we improve ? 

can we reduce WIFI and Expand BLE somehow ? 

or any other Ideas ? 

thanks, 
Franz

diplfranzhoepfinger, 2024-06-12:
Question: 

Will this improve reception ? 

what if first option is set =n ?

```
CONFIG_BT_CTRL_BLE_ADV_REPORT_FLOW_CTRL_SUPP=y
CONFIG_BT_CTRL_BLE_ADV_REPORT_FLOW_CTRL_NUM=1000
```

esp-zhp, 2024-08-07:
Set the scan_type according to your specific scenario. Below is an explanation of the differences between the two types.
The difference between BLE_SCAN_TYPE_ACTIVE and BLE_SCAN_TYPE_PASSIVE is as follows:

**BLE_SCAN_TYPE_ACTIVE**: In this mode, the scanner actively sends scan request packets to devices that are advertising. This allows the scanner to receive additional information from the advertiser, such as the complete list of services or other extended data that the advertiser may choose to send in response.

**BLE_SCAN_TYPE_PASSIVE**: In this mode, the scanner only listens for advertising packets and does not send any requests to the advertising devices. As a result, it receives less information, only what is available in the advertisement itself, but it also conserves more power since it doesn’t transmit any additional data and will find more device.
BLUETOOTH CORE SPECIFICATION Version 5.4 | Vol 6, Part B page 2787
[image]

esp-zhp, 2024-08-07:
"You mentioned that 'with PASSIVE, we do not get any Sensors.' This could be because the Sensors place their information in the scan response (scan rsp) packet, with no useful data in the advertising (adv) packet. If the Sensors cannot be modified, you will need to use an active scan.

We recommend that the Sensors place their information in the advertising packet rather than the scan response packet. This way, the scanner can more easily detect the sensors."

esp-zhp, 2024-08-07:
If there are many packets in the air, we recommend enabling duplicate scan. This way, duplicate packets will only be reported once within a specific time frame.

To enable duplicate scan, set scan_duplicate to true in the esp_ble_scan_params_t structure. You can configure the duplicate scan rules in your project settings.
[image]

These settings can be adjusted using menuconfig. The specific location is as follows:
(Top) → Component config → Bluetooth → Controller Options
[*] BLE Scan Duplicate Options"

diplfranzhoepfinger, 2024-08-07:
we now got very good results with: 

```
    .scan_interval          = 0x180,
    .scan_window            = 0x180,
```

with ACTIVE SCAN. 
Passive Scan is not working, because of the Kind of Sensor we use yes. 
we cannot modify. It is a off-the-shelf Product.

diplfranzhoepfinger, 2024-08-07:
> If there are many packets in the air, we recommend enabling duplicate scan. This way, duplicate packets will only be reported once within a specific time frame.
> 

this we disabled, and found reception getting better. 
As it is a Sensor, we are interested in Every Packet of our Sensor (as we are having Tires, it is 4 at a Time) but not interested in other BLE Packets. 

so disabling duplicate scan was improving our Reception a lot. before we missed packages. 
but this only did not help too much when there were many other stuff around. 

i am usure which helped more, Whitelist or the longer FlowControl list. 

see our sdkconfig.defaults here, which works pretty good now. 
https://gist.github.com/diplfranzhoepfinger/3b3a34e72d0e551c33f4ef0c542634d1

unfortunately with this settings the Wifi AP does not work good anymore. 
but this is expected:

esp-zhp, 2024-08-07:
Whitelist -and then-> FLOW_CTRL

esp-zhp, 2024-08-07:
Whitelist: Filtering at the hardware level can be done by setting a whitelist, allowing scan requests to be sent only to specific devices. This method is the most effective but is limited to adding only 12 devices.

Duplicate Scan: This method filters out duplicate advertising packets. The filtering rules can be set using BTDM_SCAN_DUPL_TYPE, which allows you to filter out reports where both the address and data are identical.

Flow Control: If the application layer does not process advertising reports in a timely manner, some of these reports may be discarded.

Additionally, you should check whether there are any time-consuming operations being performed in the application layer while handling advertising reports.

diplfranzhoepfinger, 2024-08-07:
> Whitelist(Hardware level) --> duplicate scan（controller level）-->FLOW_CTRL （controller level）

I mean this is very cool

closed 2024-10-23 by Alvin1Zhang