espressif/esp-idf #13918: remove_free_block StoreProhibited panic during starting GATT service

Omegaki113r, 2024-06-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-beta1-226-ga36d8bc742-dirt

### Espressif SoC revision.

ESP32S3 chip revision v0.2

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-WROOM-1

### Power Supply used.

USB

### What is the expected behavior?

I work a lot with ESP32 BLE stack therefore i recently started writing my own library on top of ESPIDF API in a way that during startup the ESP32S3 reads a specific file stored in the SPIFFS and configure itself according to that file and starts advertising itself. 

If i use 2 or more services inside a single GATT profile the ESP32S3 panics. As long as i keep 1 service per 1 GATT profile all seems good. But what i expect is to have more than 1 service in a GATT profile.

### What is the actual behavior?

When i try to configure the ESP32S3 to have more than 1 service inside the GATT profile it crashes. 

inside the stack trace i don't see any references that indicate my code is the problem. 

inside tlsf.c,

next->prev_free = prev;

above line seems to be the culprit for the crash. Also the crashes are not happening always. if i restart the ESP32 few times. ESP32 started behaving as expected (configure itself to have more than 1 service inside GATT profile). but 4/5 times the ESP32 crashes during GATT service start "esp_ble_gatts_start_service"

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Debug Logs.

```plain
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4038443e  PS      : 0x00060433  A0      : 0x80383ddc  A1      : 0x3fcb6e80
0x4038443e: remove_free_block at C:/dev/esp/esp_idf_versions/release_5_2/esp-idf/components/heap/tlsf/tlsf.c:332
 (inlined by) block_locate_free at C:/dev/esp/esp_idf_versions/release_5_2/esp-idf/components/heap/tlsf/tlsf.c:567
 (inlined by) tlsf_malloc at C:/dev/esp/esp_idf_versions/release_5_2/esp-idf/components/heap/tlsf/tlsf.c:1005

Backtrace: 0x4038443b:0x3fcb6e80 0x40383dd9:0x3fcb6ea0 0x4037601d:0x3fcb6ec0 0x40376078:0x3fcb6ee0 0x403760ad:0x3fcb6f00 0x40384f29:0x3fcb6f20 0x4203cae9:0x3fcb6f40 0x420335fe:0x3fcb6f70 0x4202be32:0x3fcb6fb0 0x4202c2e0:0x3fcb6fd0 0x42058bd1:0x3fcb6ff0 0x420466ed:0x3fcb7030 0x42046ad5:0x3fcb7050 0x420432eb:0x3fcb7070 0x4037f4ed:0x3fcb70a0
0x4038443b: remove_free_block at C:/dev/esp/esp_idf_versions/release_5_2/esp-idf/components/heap/tlsf/tlsf.c:331 (discriminator 1)
 (inlined by) block_locate_free at C:/dev/esp/esp_idf_versions/release_5_2/esp-idf/components/heap/tlsf/tlsf.c:567 (discriminator 1)
 (inlined by) tlsf_malloc at C:/dev/esp/esp_idf_versions/release_5_2/esp-idf/components/heap/tlsf/tlsf.c:1005 (discriminator 1)

0x40383dd9: multi_heap_malloc_impl at C:/dev/esp/esp_idf_versions/release_5_2/esp-idf/components/heap/multi_heap.c:210

0x4037601d: heap_caps_malloc_base at C:/dev/esp/esp_idf_versions/release_5_2/esp-idf/components/heap/heap_caps.c:179

0x40376078: heap_caps_malloc at C:/dev/esp/esp_idf_versions/release_5_2/esp-idf/components/heap/heap_caps.c:202

0x403760ad: heap_caps_malloc_default at C:/dev/esp/esp_idf_versions/release_5_2/esp-idf/components/heap/heap_caps.c:228

0x40384f29: malloc at C:/dev/esp/esp_idf_versions/release_5_2/esp-idf/components/newlib/heap.c:24

0x4203cae9: aes_cipher_msg_auth_code at C:/dev/esp/esp_idf_versions/release_5_2/esp-idf/components/bt/host/bluedroid/stack/smp/smp_cmac.c:296 

0x420335fe: gatts_calculate_datebase_hash at C:/dev/esp/esp_idf_versions/release_5_2/esp-idf/components/bt/host/bluedroid/stack/gatt/gatt_sr_hash.c:191

0x4202be32: gatt_update_for_database_change at C:/dev/esp/esp_idf_versions/release_5_2/esp-idf/components/bt/host/bluedroid/stack/gatt/gatt_api.c:129

0x4202c2e0: GATTS_StartService at C:/dev/esp/esp_idf_versions/release_5_2/esp-idf/components/bt/host/bluedroid/stack/gatt/gatt_api.c:530      

0x42058bd1: bta_gatts_start_service at C:/dev/esp/esp_idf_versions/release_5_2/esp-idf/components/bt/host/bluedroid/bta/gatt/bta_gatts_act.c:598

0x420466ed: bta_gatts_hdl_event at C:/dev/esp/esp_idf_versions/release_5_2/esp-idf/components/bt/host/bluedroid/bta/gatt/bta_gatts_main.c:128 

0x42046ad5: bta_sys_event at C:/dev/esp/esp_idf_versions/release_5_2/esp-idf/components/bt/host/bluedroid/bta/sys/bta_sys_main.c:499

0x420432eb: osi_thread_run at C:/dev/esp/esp_idf_versions/release_5_2/esp-idf/components/bt/common/osi/thread.c:165

0x4037f4ed: vPortTaskWrapper at C:/dev/esp/esp_idf_versions/release_5_2/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134
```

AxelLin, 2024-06-06:
@Omegaki113r 
v5.2-beta1-226-ga36d8bc742 is quite old version, could you try updating your v5.2 branch?

Omegaki113r, 2024-06-06:
> @Omegaki113r v5.2-beta1-226-ga36d8bc742 is quite old version, could you try updating your v5.2 branch?

I just updated to release/v5.3 and it fixed the issue. i have restarted/power cycled the ESP32S3 multiple times (almost close to 100 times) and the crash didn't happen. 

So i think we can close this issue as resolved. 

But quick question. is there a place/link that i can look up where, when each idf versions were released. 
As you saw i didn't knew v5.2-beta1-226-ga36d8bc742 was very old. So i think it will be good if there is a URL so i can keep track of each release and how old my IDF version is

AxelLin, 2024-06-06:
o i think we can close this issue as resolved.
> 
> But quick question. is there a place/link that i can look up where, when each idf versions were released. 

https://github.com/espressif/esp-idf/releases

closed 2024-06-09 by Omegaki113r