espressif/esp-idf #13919: 10% memory usage increase from IDF 4.4.6 to 4.4.7

hitecSmartHome, 2024-06-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

4.4.6 2nd stage bootloader

### Espressif SoC revision.

ESP-32 Wrover-E revision v3.0

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP-32 Wrover-E

### Power Supply used.

External 3.3V

### What is the expected behavior?

I want my memory back!

### What is the actual behavior?

Huge memory consumption!!

### Steps to reproduce.

Just update from 4.4.6 to 4.4.7

### More Information.

### 4.4.6 system info

```json
"chip": {
  "features": {
    "Embedded Flash": "No",
    "Embedded PSRAM": "No",
    "Wi-Fi 2.4GHz support": "Yes",
    "IEEE 802.15.4 support": "No",
    "Bluetooth Classic support": "Yes",
    "Bluetooth LE support": "Yes"
  },
  "model": "ESP32",
  "cores": 2,
  "revision": 3,
  "fullRevision": 300,
  "featuresRep": 50,
  "idfVersion": "4.4.6"
 },
"memory": {
  "external": {
    "total": 4181375,
    "free": 3903611,
    "largestFree": 3801088
  },
  "internal": {
    "total": 371607,
    "free": 144751,
    "largestFree": 81920
  },
  "systemSpace": {
    "total": 6144000,
    "free": 3248128,
    "used": 2895872
  },
  "userSpace": {
    "total": 4194304,
    "free": 3960832,
    "used": 233472
  }
 }
```

### 4.4.7 system info

```json
"chip": {
  "features": {
    "Embedded Flash": "No",
    "Embedded PSRAM": "No",
    "Wi-Fi 2.4GHz support": "Yes",
    "IEEE 802.15.4 support": "No",
    "Bluetooth Classic support": "Yes",
    "Bluetooth LE support": "Yes"
  },
  "model": "ESP32",
  "cores": 2,
  "revision": 3,
  "fullRevision": 300,
  "featuresRep": 50,
  "idfVersion": "4.4.7"
},
"memory": {
  "external": {
    "total": 4179927,
    "free": 3812051,
    "largestFree": 3801088
  },
  "internal": {
    "total": 366963,
    "free": 104271,
    "largestFree": 40960
  },
  "systemSpace": {
    "total": 6144000,
    "free": 3235840,
    "used": 2908160
  },
  "userSpace": {
    "total": 4194304,
    "free": 4022272,
    "used": 172032
  }
}
```

**This is the exact same code but different IDF versions**

### Main thing to look for:

**4.4.6**
- free memory: 144751 bytes
- largest free block: 81920 bytes

**4.4.7**
- free memory: 104271 bytes
- largest free block: 40960 bytes

What happened and what can i do with this??

Some screenshots about my deployments...

[image]

Where did my ram go and how can i get it back?

AxelLin, 2024-06-06:
It looks strange that the "total" field for both "external" and "internal" are different in different versions.

hitecSmartHome, 2024-06-06:
Strange that you don't experience this.
So this is the part which checks the memory

```cpp
extRam["total"] = heap_caps_get_total_size(MALLOC_CAP_SPIRAM);
extRam["free"] = heap_caps_get_free_size(MALLOC_CAP_SPIRAM);
extRam["largestFree"] = heap_caps_get_largest_free_block(MALLOC_CAP_SPIRAM);

internalRam["total"] = heap_caps_get_total_size(MALLOC_CAP_INTERNAL);
internalRam["free"] = heap_caps_get_free_size(MALLOC_CAP_INTERNAL);
internalRam["largestFree"] = heap_caps_get_largest_free_block(MALLOC_CAP_INTERNAL);
```

This part is totally the same on both versions.

> What type of application are you running, i.e., which components are involved?

I use a lot of components.

- http webserver
- websocket client
- websocket server
- ota
- littlefs
- mdns
- eth
- wifi
- uart

I'm dinamically creating tasks and these also fail after 4.4.7 update because there is not enough ram for them.
One task tries to take 10k, the other try to take 20k of task space. Out of the 40k which is in theory the largest free block
They both fail. I lowered these tasks to 5-5k and now they can be created.

> It looks strange that the "total" field for both "external" and "internal" are different in different versions.

That is indeed strange. I did not see that before.

hitecSmartHome, 2024-06-06:
I have a setup where I set the preferred memory allocation for ps ram ( because I have 4mb external ram and it's just unused by the application......). So in theory every malloc call should allocate from the ext ram. I have also setup these

```cpp
heap_caps_malloc_extmem_enable(2000); // So the system tries to allocate almost everything from the unused 4mb of psram!!!
heap_caps_register_failed_alloc_callback([](size_t requested_size, uint32_t caps, const char *function_name) {
    if (requested_size == 0) { return; }
    logger.write(
        "[SYSTEM] - %s was called but failed to allocate %d bytes with 0x%X capabilities. \n",
        function_name,
        requested_size,
        caps
    );
});
```

hitecSmartHome, 2024-06-06:
Also I'm on arduino 2.0.16 and IDF 4.4.7.
I use arduino as a component of IDF within PlatformIO.
Here is a screenshot of the psram settings in menuconfig

[image]

hitecSmartHome, 2024-06-07:
I also have utility functions from back when this option wasn't available so my app always uses the external ram.
There are several modules uses these utility functions to allocate ram from external.

```cpp
boolean Sys::alloc(char *&ptr, int size) {
    ptr = (char *)ps_malloc(size * sizeof(char));
    if (!ptr || ptr == NULL) { return false; }
    return true;
}

boolean Sys::alloc(uint16_t *&ptr, int size) {
    ptr = (uint16_t *)ps_malloc(size * sizeof(uint16_t));
    if (!ptr || ptr == NULL) { return false; }
    return true;
}

boolean Sys::alloc(uint8_t *&ptr, int size) {
    ptr = (uint8_t *)ps_malloc(size * sizeof(uint8_t));
    if (!ptr || ptr == NULL) { return false; }
    return true;
}

boolean Sys::alloc(uint32_t *&ptr, int size) {
    ptr = (uint32_t *)ps_malloc(size * sizeof(uint32_t));
    if (!ptr || ptr == NULL) { return false; }
    return true;
}
```

If one module needs a block of chunk for example a serialized JSON document, it can call one of these utility functions like this

```cpp
char* buffer;
if( !hshSystem.alloc(buffer, MSG_SIZE) ){ return; }
```

This way I can be sure that it was allocated from the external memory and the menu config settings doesn't matter that much.
But I know idf libraries and other libraries using `malloc()`

Jason2866, 2024-06-07:
mbedtls internal settings have changed. In my experience this is responsible for an increase

hitecSmartHome, 2024-06-07:
Thanks for the suggestion. Unfortunatelly the bluetooth related things are turned off in my app from menuconfig. But will check it out for sure!

0xjakob, 2024-06-07:
You should have the `.map` file of the application available somewhere from the build. You can run `idf_size.py` from the toolchain manually like this: `idf_size.py <mapfile>`.

hitecSmartHome, 2024-06-07:
I see. Will try that. I have installed ESP_IDF 5.2 PowerShell to debug a coredump file saved from the esp's file system, that should do the trick. Thank you for the tip!

hitecSmartHome, 2024-06-07:
Well, it just opens up idf_size.py file for me for some reason...

hitecSmartHome, 2024-06-07:
I have successfully tweaked PIO Inspect. ( I had to change my partition table for it to work.... )

[image]

172.4KB unknown ram usage. These must be the IDF libraries.
I also have this with 169kb

[image]

0xjakob, 2024-06-07:
I'm afraid we need some more information to isolate the problem (or at least see the main priorities when analyzing this issue).

The easiest way would be a minimal reproducing example that builds for both IDF versions. Since this is a build issue, we have the advantage that it does not need to run. You can exclude your own code, unless, of course, the size increase disappears.

Another option to at least check the static memory sizes (and see if they are responsible for the increase) is the .map files which are part of the build output. Are you able to create the .map files for both IDF versions and post them here? If yes, we could check them ourselves on our side.

hitecSmartHome, 2024-06-07:
Will investigate it further myself. Thank you very much. In case i can't figure it out, will send the map files

hitecSmartHome, 2024-06-14:
Okay sorry it turned out that it is a memory leak on one of my libraries I use.
It wasn't idf's fault. Thank you everyone for the help and sorry for the ticket.

closed 2024-06-14 by hitecSmartHome

0xjakob, 2024-06-14:
@hitecSmartHome Welcome! Nevertheless, we're happy this issue could be resolved.