espressif/esp-idf #13920: Cannot connect Bluetooth HF-AG SLC while SPP channel is open

DirkMeintjiesCtrack, 2024-06-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.2.1

### Espressif SoC revision.

ESP32 Rev01

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

Custom board

### Power Supply used.

External 3.3V

### What is the expected behavior?

When calling esp_hf_ag_slc_connect() while having active SPP connection then HF_AG callback with ESP_HF_CONNECTION_STATE_EVT  and state ESP_HF_CONNECTION_STATE_SLC_CONNECTED should occur.

### What is the actual behavior?

When calling esp_hf_ag_slc_connect() **while SPP channel is connected** I get HF_AG callback with ESP_HF_CONNECTION_STATE_EVT and connection state ESP_HF_CONNECTION_STATE_DISCONNECTED instead of ESP_HF_CONNECTION_STATE_SLC_CONNECTED.

FIY: When calling esp_hf_ag_slc_connect() **without having active SPP connection** then HF_AG callback with ESP_HF_CONNECTION_STATE_EVT  and state ESP_HF_CONNECTION_STATE_SLC_CONNECTED occurs successfully.

### Steps to reproduce.

1. Init both SPP server and HF_AG module after Bluedroid init completes.
2. Open an incoming SPP connection
3. Call esp_hf_ag_slc_connect() to establish connection to HF_AG device.

DirkMeintjiesCtrack, 2024-06-12:
Can I bump this issue back to the top?

ESP-YTGerd, 2024-06-24:
Hi @DirkMeintjiesCtrack 
Sorry for the late reply. I combined our IDF HF-AG and SPP-acceptor/initiator examples and ran them together. The AG module acts as an SPP server, maintaining an SPP connection while connecting to HF. The logs show that the HF-AG connection was successfully established.
[image]

DirkMeintjiesCtrack, 2024-06-24:
MMM, that is strange. When I close all my SPP connections, even though the SPP server is still running, then I can connect to HF kit successfully, but not while a SPP connection is active.

Once the HF kit connection is established I can reconnect the SPP connections successfully.

Did you test this on 5.2.1?

Could it be a (out of) memory related issue?

ESP-YTGerd, 2024-06-24:
> MMM, that is strange. When I close all my SPP connections, even though the SPP server is still running, then I can connect to HF kit successfully, but not while a SPP connection is active.
>
> Once the [...]

Yes, It works for me on both release/v5.2.1 and master.  Currently, we support Max 8 RFCOMM connections, so it shouldn't be related to insufficient memory. Maybe can provide some detailed logs to help us with debugging?

DirkMeintjiesCtrack, 2024-06-24:
I will try to capture some logs, thank you.