espressif/esp-idf #13921: 有关I2S配置在4.X与5.X不同的疑惑

murarduino, 2024-06-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

我在努力将项目的代码由4.X版本改为5.X版本，以适应未来的变化。
在I2S外设方面，我发现4.X与5.X有很大的不同。当然大部分配置，还可能尝试理解并修改它。对于有关DMA的选项，使我有些看不懂。我这样理解不知是否正确：
dma_desc_num 相当于4.X中的dma_buf_count，而dma_frame_num相当于4.X中的dma_buf_len。
另外，还有其它需要注意的地方么？我将部分5.X的代码附上。

`#define MIX_CHANNEL_CONFIG(i2s_num, i2s_role,buff_n,frame_n) { \
    .id = i2s_num, \
    .role = i2s_role, \
    .dma_desc_num = buff_n, \
    .dma_frame_num = frame_n, \
    .auto_clear = false, \
    .intr_priority = 0, \
}

/*I2S通信配置  */
    i2s_chan_handle_t amix_handle;
    i2s_chan_config_t chan_cfg = MIX_CHANNEL_CONFIG(I2S_MIX_NUM, I2S_ROLE_MASTER, 16, 256);
    i2s_new_channel(&chan_cfg, &amix_handle, NULL);
    /*标准I2S参数配置  */
    i2s_std_config_t max98357_cfg = {
        .clk_cfg = I2S_STD_CLK_DEFAULT_CONFIG(22050),
        .slot_cfg = I2S_STD_MSB_SLOT_DEFAULT_CONFIG(I2S_DATA_BIT_WIDTH_8BIT, I2S_SLOT_MODE_MONO),
        .gpio_cfg = {
            .mclk = I2S_GPIO_UNUSED,
            .bclk = BCK_PIN,
            .ws = WS_PIN,
            .dout = DOUT_PIN,
            .din = I2S_GPIO_UNUSED,
            .invert_flags = {
                .mclk_inv = false,
                .bclk_inv = false,
                .ws_inv = false,
...
`

murarduino, 2024-06-06:
对了，补充一下。项目中也使用了ADC连续模式，如果我理解没有错的的话，它将点用I2S0的DMA，因此，我特定将I2S_MIX_NUM改为I2S1。

`#define I2S_MIX_NUM I2S_NUM_1 // I2S0(的DMA)被getBEMF中ADC连续转换模式所占用`

L-KAYA, 2025-01-10:
> dma_desc_num 相当于4.X中的dma_buf_count，而dma_frame_num相当于4.X中的dma_buf_len。
对的~
> 另外，还有其它需要注意的地方么？
没啥了，默认配置基本都够用的，其他区别不大。不过看你用的是 8 bit，ESP32上发送数据的时候需要手动补齐到16 bit，有效数据放高八位。
> 如果两个I2S通道都会使用DMA，哪么，会不会有所冲突，或者DMA空间不足以分配两个任务的DMA操作？
不会，两个 I2S 端口的 DMA 是独立的。空间上，如果不够的话可以适当减小 `dma_buf_count` 和 `dma_frame_num`，因为你的采样率不高，通道位数也不多，所以一般都够的。

closed 2025-01-10 by Alvin1Zhang