espressif/esp-idf #13924: IDF5.2.1: Failed to configure netif (duplicate key) with ESP_NETIF_DEFAULT_ETH and single netif instance

espressing, 2024-06-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.1

### Espressif SoC revision.

 ESP32-D0WD rev3

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-WROVER-E-N16R8

### Power Supply used.

USB

### What is the expected behavior?

Based on Ethernet example, it should connect with default ETH configuration that contains default / inherent key, too.

### What is the actual behavior?

With just ESP_NETIF_DEFAULT_ETH, it fails with  esp_netif_lwip: esp_netif_new_api: Failed to configure netif with config=xxxxxx (config or if_key is NULL or duplicate key). 

...where xxxxxx is non-NULL, and neither is the key (obviously). 

The only reason is this condition:
NULL != esp_netif_get_handle_from_ifkey_unsafe(esp_netif_config->base->if_key)  
that is in esp_netif_new_api() call. 

If this is eliminated, it connects absolutely fine!

How can the default key ("ETH_DEF") make esp_netif_get_handle_from_ifkey_unsafe() return non-NULL or this to be a failure condition, when there is only one single NETIF instance as per example and our code, too? 
Why is the default key seen as duplicate if there is no other Netif instance as per below sequence?

See below. 

### Steps to reproduce.

// Initialize TCP/IP network interface aka the esp-netif (should be called only once in application)
ESP_ERROR_CHECK(esp_netif_init());
// Create default event loop that running in background
ESP_ERROR_CHECK(esp_event_loop_create_default());

esp_netif_config_t cfg = ESP_NETIF_DEFAULT_ETH();
esp_netif_t *eth_netif = esp_netif_new(&cfg);   <------------------------ this causes the described error due to duplicate key

  /* attach Ethernet driver to TCP/IP stack */
  ESP_ERROR_CHECK(esp_netif_attach(eth_netif, esp_eth_new_netif_glue(eth_handle)));

### Debug Logs.

```plain
esp_netif_lwip: esp_netif_new_api: Failed to configure netif with config=0x3ffbbbd0 (config or if_key is NULL or duplicate key)
```

### More Information.

By eliminating just that mentioned condition that checks for duplicate key, everything runs and connects perfectly fine. 

Based on exact same example sequence, cannot see how actual sequence is wrong - after a single netif init() and single "new", it decides the default key is a duplicate one... It clearly can't be.

david-cermak, 2024-06-06:
Are you sure that there's no other task, component or function that created the default Ethernet interface before?

I've just tested the code you shared (with IDF `v5.2.1`):
```cpp
void app_main(void)
{
    ESP_ERROR_CHECK(esp_netif_init());
    // Create default event loop that running in background
    ESP_ERROR_CHECK(esp_event_loop_create_default());

    esp_netif_config_t cfg = ESP_NETIF_DEFAULT_ETH();
    esp_netif_t *eth_netif = esp_netif_new(&cfg); // <------------------------ this causes the described error due to duplicate key
    ESP_LOGI(TAG, "%p", eth_netif);
}
```

and it created the netif handle without issues...

espressing, 2024-06-06:
This is quite interesting, thank you - the netif init() and subsequent
calls only occur pre-task creation therefore we could not see how something
would have called this (or a wifi-related netif) initialisation before.

But then it means something is escaping us - since an isolated test
demonstrated it's not inherently wrong.

Thanks again for looking into this, will start a chase for any
unaccounted-for init-time other call or some logic issue that caused it
somehow end up causing actual duplication.

So we can close it as a bug, something else if afoot.

espressing, 2024-06-10:
As quick update, and thanks for confirming model code being OK: there was a default init sideways - after some digging found the culprit outside the normal build-up sequence - so the key duplication was due to this, not the mentioned condition jumping the gun.
Will close this.

closed 2024-06-10 by espressing