espressif/esp-idf #13929: `esp_secure_boot_verify_signature` works incorrectly with already mapped partitions

ermacv, 2024-06-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.2

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

I tried to use signature verification for custom partition and I used function `esp_secure_boot_verify_signature`, but it always failed. I checked how it works and found out that it uses `bootloader_mmap` internally. `bootloader_mmap` returns virtual address to the start of the partition because it is already mapped by IDF internally (at the boot I guess) and return address is not valid (should be address with offset where signature starts but returned address points to the start of the partition).
I expect that even partition is mapped it should return requested virtual address to the caller.

### What is the actual behavior?

`bootloader_mmap` returns virtual address to the start of the partition because it is already mapped by IDF internally.

### Steps to reproduce.

Create custom non app/ota partition.
Enable secure boot.
Try to verify signature of that partition with `esp_secure_boot_verify_signature` function.

### Debug Logs.

```plain
Log when using `esp_secure_boot_verify_signature` function:

W (2461) mmap: paddr block is mapped already, vaddr_start: 0x3c0f0000, size: 0x300000
W (2471) mmap: paddr block is mapped already, vaddr_start: 0x3c0f0000, size: 0x300000
I (2471) secure_boot_v2: Take trusted digest key(s) from eFuse block(s)
E (2471) secure_boot_v2: Secure Boot V2 verification failed.
E (2471) frogfs: Can't verify partition "fs_web" signature (size: 147376)
```
```

### More Information.

I added manual mapping to the start of the partition and manually calculate offset to the signature block and it works as expected, but I have to include private headers:
```
#include "../private_include/bootloader_signature.h"
#include "../private_include/bootloader_utility.h"
#include "bootloader_flash_priv.h"
```
To be able to use `esp_secure_boot_verify_sbv2_signature_block` function.

ermacv, 2024-06-07:
Oh, I made a mistake. The partition is mapped by library that works with this partition, so it is not mapped by IDF internally. But still calling `bootloader_mmap` inside `esp_secure_boot_verify_signature` after partition is already mapped returns not expected result.

Harshal5, 2024-06-13:
Hi @ermacv,

I tried reproducing your use case in a minimal application so that I could confirm if this is similar to your use case and then we could discuss the issue that you are facing. Attaching the minimal project here: 
[sample_project.zip](https://github.com/user-attachments/files/15818503/sample_project.zip)

In this application, similar to what you mentioned in the above "steps to reproduce" section, the custom partition can be considered as my "custom non app/ota partition", signed and flashed the custom partition, enabled secure boot, mmap'ed the partition, and then called `esp_secure_boot_verify_signature()` to verify the custom partition which returned success in my case.

<hr />

You could quickly try out this application using QEMU for ESP32-C3 as well:
1. Install QEMU using:
```
python $IDF_PATH/tools/idf_tools.py install qemu-riscv32
```

2. Build the application
```
idf.py build
```
4. Add the entry of the signed custom partition (`custom_part_signed.bin`) in the `build/flash_args` file to flash it along with the existing artifacts
```
echo "0xc000 ../custom_part_signed.bin" >>  build/flash_args
```
5. Merge the binaries
```
(cd build; esptool.py --chip esp32c3 merge_bin --fill-flash-size 4MB -o flash_image.bin @flash_args)
```
6. Run the application using QEMU:
```
qemu-system-riscv32 -M esp32c3 \
                    -drive file=build/flash_image.bin,if=mtd,format=raw \
                    -drive file=qemu_efuse.bin,if=none,format=raw,id=efuse \
                    -global driver=nvram.esp32c3.efuse,property=drive,value=efuse \
                    -global driver=timer.esp32c3.timg,property=wdt_disable,value=true \
                    -nic user,model=open_eth -nographic -serial mon:stdio
```

(Press CTRL + A and then X to close the QEMU instance)

I have also attached the logs of my run of the application (`custom_part_verification.log`).

closed 2024-07-17 by espressif-bot