espressif/esp-idf #13930: esp_send_now aborts thread

kumado97, 2024-06-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

platformio core 6.1.15 - 3.4.4

### Espressif SoC revision.

esp32-s3-devkitc-1

### Operating System used.

Windows

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32-s3-devkitc-1

### Power Supply used.

USB

### What is the expected behavior?

Am using VSC with platformio Core 6.1.15  ·Home 3.4.4
Have a esp32 S3 woom (makerfabs S3 parallel TFT with Touch 3.5"
The code with esp_send_now did work some months ago 
classical onDatasent and  xtask create
Now it does work not anymore without changes in the code
WIFi ist not started and mode is WiFi_STA

### What is the actual behavior?

the esp_send_now is finished with a zero and the thread calling it aborts

also tried:  WiFi.useStaticBuffers(true);  // it will keep sdkconfig default for cache and static TX buffers in WiFi

### Steps to reproduce.

void OnDataSent(const uint8_t *mac_addr, esp_now_send_status_t status) 
{
  if (status == ESP_NOW_SEND_SUCCESS)
  {
    enow_success = "data sent";
  }
  else
  {
    enow_success = "nothing sent ";
  }
  Serial.print(enow_success);
  // Serial.printf(" - von: %02X:%02X:%02X:%02X:%02X:%02X  an Mac:%02X:%02X:%02X:%02X:%02X:%02X \n", th1_mac_addr[0],th1_mac_addr[1],th1_mac_addr[2],th1_mac_addr[3],th1_mac_addr[4],th1_mac_addr[5],mac_addr[0],mac_addr[1],mac_addr[2],mac_addr[3],mac_addr[4],mac_addr[5]);
}

/// assign outgoing data:
void assign_outgoingreadings(uint8_t nmc)
{ 
  wifi_second_chan_t b;
  uint8_t a;
[... 320 lines cut ...]
anz_mc;1;
pmc1;1;1;1;00:01:00:00:02:01;
pmc2;2;1;2;00:01:00:00:02:02;
pmc3;3;1;3;00:01:00:00:02:03;
pid1;0.100000;0.100000;0.100000;
pid2;0.200000;0.200000;0.200000;
pid3;0.300000;0.300000;0.300000;
dsx1;5;3;2;
dsx2;0;1;4;
dsx3;6;7;8; 17 Zeilen eingelesen
Wifi Scan durchführen - Bitte warten
1 Netzwerke gefunden:
Nr:1  - Name:Kuma-Dach  - RSSI:(-50)dB  - Kanal:11 - Encrypted: j
Ausgewähltes Netzwerk nach index: 0 - Channel für esp-now: 8
 - Check Time set: 01-01-1970  00:00:07 
  - Eigene Mac Adresse: 00:01:00:00:02:00 - WiFi Mac:48:27:E2:07:43:D0
now_wifi_channel = 8
# Peer mc1: 00:01:00:00:02:01 hinzugefügt
regerr_send: 0
ESP-NOW gestartet auf Kanal: 8
```

### More Information.

It has worked 6 months ago properly and now it doesn't !!!??????

0xjakob, 2024-06-10:
@kumado97 I assume you mean `esp_now_send()`, instead of `esp_send_now()`, correct?

kumado97, 2024-06-10:
> [Verfolgungsbild][https://github.com/notifications/beacon/AZCG7CDZRUSZCQG6QMAACV3ZGVLYZA5CNFSM6AAAAABI7AYY6SWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTUATJEJA.gif]

Yes,
but i solved it now. It would help extremely if you add code examples at the bottom of the reference chapters, e. g. how the code would change for the receive call back from core 2. to core 3.
kind regards
Kurt Orthmann

0xjakob, 2024-06-11:
@kumado97 Thanks for letting us know. We're working on improving our documentation and we do plan to introduce how-to sections for each area, which should help in your case.

If the problem is resolved, would you mind giving a brief feedback what solved the issue and close it? This helps other users with a similar issue and let's us easily keep track of issues that haven't been resolved, respectively.

closed 2024-06-14 by Sherry616