espressif/esp-idf #13933: HUB: Root port reset failed on USB host HID example

krouma, 2024-06-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.1

### Espressif SoC revision.

Chip is ESP32-S3 (QFN56) (revision v0.2)

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

Battery

### What is the expected behavior?

After plugging a HID device I expected the console to say what keys were pressed.

### What is the actual behavior?

Instead console says `E (25820) HUB: Root port reset failed` repeatedly. When I unplug the keyboard, it stops.

### Steps to reproduce.

1. Initialize project from USB host HID example.
2. Set target to esp32s3 via USB bridge. I have an external uart chip.
3. Build and flash the esp.

### Debug Logs.

```plain
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x2a (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x178c
load:0x403c9700,len:0x4
load:0x403c9704,len:0xcbc
load:0x403cc700,len:0x2da0
entry 0x403c9914
I (27) boot: ESP-IDF v5.2.1 2nd stage bootloader
I (27) boot: compile time Jun  8 2024 16:51:57
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.2
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (39) boot.esp32s3: SPI Mode       : DIO
I (43) boot.esp32s3: SPI Flash Size : 2MB
I (48) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  2 factory          factory app      00 00 00010000 00100000
I (87) boot: End of partition table
I (91) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=0c294h ( 49812) map
I (108) esp_image: segment 1: paddr=0001c2bc vaddr=3fc92200 size=02760h ( 10080) load
I (111) esp_image: segment 2: paddr=0001ea24 vaddr=40374000 size=015f4h (  5620) load
I (118) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=1e61ch (124444) map
I (147) esp_image: segment 4: paddr=0003e644 vaddr=403755f4 size=0cb60h ( 52064) load
I (164) boot: Loaded app from partition at offset 0x10000
I (165) boot: Disabling RNG early entropy source...
I (176) cpu_start: Multicore app
I (186) cpu_start: Pro cpu start user code
I (186) cpu_start: cpu freq: 160000000 Hz
I (186) cpu_start: Application information:
I (189) cpu_start: Project name:     <redacted>
I (194) cpu_start: App version:      1
I (198) cpu_start: Compile time:     Jun  8 2024 16:51:52
I (204) cpu_start: ELF file SHA256:  87a03f256...
I (210) cpu_start: ESP-IDF:          v5.2.1
I (214) cpu_start: Min chip rev:     v0.0
I (219) cpu_start: Max chip rev:     v0.99 
I (224) cpu_start: Chip rev:         v0.2
I (229) heap_init: Initializing. RAM available for dynamic allocation:
I (236) heap_init: At 3FC951D0 len 00054540 (337 KiB): RAM
I (242) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (248) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (254) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (261) spi_flash: detected chip: gd
I (265) spi_flash: flash io: dio
W (269) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (282) sleep: Configure to isolate all GPIO pins in sleep state
I (289) sleep: Enable automatic switching of GPIO sleep configuration
I (296) main_task: Started on CPU0
I (306) main_task: Calling app_main()
I (306) example: HID Host example
I (336) example: Waiting for HID Device to be connected
E (5726) HUB: Root port reset failed
E (6036) HUB: Root port reset failed
E (6346) HUB: Root port reset failed
E (6656) HUB: Root port reset failed
E (6966) HUB: Root port reset failed
E (7216) HUB: Root port reset failed
```

### More Information.

I tried:

- Setting CONFIG_USB_HOST_CONTROL_TRANSFER_MAX_SIZE to 255, as in #10086
- Unsoldering D+ pull-up resistor (having it is suggested in schematic checklist)
- Other USB host examples, all with the same issue
- Multiple keyboards and mice

I also have 0 ohm resistors and 100nF capacitors to ground on both data traces.

roma-jam, 2024-06-08:
Hi @krouma, 

based on the information it seems tha the problem is in USB signalling, as the device connection is detected, but the root port reset finished with the error. 

Do you have an option to verifry your usb devices with any esp dev kit, that is supported usb host?  

1. When D+ and D- are swapped, the result should be a bit different, but it is also better to double check that. 
2. What is more possible: bad soldering on D+ or D-, so it makes sense to check all soldering points on these lines. 
3. What type of usb port connector you are using on your custom board?

krouma, 2024-06-08:
I don't have any other dev kit around. I checked the connection and D+ and D- are not swapped. The soldering is from a pcb factory and everything seems fine. It is a type A connector and the peripheral is connected directly into it.

There is also an ESD protection chip in the path, I hope I didn't place that the other way around.

krouma, 2024-06-08:
I clicked the wrong button

roma-jam, 2024-06-08:
Hi @krouma,

Could you try `usb_host_lib` example from
`esp-idf`? 

You can use any USB devices, not only keyboards and mice.

Which hardware design you used as a reference to design your custom board?

krouma, 2024-06-08:
Still the same error with no additional info. I followed [https://docs.espressif.com/projects/esp-hardware-design-guidelines/en/latest/esp32s3/schematic-checklist.html](https://docs.espressif.com/projects/esp-hardware-design-guidelines/en/latest/esp32s3/schematic-checklist.html) and [https://docs.espressif.com/projects/esp-hardware-design-guidelines/en/latest/esp32s3/pcb-layout-design.html](https://docs.espressif.com/projects/esp-hardware-design-guidelines/en/latest/esp32s3/pcb-layout-design.html).

krouma, 2024-06-08:
I uploaded KiCad file to [repository](https://gitlab.com/krouma/enigma-kicad).

krouma, 2024-06-10:
I removed the capacitors to ground and now it works.

roma-jam, 2024-06-10:
Hi @krouma ,

thanks for the update. 

Feel free to close the issue if there is nothing more to verify.

closed 2024-06-10 by krouma