espressif/esp-idf #13935: [v5.1.4][s3] `usb_serial_jtag_is_connected` doesn't detect my computer ~10% of the time

chipweinberger, 2024-06-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.4

### Espressif SoC revision.

ESP32-s3

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

custom pcb

### Power Supply used.

USB

### What is the expected behavior?

I expect `usb_serial_jtag_is_connected` to always work

### What is the actual behavior?

~10% of the time, USJ is connected, but `usb_serial_jtag_is_connected` returns false

### Steps to reproduce.

**With `idf.py monitor`:**
1. plug in USJ to PC
2. device boots up
3. run `idf.py monitor`
4. device reboots
5. ~16 seconds after boot, I check `usb_serial_jtag_is_connected` and it returns `false`

### Debug Logs.

```plain
Waiting on /dev/cu.usbmodem2101
found device!
Waiting for Console...
I (2238) esp_image: segment 4: paddr=0058bd70 vaddr=4037576c size=1cb9ch (117660) load
	  > I (2279) boot: Loaded app from partition at offset 0x300000
	  > I (2296) boot: Set actual ota_seq=1 in otadata[0]
	  > I (2296) boot: Disabling RNG early entropy source...
	  > I (2299) cpu_start: Multicore app
	  > I (2299) esp_psram: Found 2MB PSRAM device
	  > I (2300) esp_psram: Speed: 40MHz
	  > I (2300) cpu_start: Pro cpu up.
	  > I (2300) cpu_start: Starting app cpu, entry point is 0x4037590c
	  > I (0) cpu_start: App cpu up.
	  > I (2788) esp_psram: SPI SRAM memory test OK
	  > I (2797) cpu_start: Pro cpu start user code
	  > I (2797) cpu_start: cpu freq: 240000000 Hz
[... 701 lines cut ...]
	  > I (15213) pd mss: stream_file_loop started
	  > I (15233) pd library player: init
	  > I (15233) pd library player: xTaskCreate playerTask
	  > I (15233) pd library player: player task resumed
	  > I (15233) pd library player: execute: exit
	  > I (15233) pd library player: player task suspended
	  > I (15233) pd midi usb: init
	  > I (15233) pd midi usb: xTaskCreate midi_usb_open_loop
	  > I (15243) pd midi usb: midi_usb_open_loop task started
	  > I (15243) pd main: Wait for usb midi to init...
	  > I (15243) pd usbh: oneTimeInit()
	  > I (15243) pd usbh: [oneTimeInit] create xUsbhInitializeMutex
	  > I (15243) pd usbh: [oneTimeInit] create xOpenedDeviceHandlesMutex
	  > I (15243) pd usbh: [oneTimeInit] create queue xDeviceConnectedEvents
	  > I (16213) pd mss: all clients up to date. sleeping
	  > semaphore: 10
	  > I (16243) pd usbh: [openSingleDevice] called
	  > I (16243) pd usbh: [openSingleDevice] all devices closed. continuing...
	  > I (16243) pd usbh: [openSingleDevice] initializing usbh
	  > I (16243) pd usbh: [init] initialize
	  > semaphore: 11
	  > I (16243) pd usbh: [init] got mutex
	  > I (16243) pd usbh: [init] xSemaphoreCreateRecursiveMutex() xClaimedInterfacesMutex
	  > I (16253) pd usbh: [init] usb_host_install()
```

chipweinberger, 2024-06-08:
Using Visual Studio Code I can also repro by doing this:

1. plug in USJ to PC
2. device boots up
3. click 'Terminal' button in VSC
4. device reboots
5. USJ is detected, serial console works (90% of the time!)
.... repeat last 3 steps again and again, until eventually....
6. USJ is not detected, serial console disabled

chipweinberger, 2024-06-21:
@mythbuster5 

@songruo 

@igrr 

Anything I can look into?

chipweinberger, 2024-08-12:
this problem is very annoying

chipweinberger, 2024-08-12:
note, im using full security. flash encryption. jtag disabled. etc.

songruo, 2024-08-16:
@chipweinberger If you increase the value [here](https://github.com/espressif/esp-idf/blob/d7b0a45ddbddbac53afb4fc28168f9f9259dbb79/components/driver/usb_serial_jtag/usb_serial_jtag_connection_monitor.c#L25), does it help you?

chipweinberger, 2024-08-21:
I'll do some testing soon. 

Here is my CMakeLists.txt

```
CONFIG_IDF_TARGET="esp32s3"
CONFIG_BOOTLOADER_WDT_ENABLE=y
CONFIG_BOOTLOADER_WDT_TIME_MS=120000
CONFIG_ESPTOOLPY_FLASHSIZE_8MB=y
CONFIG_ESPTOOLPY_MONITOR_BAUD=921600
CONFIG_PARTITION_TABLE_CUSTOM=y
CONFIG_PARTITION_TABLE_CUSTOM_FILENAME="partitions.csv"
CONFIG_PARTITION_TABLE_OFFSET=0xf000
CONFIG_COMPILER_OPTIMIZATION_PERF=y
CONFIG_COMPILER_CXX_EXCEPTIONS=y
CONFIG_COMPILER_CXX_EXCEPTIONS_EMG_POOL_SIZE=2048
CONFIG_COMPILER_STACK_CHECK_MODE_ALL=y
CONFIG_COMPILER_STACK_CHECK=y
CONFIG_COMPILER_WARN_WRITE_STRINGS=y

CONFIG_BT_ENABLED=y
CONFIG_BT_NIMBLE_ENABLED=y
CONFIG_BT_NIMBLE_MEM_ALLOC_MODE_EXTERNAL=y
CONFIG_BT_NIMBLE_MAX_CONNECTIONS=8
CONFIG_BT_NIMBLE_PINNED_TO_CORE_0=y
CONFIG_BT_NIMBLE_HOST_TASK_STACK_SIZE=5120
CONFIG_BT_NIMBLE_NVS_PERSIST=y
CONFIG_BT_NIMBLE_USE_ESP_TIMER=y
CONFIG_BT_NIMBLE_MSYS_1_BLOCK_COUNT=18
CONFIG_BT_NIMBLE_MSYS_2_BLOCK_COUNT=36

CONFIG_HTTPD_MAX_REQ_HDR_LEN=2048
CONFIG_HTTPD_MAX_URI_LEN=2048
CONFIG_HTTPD_WS_SUPPORT=y

CONFIG_SPIRAM=y
CONFIG_SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY=y
CONFIG_SPIRAM_CLK_IO=30
CONFIG_SPIRAM_CS_IO=26
CONFIG_SPIRAM_USE_MALLOC=y
CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL=94
CONFIG_SPIRAM_TRY_ALLOCATE_WIFI_LWIP=y
CONFIG_SPIRAM_MALLOC_RESERVE_INTERNAL=40960
CONFIG_ESP_CONSOLE_USE_PSRAM=y

CONFIG_ESP_DEFAULT_CPU_FREQ_MHZ_240=y
CONFIG_ESP32S3_DATA_CACHE_16KB=y
CONFIG_ESP_SYSTEM_PANIC_PRINT_REBOOT=y
CONFIG_ESP_SYSTEM_PANIC_REBOOT_DELAY_SECONDS=4
CONFIG_ESP_SYSTEM_EVENT_TASK_STACK_SIZE=4096
CONFIG_ESP_MAIN_TASK_STACK_SIZE=8192

CONFIG_ESP_CONSOLE_UART_CUSTOM=y
CONFIG_ESP_CONSOLE_UART_CUSTOM_NUM_0=y
CONFIG_ESP_CONSOLE_UART_TX_GPIO=43
CONFIG_ESP_CONSOLE_UART_RX_GPIO=44
CONFIG_ESP_CONSOLE_UART_BAUDRATE=921600

CONFIG_ESP_INT_WDT_TIMEOUT_MS=10000
CONFIG_ESP_TASK_WDT_PANIC=y
CONFIG_ESP_TASK_WDT_TIMEOUT_S=60
CONFIG_ESP_IPC_TASK_STACK_SIZE=2048
CONFIG_ESP_TIMER_TASK_STACK_SIZE=4096

CONFIG_ESP_WIFI_STATIC_RX_BUFFER_NUM=6
CONFIG_ESP_WIFI_STATIC_TX_BUFFER=y
CONFIG_ESP_WIFI_TX_BUFFER_TYPE=0
CONFIG_ESP_WIFI_STATIC_TX_BUFFER_NUM=8
CONFIG_ESP_WIFI_CACHE_TX_BUFFER_NUM=32
CONFIG_ESP_WIFI_TX_BA_WIN=8
CONFIG_ESP_WIFI_RX_BA_WIN=12

CONFIG_ESP_COREDUMP_ENABLE_TO_FLASH=y
CONFIG_ESP_COREDUMP_DATA_FORMAT_ELF=y
CONFIG_ESP_COREDUMP_CHECKSUM_CRC32=y
CONFIG_ESP_COREDUMP_ENABLE=y
CONFIG_ESP_COREDUMP_MAX_TASKS_NUM=64
CONFIG_ESP_COREDUMP_USE_STACK_SIZE=y
CONFIG_ESP_COREDUMP_STACK_SIZE=1280

CONFIG_FATFS_LFN_HEAP=y
CONFIG_FATFS_MAX_LFN=255
CONFIG_FATFS_API_ENCODING_ANSI_OEM=y
CONFIG_FATFS_FS_LOCK=16
CONFIG_FATFS_ALLOC_PREFER_EXTRAM=y

CONFIG_FREERTOS_THREAD_LOCAL_STORAGE_POINTERS=3
CONFIG_FREERTOS_IDLE_TASK_STACKSIZE=1880
CONFIG_FREERTOS_TIMER_TASK_STACK_DEPTH=4096
CONFIG_FREERTOS_USE_TRACE_FACILITY=y
CONFIG_FREERTOS_USE_STATS_FORMATTING_FUNCTIONS=y
CONFIG_FREERTOS_VTASKLIST_INCLUDE_COREID=y
CONFIG_FREERTOS_GENERATE_RUN_TIME_STATS=y

CONFIG_FREERTOS_WATCHPOINT_END_OF_STACK=y
CONFIG_FREERTOS_ISR_STACKSIZE=2096
CONFIG_FREERTOS_RUN_TIME_STATS_USING_ESP_TIMER=y
CONFIG_FREERTOS_PLACE_SNAPSHOT_FUNS_INTO_FLASH=y

CONFIG_HEAP_POISONING_COMPREHENSIVE=y
CONFIG_HEAP_ABORT_WHEN_ALLOCATION_FAILS=y

CONFIG_LWIP_MAX_SOCKETS=16
CONFIG_LWIP_MAX_ACTIVE_TCP=32
CONFIG_LWIP_MAX_LISTENING_TCP=32
CONFIG_LWIP_TCP_SND_BUF_DEFAULT=6500
CONFIG_LWIP_TCP_WND_DEFAULT=6500
CONFIG_LWIP_TCPIP_TASK_STACK_SIZE=4096

CONFIG_MBEDTLS_HARDWARE_AES=n
CONFIG_MBEDTLS_MPI_USE_INTERRUPT=n

CONFIG_PTHREAD_TASK_STACK_SIZE_DEFAULT=4096

CONFIG_MDNS_TASK_STACK_SIZE=3072
CONFIG_MONITOR_BAUD=921600
```

chipweinberger, 2024-08-21:
Logs. 

I periodically print the value of `usb_serial_jtag_is_connected()` during boot. You can see it is "1" (true) for nearly the entire boot, but then switches to "0" right before I actually use it. 

```
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x400490d5
0x400490d5: usb_uart_device_rx_one_char in ROM

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3af0,len:0x23e4
load:0x403c9700,len:0x4
load:0x403c9704,len:0x1c94
load:0x403cc700,len:0x2e68
entry 0x403c9a8c
I (32) boot: ESP-IDF v5.1.4 2nd stage bootloader
I (32) boot: compile time Jun  2 2024 22:27:20
I (32) boot: Multicore bootloader
I (33) boot: chip revision: v0.2
I (33) boot.esp32s3: Boot SPI Speed : 80MHz
I (35) boot.esp32s3: SPI Mode       : DIO
I (36) boot.esp32s3: SPI Flash Size : 8MB
I (37) boot: Enabling RNG early entropy source...
I (39) boot: Partition Table:
I (39) boot: ## Label            Usage          Type ST Offset   Length
I (41) boot:  0 jOtaData         OTA data         01 00 00010000 00002000
I (43) boot:  1 jCoredump        Unknown data     01 03 00012000 00010000
I (45) boot:  2 nvsKeys          NVS keys         01 04 00022000 00001000
I (47) boot:  3 jPhyInit         RF data          01 01 00023000 00001000
I (49) boot:  4 efuse_em         efuse            01 05 00024000 00001000
I (50) boot:  5 jOtaAesKey       Unknown data     01 06 00025000 00001000
I (52) boot:  6 jFabInfo         Unknown data     01 06 00026000 00013000
I (54) boot:  7 jStSafe          Unknown data     01 06 00039000 00004000
I (56) boot:  8 jJamcorderUuid   Unknown data     01 06 0003d000 00001000
I (58) boot:  9 nvs              WiFi data        01 02 0003e000 00033000
I (60) boot: 10 jSafeBoot        Unknown data     01 06 00071000 00006000
I (62) boot: 11 jDevCmds         Unknown data     01 06 00077000 00004000
I (63) boot: 12 jLifeStats       Unknown data     01 06 0007b000 00042000
I (65) boot: 13 jImmortal        Unknown data     01 06 000bd000 00008000
I (67) boot: 14 jJfuArgs         Unknown data     01 06 000c5000 00008000
I (69) boot: 15 jJfwmArgs        Unknown data     01 06 000cd000 00008000
I (71) boot: 16 jExtra1          Unknown data     01 06 000d5000 00004000
I (73) boot: 17 jExtra2          Unknown data     01 06 000d9000 00004000
I (75) boot: 18 jExtra3          Unknown data     01 06 000dd000 00002000
I (77) boot: 19 jExtra4          Unknown data     01 06 000df000 00001000
I (78) boot: 20 jJfuApp          OTA app          00 11 000e0000 00220000
I (80) boot: 21 jJfwmApp         OTA app          00 10 00300000 00500000
I (82) boot: End of partition table
E (83) boot: ota data partition invalid and no factory, will try all partitions
I (85) jboot: bootloader_utility_get_selected_boot_partition: -1
I (87) jboot: Clearing ota_data Partition...
I (88) boot_comm: ## Label            Usage Offset   Length   Cleaned
I (121) boot_comm:  0 jOtaData         data  00010000 00002000 [yes]
I (122) boot_comm:  1 jCoredump        data  00012000 00010000 [no]
I (122) boot_comm:  2 nvsKeys          data  00022000 00001000 [no]
I (124) boot_comm:  3 jPhyInit         data  00023000 00001000 [no]
I (125) boot_comm:  4 efuse_em         data  00024000 00001000 [no]
I (127) boot_comm:  5 jOtaAesKey       data  00025000 00001000 [no]
I (129) boot_comm:  6 jFabInfo         data  00026000 00013000 [no]
I (131) boot_comm:  7 jStSafe          data  00039000 00004000 [no]
I (132) boot_comm:  8 jJamcorderUuid   data  0003d000 00001000 [no]
I (134) boot_comm:  9 nvs              data  0003e000 00033000 [no]
I (136) boot_comm: 10 jSafeBoot        data  00071000 00006000 [no]
I (138) boot_comm: 11 jDevCmds         data  00077000 00004000 [no]
I (139) boot_comm: 12 jLifeStats       data  0007b000 00042000 [no]
I (141) boot_comm: 13 jImmortal        data  000bd000 00008000 [no]
[... 784 lines cut ...]
I (8459) pd uart midi: init
I (8459) uart: queue free spaces: 100
I (8459) pd uart midi in: init
I (8459) pd uart midi in: xTaskCreate midi_uart_rx_task
I (8459) pd uart midi in: rx task loop
I (8459) pd uart midi out: init
I (8459) pd uart midi out: allocate msg queue: 44000 bytes
I (8479) pd uart midi out: xTaskCreate midi_uart_tx_task
I (8479) pd uart midi out: UART OUT task started
E (8479) pd main: usb_serial_jtag_is_connected() 1
I (8479) pd mss: init
I (8479) pd mss: xTaskCreate mss
I (8479) pd mss: stream_file_loop started
E (8499) pd main: usb_serial_jtag_is_connected() 0
I (8499) pd midi usb: init
I (8499) pd midi usb: xTaskCreate midi_usb_open_loop
I (8499) pd main: Wait for usb midi to init...
I (8499) pd midi usb: midi_usb_open_loop task started
I (8499) pd usbh: oneTimeInit()
I (8509) pd usbh: [oneTimeInit] create xUsbhInitializeMutex
I (8509) pd usbh: [oneTimeInit] create xOpenedDeviceHandlesMutex
I (8509) pd usbh: [oneTimeInit] create queue xDeviceConnectedEvents
I (9479) pd mss: all clients up to date. sleeping
semaphore: 0
```

chipweinberger, 2024-08-21:
> @chipweinberger If you increase the value [here](https://github.com/espressif/esp-idf/blob/d7b0a45ddbddbac53afb4fc28168f9f9259dbb79/components/driver/usb_serial_jtag/usb_serial_jtag_connection_monitor.c#L25), does it help you?

Yes. I tried a value of 15 instead of 3, and it worked!

I have not tried other values.

songruo, 2024-08-22:
Yah, I guess the default tolerance `ALLOWED_NO_SOF_TICKS` was not enough for your application, maybe due to too many modules are involved, and led to a big delay in handling the FreeRTOS tick interrupts. The value of `USJ_DISCONNECT_CONFIRM_PERIOD_MS` can be increased as you want, just to notice that the larger the value is, the longer it takes to confirm a real disconnection.

chipweinberger, 2024-08-22:
please make the value configurable, without forking ESP-IDF :) 

and you should probably increase the default value too, so that it works. delay is better than a false signal.

richlegrand, 2024-09-23:
After calling nvs_flash_init(), usb_serial_jtag_is_connected() returns false, whereas before calling nvs_flash_init() returns true.  I'm using an esp32-s3 platform, esp-idf 5.2.1, setting  ALLOWED_NO_SOF_TICKS to 15.

chipweinberger, 2026-03-14:
@richlegrand same. For best reliability, you have to check `usb_serial_jtag_is_connected()` early during boot in `main()`. That worked for me.

But this still needs a more proper fix in ESP-IDF