espressif/esp-idf #13937: tusb_composite_msc_serialdevice and adc

Matthias199, 2024-06-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

hey guys is used this for testing and also the adc code 

https://github.com/espressif/esp-idf/tree/release/v5.3/examples/peripherals/usb/device/tusb_composite_msc_serialdevice

But i seems that teh example will not run together with the adc 

fatal error: esp_adc/adc_oneshot.h: No such file or directory
   24 | #include "esp_adc/adc_oneshot.h"

Standalone the adc example works fine in my vscode ... I guess there is something else going on

Dazza0, 2024-06-10:
@Matthias199 I don't see `#include "esp_adc/adc_oneshot.h"` in the example. Could you...

- post the exact commit ID you attempted the build on
- Full build error log showing which source file is calling the `#include "esp_adc/adc_oneshot.h"` statement.

tore-espressif, 2024-06-24:
@Matthias199 This should be resolved by adding `REQUIRES esp_adc` to your `idf_component_register()` command in CMakeLists.txt

closed 2024-06-24 by tore-espressif