espressif/esp-idf #13940: USB serial jtag console does not take input if enter is being pressed too soon

congyue, 2024-06-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

### Setup
- HW: ESP32-S3-DevKitM-1
- IDF_version: 5.2.2 tag
- Code: official [basic console example](https://github.com/espressif/esp-idf/tree/master/examples/system/console/basic)
- The final sdkconfig contains `CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG=y`

### Expected behavior
The console should work as expected:
```
esp32s3>
esp32s3> version
IDF Version:v5.2.1-538-g3b8741b172-dirty
Chip info:
	model:ESP32-S3
	cores:2
	feature:/802.11bgn/BLE/External-Flash:4 MB
	revision number:1
esp32s3>
```

### Actual behavior
If developer tries to press enter too soon when the chip is still booting up, then the console won't take input anymore and the console seems frozen. e.g.
```
esp32s3>
esp32s3> restart
I (14284) cmd_system_commonESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
...
<Keep pressing Enter key while the device is booting>
...
I (214) main_task: Started on CPU0
I (224) main_task: Calling app_main()
I (264) example: Command history enabled

esp32s3> I (774) main_task: Returned from app_main()
<The console won't take any input for this boot cycle, has to power cycle the device from HW>
<However, if there is any periodic logging, they will still be printed properly>
```

The impact for this issue would be:
- Make developer's life harder since we usually use Enter key to test if the console is ready
- When this issue (console frozen) happens, if someone without context tries to attach USB serial, they may consider the device is hang or not responding, which cause further confusion.

If this is a feature instead of bug, may I know how to disable it? Thanks!
@igrr @mythbuster5

0xjakob, 2024-06-11:
@congyue Thanks for noticing this! I could reproduce this and it looks like a bug. We'll take a look and come back here once we know more about this.

SoucheSouche, 2024-07-30:
Closing the issue since it was fixed and merged (see https://github.com/espressif/esp-idf/commit/87b1e455648e4d6cfc7c636296109319fd229f91)

closed 2024-07-30 by SoucheSouche