espressif/esp-idf #13941: Maximise chip power consumption

DCSBL, 2024-06-10:
### Is your feature request related to a problem?

A product is connected to a power supply that is not always capable of providing the current we require, this is especially a problem when the ESP tries to connect to a WiFi network. A brownout can happen when the power supply is not sufficient. 

This brownout can happen at a random moment, but we would like to have control about this situation.

### Describe the solution you'd like.

We would like to 'test' the power supply on boot, so we can detect if the power supply is sufficient. When it detects a brownout we shutdown the ESP and stop trying to turn it on.

Is there some peripheral we can use to draw as much power as the ESP is able to use, to test the power supply?

### Describe alternatives you've considered.

- Start WiFi and connect to a non-existent network → This only peaks the power for a very short time, which is not enough
- Perform some crypto-stuff → Nothing significant happens
- Random ADC measurements → Nothing significant happens

### Additional context.

Our product can be connect to a smart meter, some meters do not provide enough power. When the product is used with an older smart meter that has insufficient power the user has to use an external USB cable to supply power.

kriegste, 2024-06-10:
Open a softAP?
Disable WiFi modem sleep?
Activate pulldown+pullup on every GPIO?

kriegste, 2024-06-10:
If you have a GPIO that is tied to VCC or GND, you could make it an output and create a short circuit... you can change the pad drive strength (see the docs) to a weaker setting to make it less crude.

0xjakob, 2024-06-10:
Is the available current of the power supply not known in advance? Would the users or whoever installs the device need to make sure that the power supply is sufficient? And if the answer is no for both, how about developing a dedicated hardware circuit that draws a specific amount of current?

DCSBL, 2024-06-10:
@kriegste thanks, that GPIO-shortcurcuit is.. scary. but creative. Thanks for the other suggestions, sadly none of them are suitable or are already enabled.

@0xjakob, to give more context; Our users are not 'technical' and have no idea. They just plug in our device in the smart meter and hope for the best. The older smart meters do not provide enough power to enable Wi-Fi.

Our previous solution was to read the hardware version from the smart meter and notify the user when it was an older model. But we have added support for new variants that do not tell the hardware version so we cannot detect it via software...

The product is already in-field so we cannot adjust the hardware.

kriegste, 2024-06-10:
I think the hall sensor of the ESP32 used a short, but relatively high current if enabled. It is not available in the newer IDF versions, though.

esp-lis, 2024-06-27:
@DCSBL perhaps, your can refer to [13624](https://github.com/espressif/esp-idf/issues/13624)

0xjakob, 2024-08-05:
@DCSBL I'm not exactly sure if this helps, but if you need some more or less "realistic" consumption that might be comparable for using WiFi in your application, you could tweak the [UDP client example](https://github.com/espressif/esp-idf/tree/master/examples/protocols/sockets/udp_client) to continuously send messages to a local address. When I do this, I get a consistent average power consumption of more than 100mA.

ESP-Marius, 2024-08-08:
Another suggestion was to use the wifi power save ($IDF_PATH/examples/wifi/powere_save/) example to test for this. if you want to get the maximum power, make sure the option of ESP_PHY_MAX_WIFI_TX_POWER to set to 20 dBm.

ESP-Marius, 2024-08-15:
Since there is a few different approaches suggested here I think we can consider this one closed. Feel free to reopen if you think there is anything missing on our side @DCSBL

closed 2024-08-15 by ESP-Marius

DCSBL, 2024-08-15:
Yeah, sorry for not replying. This project was abandoned by us so I forgot about this issue. Thank you all for the suggestions.