espressif/esp-idf #13942: `i2c_master_cmd_begin()` always returns `ESP_ERR_TIMEOUT` on ESP-IDF 5.2.2 (fine on ESP-IDF 5.0.3)

RobMeades, 2024-06-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.2

### Espressif SoC revision.

ESP32-D0WD (revision v1.0)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-DevKitC

### Power Supply used.

USB

### What is the expected behavior?

Our code, using the legacy I2C API, works perfectly fine with ESP-IDF version 5.0.3.  We would now like to move to the latest version, ESP-IDF version 5.2.2, still using the legacy I2C API.

However when we do so `i2c_master_cmd_begin()` _always_ returns `ESP_ERR_TIMEOUT` (0x107); `ERR_OK` with 5.0.3, always `ESP_ERR_TIMEOUT` with 5.2.2.

We've checked the release notes for changes between the two versions but nothing is mentioned about changes in I2C aside from the introduction of the new I2C API.  There has been a bug fix in I2C timing, 793ca6d7a48120722e848f408e363345ee63fd09, but that doesn't _look_ as though it should cause what we are seeing.

### What is the actual behavior?

See above.

### Steps to reproduce.

Our full code can be found [here](https://github.com/u-blox/ubxlib/blob/1e6b3583ad21cd79c82018949510ff652891c880/port/platform/esp-idf/src/u_port_i2c.c#L182); below I have paraphrased it so that you can see more easily our configuration etc. (all of which is the same for 5.0.3),

`pData` is passed in and points to the value `0xff`:

```
i2c_config_t cfg = {0};

cfg.mode = I2C_MODE_MASTER;
cfg.sda_io_num = 21;
cfg.scl_io_num = 22;
cfg.sda_pullup_en = true;
cfg.scl_pullup_en = true;
cfg.master.clk_speed = 100000;
cfg.clk_flags = 0;

if ((i2c_param_config(0, &cfg) == ESP_OK) &&
    (i2c_set_timeout(0, 800000) == ESP_OK) &&  // 10 ms timeout at 80 MHz APB
    (i2c_driver_install(0, I2C_MODE_MASTER, 0, 0, 0) == ESP_OK)) {
    int32_t x = i2c_master_start(cmd);
    printf("### A i2c_master_start(cmd) returned 0x%02x.\n", x);
    x = i2c_master_write_byte(cmd, 0x84, true); // 7-bit device address 0x42
    printf("### B i2c_master_write_byte(cmd, 0x84) returned 0x%02x.\n", x);
    if (pData != NULL) {
        x = i2c_master_write(cmd, (const uint8_t *) pData, size, true);
        printf("### C i2c_master_write(cmd, (const uint8_t *) pData, size, true) returned 0x%02x, pData %p, *pData 0x%02x, size %d.\n", x, pData, *pData, size);
    } else {
        printf("### C pData is NULL.\n");
    }
    x = i2c_master_stop(cmd);
    printf("### D i2c_master_stop(cmd) returned 0x%02x.\n", x);
    x = i2c_master_cmd_begin(handle, cmd, 0);
    printf("### E i2c_master_cmd_begin(0, cmd, 0) returned 0x%02x.\n", x);
    i2c_cmd_link_delete(cmd);
}
```

...and this prints out:

```
### A i2c_master_start(cmd) returned 0x00.
### B i2c_master_write_byte(cmd, 0x84) returned 0x00.
### C i2c_master_write(cmd, (const uint8_t *) pData, size, true) returned 0x00, pData 0x3ffbfc78, *pData 0xff, size 1.
### D i2c_master_stop(cmd) returned 0x00.
### E i2c_master_cmd_begin(0, cmd, 0) returned 0x107.
```

### Debug Logs.

```plain
We have tried setting

#define LOG_LOCAL_LEVEL ESP_LOG_VERBOSE
#include "esp_log.h"

...and then calling:

esp_log_level_set("esp32-hal-i2c", ESP_LOG_DEBUG);

...before the code above but we never seem to get any more debug out of ESP-IDF.
```

RobMeades, 2024-06-10:
Gah, no, ignore me: 793ca6d7a48120722e848f408e363345ee63fd09 _was_ the relevant change and the issue was with us, not with ESP-IDF.

We had always passed 0 as the final parameter to `i2c_master_cmd_begin()`, thinking that would result in an infinite timeout, but of course we _should_ have been passing `portMAX_DELAY` and the thing that 793ca6d7a48120722e848f408e363345ee63fd09 fixed had been masking our mistake all this time (four years!).

Setting the last parameter of `i2c_master_cmd_begin()` to `portMAX_DELAY` instead fixes the problem.

kiss81, 2024-06-26:
@RobMeades I do have a similar issue with v5.2.2. Is there a hotfix I can pull?

kiss81, 2024-06-26:
> @kiss81: if you mean a hotfix for the issue in [ubxlib](https://github.com/u-blox/ubxlib) code, it was fixed in [u-blox/ubxlib@ef194bf](https://github.com/u-blox/ubxlib/commit/ef194bf81ea502a30c73fe5d1e98f398e12ba33a); there is no problem in the ESP-IDF code, we (I) had just misinterpreted their API.

Thank you. I thought it was an IDF issue, but found the issue in my code now as well:  2ms timeout with a 200Hz tick clock wasn't working too well. ;)