espressif/esp-idf #13944: SHT31 I2C communication fails after SNTP gets time

paduc77, 2024-06-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.2.1-dirty

### Espressif SoC revision.

ESP32 v3.0

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32 LoLin

### Power Supply used.

USB

### What is the expected behavior?

SHT31 and SNTP work together without any issue.

### What is the actual behavior?

SHT31 measure the temperature and humidity perfectly but when the SNTP time synchronization is finished the I2C communication  will be broken and ESP can't communicate with the sensor anymore. The strange thing is that on the same pins there is a OLED display which continoues working.

### Steps to reproduce.

Connect SHT31 to ESP32 board.
Connect WiFi.
Synchronize time.

### Debug Logs.

```plain
-
```

### More Information.

-

0xjakob, 2024-06-11:
@paduc77 If an OLED display is connected to the very same pins as the SHT31, isn't it possible that these two devices interact with each other and cause the issue?

paduc77, 2024-06-11:
@0xjakob Thanks for the hints, but without SNTP or when the SNTP is not synchronized the OLED and SHT31 work together without any issue.

AxelLin, 2024-11-06:
@paduc77 Does  https://github.com/espressif/esp-idf/issues/12860#issuecomment-2413975957 work for you?

mythbuster5, 2025-02-07:
So long time no reply, I will close this issue. Feel free to reopen it if there is more update, thanks!

closed 2025-02-07 by mythbuster5