espressif/esp-idf #13946: ESP32-S3: Permanently disabling USB-JTAG controller

benjaminaigner, 2024-06-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Following problem:
On some Win11 machines, the USB-OTG interface is kicked out, because of the sudden changes between the USB-OTG & USB-JTAG/serial interfaces in the ESP32-S3.

So I thought disabling it via eFuses:
* DIS_USB_JTAG
* DIS_USB_SERIAL_JTAG

But: since then, my chip is bricked. Entering the UART0 download mode results in an immediate core panic, without the possibility to recover:
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x0 (DOWNLOAD(USB/UART0))
waiting for download
uru Meditation Error: Core 0 panic'ed (LoadProhibited)
Core 0 register dump:
PC      : 0x400511b1  PS      : 0x00060330  A0      : 0x80049188  A1      : 0x3fceb640
A2      : 0x00000000  A3      : 0x3ff1e3fb  A4      : 0x00000001  A5      : 0x00000000
A6      : 0x00000000  A7      : 0x00000088  A8      : 0x800533dc  A9      : 0x3fceb620
A10     : 0xfffffffb  A11     : 0x3fceeebc  A12     : 0x3fceefbc  A13     : 0x00000000
A14     : 0x00000000  A15     : 0x00000006  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x400570e8  LEND   : 0x400570f3  LCOUNT  : 0x00000000

Backtrace: 0x400511b1:0x3fceb640 0x40049185:0x3fceb670 0x400491e5:0x3fceb690 0x40043917:0x3fceb6b0 0x4004392a:0x3fceb6d0 0x40043c33:0x3fceb6f0 0x40034c45:0x3fceb710
```

I found no information in the documentation that this setting can brick my chip, maybe this is just related to the bootloader code.

Please provide instructions how to permanently disable the USB Serial/JTAG controller in the chip and if possible how to unbrick my ESP32-S3.

Thank you very much, keep up with the good work!

igrr, 2024-06-13:
Hi @benjaminaigner, so far I can only say that this indeed seems to be a bug in the boot ROM. We do have an internal issue report about it, but we only met this once and didn't pursue debugging the issue.

I'm afraid there probably isn't a way to get the chip back to the working state, you'll have to use a new one.

We'll confirm which eFuse is sufficient to disable USB Serial/JTAG and get back to you.

terrafirma2021, 2024-06-27:
@igrr 

Can you kindly confirm the correct efuse to burn please, Thanks

terrafirma2021, 2024-08-24:
Hi, for anybody who may find themselves here, 
DIS_USB_JTAG will not stop the USB JTAG from appearing when the mcu boots!

To prevent USB/JTAG from appearing at all, use the following command:
python espefuse.py --port <COM_PORT> burn_efuse USB_PHY_SEL

Running this command will stop the MCU from appearing at all forever, as the internal PHY selector will always be in OTG mode
If you want to enable USB CDC to flash again, you must hold down the Boot button when connecting the device to your computer; otherwise, your microcontroller won't show up. Please note that USB JTAG functionality will be permanently disabled.

[image]

You can find more details about this topic in the documentation at: [Espressif USB PHY Documentation](https://docs.espressif.com/projects/esp-iot-solution/en/latest/usb/usb_overview/usb_phy.html).

[image]

Note: For ESP32-S3 modules and development boards produced before DateCode 2219 (PW No. earlier than PW-2022-06-XXXX), because EFUSE_DIS_USB_OTG_DOWNLOAD_MODE (BLK0 B19[7]) has already been burned to 1 (USB OTG download disabled), if users burn the efuse_usb_phy_sel bit to 1, it will result in both USB-Serial-JTAG and USB-OTG download functionalities being disabled when the chip enters Boot mode.

terrafirma2021, 2024-08-24:
@benjaminaigner
Here is your fix, :)

closed 2024-09-04 by KonstantinKondrashov

zjanosy, 2025-04-25:
I've ran into this same issue after burning DIS_USB_SERIAL_JTAG. The UART download and the efuse programmer does not connect anymore. I cannot even program the chip via the external JTAG port using ESP-PROG.

Is there any way to unbrick it, or replacing the module is the only solution?

DanielSenseAI, 2025-05-25:
Hello zjanosy, did you manage to find a fix?

I burnt the same EFUSE and cant set the UART download mode anymore. However, normal UART communication seems to be working.