espressif/esp-idf #13947: vTaskSuspendAll bug can cause incorrect core to be suspended.

DirkMeintjiesCtrack, 2024-06-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.2.1

### Espressif SoC revision.

N/A

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

Any

### Power Supply used.

USB

### What is the expected behavior?

vTaskSuspendAll must suspend the core on which the task is run

### What is the actual behavior?

It is possible for vTaskSuspendAll to suspend the wrong core. Consider the code below from xt_utils.h, if TaskA calls vTaskSuspendAll and taskB is made to run after the "rsr.prid %0" instruction then TaskA moves to another core, but prid was already read and so xt_utils_get_core_id() will now return wrong core ID, and then wrong core is suspended.

`
FORCE_INLINE_ATTR __attribute__((pure)) uint32_t xt_utils_get_core_id(void)
{
    /*
    Note: We depend on SOC_CPU_CORES_NUM instead of XCHAL_HAVE_PRID as some single Xtensa targets (such as ESP32-S2) have
    the PRID register even though they are single core.
    */
#if SOC_CPU_CORES_NUM > 1
    // Read and extract bit 13 of special register PRID
    uint32_t id;
    asm volatile (
        "rsr.prid %0\n"
        "extui %0,%0,13,1"
        :"=r"(id));
    return id;
#else
    return 0;
#endif // SOC_CPU_CORES_NUM > 1
}
`

### Steps to reproduce.

Repeatedly call vTaskSuspendAll() and then xTaskResumeAll(), after a while you get assert failed:

`assert failed: xTaskResumeAll tasks.c:2554 (( ( xTaskGetSchedulerState() == ( ( BaseType_t ) 0 ) ) ? ( ( BaseType_t ) 1 ) : ( ( BaseType_t ) 0 ) ) == ( ( BaseType_t ) 1 ))`

AxelLin, 2024-06-11:
Check if v5.2.2 fixed this issue.

https://github.com/espressif/esp-idf/releases/tag/v5.2.2 mentioned
Fixed a bug where vTaskSuspendAll() could suspend the wrong core (9f4f8e2)

DirkMeintjiesCtrack, 2024-06-12:
v5.2.2 solves it, thx.

closed 2024-06-12 by DirkMeintjiesCtrack