espressif/esp-idf #13952: New I2C API does not allow stop bit to be suppressed between write and read.

RobMeades, 2024-06-11:
### Is your feature request related to a problem?

With the old I2C API one could construct an I2C command to include a stop bit or not.  This is required for I2C devices which allow writing of a memory address byte or bytes, followed by no stop bit, then the data from that memory address may be read; this ONLY works when there is no intervening stop bit between the two.

We can see no way to do this with the new I2C API: have we missed something?

### Describe the solution you'd like.

Need to be able to suppress transmission of a stop bit in the I2C master write API, as one could before.  Otherwise we cannot use the new API with our devices and would ask that you please do NOT deprecate the old one.

### Describe alternatives you've considered.

We will continue to use the old I2C API until it is possible to use the new one.

### Additional context.

All u-blox GNSS modules require this operation in order to determine the number of bytes buffered in the GNSS device; write 0xFD followed by no stop bit, perform a two-byte read and that gives the master the number of bytes that the GNSS device has buffered.

o-marshmallow, 2024-06-13:
Hello @RobMeades ,

From what I understand, you are trying to read the register `0xFD` out of your I2C device, which returns 2 bytes. The new API does implement this I2C type of transaction, as documented in our official API reference: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/i2c.html#i2c-master-write-and-read

What you should do in your case is the following:
```
uint8_t reg = 0xFD;
uint8_t value[2];
ESP_ERROR_CHECK(i2c_master_transmit_receive(<your_dev_handle>, &reg, 1, value, 2, <your_timeout>));
```

RobMeades, 2024-06-13:
But that will send a stop bit at the end of the transmit, which we cannot have, the device _requires_ the transmit and the receive I2C operations to _not_ be separated by a stop bit.  This is not an uncommon I2C operation, it is sometimes called "repeated start bit".

RobMeades, 2024-06-13:
For more information, see here:

https://www.i2c-bus.org/repeated-start-condition/

RobMeades, 2024-06-13:
Ah, OK, I didn't see any mention of this in the documentation, I assumed that a "transmit/receive" was just what it said, a transmit followed by a receive.

Looking more closely, I see that the diagram further up in the description does show that there is no intervening stop bit, but I think it is worth you updating the documentation of the API function, `i2c_master_transmit_receive()`, to explicitly say that there is no stop bit between the transmit and the receive; I know I2C reasonably well so I didn't read the "getting started" part!  I could just find no mention of "repeated start" or "no stop" in the API at all, so I assumed it wasn't there :-).

o-marshmallow, 2024-06-13:
@RobMeades No problem, I understand that the naming can be confusing, since there is no official name for the write-read operation in the I2C specification, you will find it under different naming depending on the SDKs :smile: 

 We will try to make it more clear and  maybe also state that is this often used to read I2C devices registers

o-marshmallow, 2024-06-24:
The API documentation has been updated internally, it will be pushed to https://docs.espressif.com on the next sync

dpnebert, 2024-09-05:
> Ah, OK, I didn't see any mention of this in the documentation, I assumed that a "transmit/receive" was just what it said, a transmit followed by a receive.
> 
> Looking more closely, I see that the diagram further up in the description does show that there is no intervening stop bit, but I think it is worth you updating the documentation of the API function, `i2c_master_transmit_receive()`, to explicitly say that there is no stop bit between the transmit and the receive; I know I2C reasonably well so I didn't read the "getting started" part! I could just find no mention of "repeated start" or "no stop" in the API at all, so I assumed it wasn't there :-).
> 
> Also, someone who doesn't know I2C might assume the same thing, that this function does a normal transmit followed by a normal receive, since many I2C APIs have such an "exchange" or "transaction" type function, and then they would get very confused when it didn't work in standard operation where a stop bit _is_ required.

Yes, that is exactly what is going on for me. [https://github.com/espressif/esp-idf/issues/14336#issuecomment-2332152351](https://github.com/espressif/esp-idf/issues/14336#issuecomment-2332152351)

I assume that would only do one transmit followed by one receive.  Useful for setting up which register is to be read from.  But the example in the API is confusing to me:
```
uint8_t buf[20] = {0x20};
uint8_t buffer[2];
ESP_ERROR_CHECK(i2c_master_transmit_receive(dev_handle, buf, sizeof(buf), buffer, 2, -1));
```
I read that as:

Transmits 0x20 to select the register, then writes 0x20 to register 0x20.
Transmits 0x20 to select the register, then writes 0x20 to register 0x20.
Transmits 0x20 to select the register, then writes 0x20 to register 0x20.
Transmits 0x20 to select the register, then writes 0x20 to register 0x20.
Transmits 0x20 to select the register, then writes 0x20 to register 0x20.
Transmits 0x20 to select the register, then writes 0x20 to register 0x20.
Transmits 0x20 to select the register, then writes 0x20 to register 0x20.
Transmits 0x20 to select the register, then writes 0x20 to register 0x20.
Transmits 0x20 to select the register, then writes 0x20 to register 0x20.
Transmits 0x20 to select the register, then writes 0x20 to register 0x20.
Receives value from register 0x20.
Receives values from register 0x21

That doesn't seem at all correct, but it's a good starting point to get corrected.