espressif/esp-idf #13953: Linux Host Unit Testing Coverage

wilmer-suarez-simplisafe, 2024-06-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm using version 5.0.2 of ESP-IDF.

I'm trying to get code coverage reports working for a custom component that I have existing unit tests for. I'm trying to set this up using CMake + ESP-IDF. I'm using ESP-IDFs [nvs_flash](https://github.com/espressif/esp-idf/tree/release/v5.0/components/nvs_flash/host_test/nvs_page_test) component as an example.

I'm able to generate the `.gcno` files during compilation for both my test source file (file containing the unit tests) and the component under test source files. When I execute the resulting program, a `.gcda` file is only generated for the test source file and not the component under test (functions for this are called from the test source file successfully). Anyone know how to resolve this?

This is my current directory structure:

```
├── CMakeLists.txt (1)
├── component_file.c
├── component_file.h
├── host_test
 │   ├── component_unit_tests
 │   │   ├── CMakeLists.txt (2)
 │   │   ├── main
 │   │   │   ├── CMakeLists.txt (3)
 │   │   │   └── component_unit_tests.c
 │   │   └── sdkconfig.defaults
```

These are the contents of my CMakeLists.txt files: https://codefile.io/f/WEbfCgizrt

0xjakob, 2024-06-12:
@wilmer-suarez-simplisafe It looks like you didn't add the coverage flags to the source files in CMake. You can achieve this for an entire component by adding these lines after the `idf_component_register()` function call in the component's `CMakeLists.txt` file:
```
target_compile_options(${COMPONENT_LIB} PUBLIC --coverage)
target_link_libraries(${COMPONENT_LIB} PUBLIC --coverage)
```
Does this help?

Coverage is only generated if these flags are added. The documentation about coverage on Linux is not exactly enlightening, because we wanted to provide a different mechanism than what has been added to the [nvs_page_test application you mentioned above](https://github.com/espressif/esp-idf/tree/release/v5.0/components/nvs_flash/host_test/nvs_page_test).

0xjakob, 2024-06-12:
@wilmer-suarez-simplisafe OK, I see `--coverage` flags have already been added. Could you check that the compiler and linker commands also contain the `--coverage` flags? You can see the full compiler and linker commands by using `idf.py -v build` and re-build at least the corresponding source file, and link it.

wilmer-suarez-simplisafe, 2024-06-12:
@0xjakob Below is the build output (cleaned) generated by `idf.py -v build`. Looks like `--coverage` is being used by both the compiler and linker.
[output.txt](https://github.com/user-attachments/files/15807094/output.txt)

ottovis, 2024-06-13:
I had the same issue, and for me the issue is that the .elf executable does not exit upon completion of running the tests. As the hooks generating the gcda files are only called on exit, you will need to (somewhat) gracefully exit the program. 

I did this by running the executable with subprocess.Popen and scanning the output until seeing the OK message and then exiting the Python script. This fixed the issue for me, however dumb that might be.

wilmer-suarez-simplisafe, 2024-06-13:
@ottovis My issue is that the `.gcda` file is being generated, just not for all the source files that have `.gcno`. The resulting `.elf` file always exists with err code 0. I tried to do the same as you suggested in Python and it ended with the same result:
Test source code `.gcda` gets generated but not for the component under the test source code (even though the compilation generated its `.gcno` files).

0xjakob, 2024-06-13:
@wilmer-suarez-simplisafe Do you have a reproducing example that we could examine on our side? I couldn't reproduce the issue on my side. A reproducing example (i.e., an entire app including the build and configuration files) would really help us to analyze the issue further.

wilmer-suarez-simplisafe, 2024-06-13:
Here's a barebones component I put together:
[coverage_test_component.zip](https://github.com/user-attachments/files/15825916/coverage_test_component.zip)

I just placed this in an ESP-IDF (v5.0.2) environment and ran `idf.py build`. I left the generated `build` and `sdkconfig` in just incase that might be helpful.

0xjakob, 2024-06-14:
@wilmer-suarez-simplisafe How do you generate the coverage report? Are you using `lcov` and `genhtml` similar to the how it's done in the [nvs_page_test project](https://github.com/espressif/esp-idf/blob/v5.0.2/components/nvs_flash/host_test/nvs_page_test/CMakeLists.txt)?

Because when I try the test project, the coverage is reported for both the test source `coverage_test_unit_tests.c` and the component's source file `test_comp.c` (I ran the application; then ran the `lcov` and `genhtml` commands manually). However, I can also just find the `.gcda` file for `coverage_test_unit_tests.c`, not for `test_comp.c`:
```
$ find . -name "*.gcda"
./build/esp-idf/main/CMakeFiles/__idf_main.dir/coverage_test_unit_tests.c.gcda
```

wilmer-suarez-simplisafe, 2024-06-14:
Okay... I was using `gcovr` to generate the report. It only shows the test source code coverage:
[image]

but yeah, I didn't try to use `lcov` + `genhtml`. That seems to work:
[image]

I assumed its because the `.gcda` file isn't being generated.

0xjakob, 2024-06-14:
I could reproduce this exact behavior of `gcovr` when running `gcovr` in the host test application directory (`coverage_test_component/test_coverage/host_test/coverage_test`)
. However, when I run it in the component directory (`coverage_test_component/test_coverage`) above, it also shows the coverage of `test_comp.c`. Can you confirm this?
[image]

It seems to me that `gcovr` has a different search strategy for coverage artifacts. It also seems to look for source files first and then for related coverage artifacts, which prunes away everything but the application's components, if `gcovr` is ran in the application's directory. The same thing happens when `gcovr` is ran in the `nvs_page_test` project, btw.

Note that *still no `.gcda` file is ever generated on my machine*, while the *coverage report is complete*.

wilmer-suarez-simplisafe, 2024-06-14:
Yes, I'm able to see coverage for all files now. Thanks a lot!

wilmer-suarez-simplisafe, 2024-06-14:
Resolved.

closed 2024-06-14 by wilmer-suarez-simplisafe