espressif/esp-idf #13955: undefined reference to `esp_xt_wdt_init' on esp32c6

rkit1, 2024-06-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.2-89-gcc9edf2085

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

build

### What is the actual behavior?

does not

### Steps to reproduce.

CONFIG_RTC_CLK_SRC_EXT_CRYS=y
CONFIG_ESP_XT_WDT=y
CONFIG_ESP_XT_WDT_TIMEOUT=200
CONFIG_ESP_XT_WDT_BACKUP_CLK_ENABLE=y

0xjakob, 2024-06-12:
@rkit1 We could confirm this issue, thanks for noticing! We'll take a look at this and come back here once we know more.

0xjakob, 2024-06-12:
@rkit1 Only the C3, S2 and S3 chips have an [XTAL32K Watchdog Timer](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/system/wdts.html#xtal32k-watchdog-timer-xtwdt). The configuration options didn't really reflect this. We'll change this so that there shouldn't be any accidental configuration of the XTAL32K Watchdog Timer anymore.

closed 2024-06-12 by rkit1