espressif/esp-idf #13957: OpenThread: Neighbor Solicitation responses are being filtered

manuelm, 2024-06-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Since https://github.com/openthread/openthread/commit/53305d47620e8ba8019f8e9058714c3dc12930e3 the border router is sending neighbor solicitation packets to routers, however the neighbor advertisement (NA) responses are never passed to `otPlatInfraIfRecvIcmp6Nd`. I've verified this by dumping all received ICMPv6 packets and I don't see any NA packets. Router adverts (RA) and router solicit (RS) packets work fine.

The issue can also be observed from the log output. e.g:
```
I(108804) OPENTHREAD:[I] RoutingManager: Sent Neighbor Solicitation to fe80:0:0:0:acbd:dcff:fe36:fb5f - attempt:1/5
[...]
I(109814) OPENTHREAD:[I] RoutingManager: Sent Neighbor Solicitation to fe80:0:0:0:acbd:dcff:fe36:fb5f - attempt:2/5
[...]
I(110824) OPENTHREAD:[I] RoutingManager: Sent Neighbor Solicitation to fe80:0:0:0:acbd:dcff:fe36:fb5f - attempt:3/5
[...]
I(111834) OPENTHREAD:[I] RoutingManager: Sent Neighbor Solicitation to fe80:0:0:0:acbd:dcff:fe36:fb5f - attempt:4/5
[...]
I(112844) OPENTHREAD:[I] RoutingManager: Sent Neighbor Solicitation to fe80:0:0:0:acbd:dcff:fe36:fb5f - attempt:5/5
[...]
I(114854) OPENTHREAD:[I] RoutingManager: No response to all Neighbor Solicitations attempts from router fe80:0:0:0:acbd:dcff:fe36:fb5f
```

I'm using the ESP Thread Border Router as a test device. With latest stable and master.

zwx1995esp, 2024-06-12:
Hi, @manuelm  We have already fixed this issue on the latest IDF master branch, please pull the latest master branch and try again(For the submodule `components/openthread/lib`, the commit hash [5ae57e1](https://github.com/espressif/esp-thread-lib/tree/5ae57e156e4cd2ccd8dc51e90266b16b284e64de) contains the fix).

manuelm, 2024-06-12:
I can confirm that the updated openthread library fixes the issue:
```
I(166959) OPENTHREAD:[I] RoutingManager: Received NA from router fe80:0:0:0:acbd:dcff:fe36:fb5f
```

Sorry for the noise and thanks for the very fast help.

closed 2024-06-12 by manuelm