espressif/esp-idf #13959: compiling error for esp32s3(Failed to resolve component 'esp32s3')

arunkm490, 2024-06-12:
i get this error .....Failed to resolve component 'esp32s3'
 when i run this command make BOARD=GENERIC_S3

this is for lv_micropython (lvgl binding with micropython)

dobairoland, 2024-06-12:
@arunkm490 I don't think that this is right place for this report. We don't support "make BOARD" nor micropython directly in ESP-IDF. Can you please explain more what you are using and how so I can direct to you the right place?

dobairoland, 2024-06-12:
Ok, I think https://github.com/micropython/micropython/issues might be the right place to report this. 

If I'm not mistaken then `make BOARD=GENERIC_S3` is a build command for micropython.

arunkm490, 2024-06-13:
thank u

ESP-Marius, 2024-08-08:
Closing this since it seems unrelated to esp-idf

closed 2024-08-08 by ESP-Marius