espressif/esp-idf #13961: efuse写入失败

WonderfulMoment, 2024-06-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

目前想通过写efuse位来判断代码应用代码是否启动过，来判断芯片是否跑过程序，目前遇到的问题是，第一次写入的一个字节是能成功写入的，后面再写同一个块的其它字节，就写入失败；

代码如下：
static const esp_efuse_desc_t MODULE_START_CONFIRM[] = {
    {EFUSE_BLK3, 0, 8},
};
const esp_efuse_desc_t* ESP_EFUSE_START_CONFIRM[] = {
    &MODULE_START_CONFIRM[0],
    NULL
};

char start_confirm_data = 0x5d;

// burn efuse
	ret = esp_efuse_read_field_blob(ESP_EFUSE_START_CONFIRM, &read_start_confirm, 8);
	if (ret != ESP_OK){
		LS_FW_LOGI(TAG, "efuse read block3 start %d, len %d, failed ret = %d",
			ESP_EFUSE_START_CONFIRM[0]->bit_start,
			ESP_EFUSE_START_CONFIRM[0]->bit_count, ret);
	}
	if (read_start_confirm == start_confirm_data) {
		LS_FW_LOGI(TAG, "efuse block3 start %d, len %d, is aread 0x%02x, not need to write again",
			ESP_EFUSE_START_CONFIRM[0]->bit_start,
			ESP_EFUSE_START_CONFIRM[0]->bit_count, start_confirm_data);
	} else {
		LS_FW_LOGI(TAG, "efuse block3 start %d, len %d, read 0x%02x, need write data 0x%02x", 
			ESP_EFUSE_START_CONFIRM[0]->bit_start,
			ESP_EFUSE_START_CONFIRM[0]->bit_count, read_start_confirm, start_confirm_data);
		ret = esp_efuse_write_field_blob(ESP_EFUSE_START_CONFIRM, &start_confirm_data, 8);
		if (ret != ESP_OK){
			LS_FW_LOGI(TAG, "efuse write block3 start %d, len %d, failed  ret = %d", 
				ESP_EFUSE_START_CONFIRM[0]->bit_start,
				ESP_EFUSE_START_CONFIRM[0]->bit_count, ret);
		} else {
			LS_FW_LOGI(TAG, "efuse write block3 start %d, len %d, seccess",
				ESP_EFUSE_START_CONFIRM[0]->bit_start,
				ESP_EFUSE_START_CONFIRM[0]->bit_count);
		}
	}

结果如下：
[image]
第一次写入的时候，读出来是没有问题的

之后我再把
static const esp_efuse_desc_t MODULE_START_CONFIRM[] = {
    {EFUSE_BLK3, 0, 8},
};
改成如下几种数据
static const esp_efuse_desc_t MODULE_START_CONFIRM[] = {
    {EFUSE_BLK3, 8, 8},
};
static const esp_efuse_desc_t MODULE_START_CONFIRM[] = {
    {EFUSE_BLK3, 16, 8},
};
static const esp_efuse_desc_t MODULE_START_CONFIRM[] = {
    {EFUSE_BLK3, 32, 8},
};
读出来的数值是0，但是写入失败的，输出日志如下：
输出日志：
1970/01/01 00:00:00.203 [ls bridge/I] efuse block3 start 16, len 8, read 0x00, need write data 0x7f
E (540) efuse: Bits are not empty. Write operation is forbidden.
1970/01/01 00:00:00.217 [ls bridge/I] efuse write block3 start 16, len 8, failed  ret = 5636

我只写了第一个字节，为什么我再写其它字节的时候就写不进去了，是否我哪里操作的有问题？

谢谢答复！

KonstantinKondrashov, 2024-06-13:
Hi @WonderfulMoment!
You get the `Write operation is forbidden.` error which means that an efuse block can not be re-burned again.
BLOCKs 1 - 10 have [RS coding](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/system/efuse.html#supported-coding-scheme) which means that you can burn into a block only once because some coding data are burned along with user data. The coding data are calculated based on the user data. There is no way to burn user data without the coding data (hardware does not allow that).

You need to burn all fields that you want at one time. There is an efuse API that helps to do it, you need to wrap your code with these functions, the example below

```c
esp_efuse_batch_write_begin();

//Put your code here
// burn efuse
ret = esp_efuse_read_field_blob(ESP_EFUSE_START_CONFIRM, &read_start_confirm, 8);
if (ret != ESP_OK){
LS_FW_LOGI(TAG, "efuse read block3 start %d, len %d, failed ret = %d",
ESP_EFUSE_START_CONFIRM[0]->bit_start,
ESP_EFUSE_START_CONFIRM[0]->bit_count, ret);
}
if (read_start_confirm == start_confirm_data) {
LS_FW_LOGI(TAG, "efuse block3 start %d, len %d, is aread 0x%02x, not need to write again",
ESP_EFUSE_START_CONFIRM[0]->bit_start,
ESP_EFUSE_START_CONFIRM[0]->bit_count, start_confirm_data);
} else {
LS_FW_LOGI(TAG, "efuse block3 start %d, len %d, read 0x%02x, need write data 0x%02x",
ESP_EFUSE_START_CONFIRM[0]->bit_start,
ESP_EFUSE_START_CONFIRM[0]->bit_count, read_start_confirm, start_confirm_data);
ret = esp_efuse_write_field_blob(ESP_EFUSE_START_CONFIRM, &start_confirm_data, 8);
if (ret != ESP_OK){
LS_FW_LOGI(TAG, "efuse write block3 start %d, len %d, failed ret = %d",
ESP_EFUSE_START_CONFIRM[0]->bit_start,
ESP_EFUSE_START_CONFIRM[0]->bit_count, ret);
} else {
LS_FW_LOGI(TAG, "efuse write block3 start %d, len %d, seccess",
ESP_EFUSE_START_CONFIRM[0]->bit_start,
ESP_EFUSE_START_CONFIRM[0]->bit_count);
}
}

////

esp_efuse_batch_write_commit();
```

WonderfulMoment, 2024-06-13:
> Hi @WonderfulMoment! You get the `Write operation is forbidden.` error which means that an efuse block can not be re-burned again. BLOCKs 1 - 10 have [RS [...]

thanks for your reply

KonstantinKondrashov, 2024-06-17:
Hi @WonderfulMoment!
Does it work for you? If yes, could you close it if no other questions. Thanks.

closed 2024-06-17 by WonderfulMoment