espressif/esp-idf #13964: Parttool should allow bypassing readonly flag

jtmullen, 2024-06-12:
### Is your feature request related to a problem?

We recently upgraded to a newer IDF and one of the features we want to take advantage of is the read only partitions to protect our factory set data. 

However, we use parttool to set this info in the first place since it provides a nice interface to do so but parttool does not allow writing to readonly partitions. We can call esptool directly to do this instead of using parttool, but parttool provides an easier way to do this since it can address by name. We have multiple devices that get the same data to partitions with the same name, but they are not at the same address across devices.

### Describe the solution you'd like.

I think the default behavior of not writing to read only partitions is good. However there should be an override flag the user can pass to allow writing to the read only partitions. 

### Describe alternatives you've considered.

Use esptool directly. This loses the better interface of parttool

dobairoland, 2024-06-13:
I think you are looking for passing the `--force` argument. Please see https://github.com/espressif/esp-idf/issues/8196#issuecomment-2141397908.

igrr, 2024-06-13:
@dobairoland I think this is a new limitation in parttool introduced in https://github.com/espressif/esp-idf/commit/ab1eb37fe8c5624ed8efb0629747aa4c5d4ae0a8 (cc @adokitkat)

https://github.com/espressif/esp-idf/blob/8760e6d2a7e19913bc40675dd71f374bcd51b0ae/components/partition_table/parttool.py#L161-L167

so we don't even reach esptool.py, hence `--force` esptool argument is not applicable.

KonstantinKondrashov, 2024-06-13:
I can add the `--force` flag for parttool as well to suppress this error.

igrr, 2024-06-13:
Perhaps something less generic, like `--ignore-readonly` or `--write-readonly`?

That is, unless the `--force` argument would also pass `--force` down to esptool, in which case it would mean "bypass all checks" rather than just "bypass readonly partition checks".

closed 2024-06-26 by espressif-bot