espressif/esp-idf #13965: BLE HID device Gamepad button not working on iOS

alkurop, 2024-06-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.2

### Espressif SoC revision.

ESP32 c3

### Operating System used.

macOS

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C3-DevKitM-1

### Power Supply used.

USB

### What is the expected behavior?

When sending gamepad button press, event should be recognised on iOS, in the Switch Input section

### What is the actual behavior?

Not respond to any button button

### Steps to reproduce.

1. compile the ble hid device demo under bludroid, modify the demo function to:
```

esp_hid_ble_gap_adv_init(ESP_HID_APPEARANCE_GAMEPAD, ble_hid_config.device_name);

const unsigned char mediaReportMap[] = {
0x05, 0x01, // Usage Page (Generic Desktop)
0x09, 0x05, // Usage (Game Pad)
0xA1, 0x01, // Collection (Application)
0x85, 0x01, // Report ID (1)
0x05, 0x09, // Usage Page (Button)
0x19, 0x01, // Usage Minimum (Button 1)
0x29, 0x10, // Usage Maximum (Button 16)
0x15, 0x00, // Logical Minimum (0)
0x25, 0x01, // Logical Maximum (1)
0x75, 0x01, // Report Size (1)
0x95, 0x10, // Report Count (16)
0x81, 0x02, // Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0xC0, // End Collection
};

void esp_hidd_send_consumer_value(uint8_t key_cmd, bool key_pressed)
{
uint8_t buffer[1] = {0};
if (key_pressed)
{
buffer[0] = 0x01;
}

esp_err_t res = esp_hidd_dev_input_set(s_ble_hid_param.hid_dev, 0, 0x01, buffer, 1);

return;
}

```

2. flash it to board and connect it from the iOS, wait for it to perform button click; 

### Debug Logs.

```plain
Nothing strange. On Android this sample works as expected.
I (90149) HID_DEV_DEMO: Send button
V (90149) BLE_HIDD: HID CONF[0] status: 0x00, len: 0
```

### More Information.

Not working on MacOS either

esp-zhp, 2024-10-23:
Did your Mac device recognize the ESP32-C3 as a keyboard device, and none of the keys are responding?

alkurop, 2024-11-24:
I was trying to build a gamepad no a keyboard.
Yes it was recognized as a gamepad, but none of the button events were recognized by iOS.
After some investigation it appeared, that iOS will only accept events big corp gamepads. 
Hence this is not an ESP issue, but rather a requirement from iOS.

closed 2024-11-24 by alkurop