espressif/esp-idf #13966: Improving wolfSSL integration with the Espressif ESP-IDF

gojimmypi, 2024-06-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

# Improving wolfSSL integration with the Espressif ESP-IDF

This is an anchor GitHub issue to track the various upcoming pull requests and other issues related to improving the [wolfSSL](https://www.wolfssl.com/) cryptographic library integration with the Espressif ESP-IDF.

The wolfSSL libraries are available for a given project, but the integration with the ESP-IDF core is not as robust as it should be.

There's a companion issue at wolfSSL: https://github.com/wolfSSL/wolfssl/issues/7640

## Features To Do

- [ ] Include wolfSSL as an ESP-IDF component. See preview [here](https://github.com/gojimmypi/esp-idf/tree/wolfssl/components/wolfssl).
- [ ] Full wolfSSL [ESP-TLS](https://github.com/espressif/esp-idf/tree/master/components/esp-tls) integration.
- [ ] Add wolfSSL support to [bootloader_support](https://github.com/espressif/esp-idf/tree/master/components/bootloader_support)
- [ ] Add wolfSSL support to [bt](https://github.com/espressif/esp-idf/tree/master/components/bt)
- [ ] Add wolfSSL support to [espcoredump](https://github.com/espressif/esp-idf/tree/master/components/espcoredump)
- [ ] Add wolfSSL support to [esp_http_client example](https://github.com/espressif/esp-idf/tree/master/examples/protocols/esp_http_client)
- [ ] Add wolfSSL support to [esp_https_server](https://github.com/espressif/esp-idf/tree/master/components/esp_https_server)
- [ ] Add wolfSSL support to [esp_http_server](https://github.com/espressif/esp-idf/tree/master/components/esp_http_server)
- [ ] Add wolfSSL support to [esp_rom](https://github.com/espressif/esp-idf/tree/master/components/esp_rom)
- [ ] Add wolfSSL support to [lwip](https://github.com/espressif/esp-idf/tree/master/components/lwip)
- [ ] Add wolfSSL support to [mqtt](https://github.com/espressif/esp-idf/tree/master/components/mqtt)
- [ ] Add wolfSSL support to [openthread](https://github.com/espressif/esp-idf/tree/master/components/openthread)
- [ ] Add wolfSSL support to [protocomm](https://github.com/espressif/esp-idf/tree/master/components/protocomm)
- [ ] Add wolfSSL support to [wpa_supplicant](https://github.com/espressif/esp-idf/tree/master/components/wpa_supplicant)
- [ ] CI/CD, Testing, Jenkins at wolfSSL

 
## Reasons for choosing wolfSSL instead of mbedTLS

For serious commercial applications needing or users simply needing more capable, flexible, and actively supported libraries developers should choose wolfSSL.

wolfSSL is a TLS library. wolfSSL offers:

- optimal performance
- rapid integration
- hardware crypto support
- support for the most current standards.

wolfSSL is the best tested crypto support, the #1 TLS in IoT and the first embedded TLS 1.3 platform with TPM 2.0, MQTT, FIPS 140 certification, hardware crypto acceleration and secure enclave support. All products are backed by 24/7 support.  

|                   | **wolfSSL**  | **mbed TLS**    |
| ----------------- | ------------ | --------------- |
| **TECHNOLOGY**                                     |
| Copyright         | wolfSSL Inc. | Multiple Owners | 
| Development Team  | Original developers still on project | Based on XySSL/PolarSSL, not maintained by the original developers |
| Portability       | "Portable Out of the Box <br /> Win32/64, Linux, OS X, Solaris, ThreadX, VxWorks, FreeBSD, NetBSD, OpenBSD, embedded Linux, Haiku, OpenWRT, iPhone (iOS), Android, Nintendo Wii and Gamecube through DevKitPro, QNX, MontaVista, OpenCL, NonStop, TRON/ITRON/µITRON, Micrium's µC OS, FreeRTOS, SafeRTOS, Freescale MQX, Nucleus, TinyOS, HP/UX, Keil RTX, TI-RTOS, Integrity OS" | Win32/64, Linux, OS X, Solaris, FreeBSD, NetBSD, OpenBSD, OpenWRT, iPhone (iOS), Xbox, Android, SeggerOS |
| Standards Support | SSLv3 - TLS 1.3, DTLS 1.0,1.2, 1.3 | TLS 1.2/TLS 1.3 and DTLS 1.2 |
| Server Support | YES | YES |
| Performance | Awesome! See our benchmarks page: https://www.wolfssl.com/docs/benchmarks/ | Average |
| Hardware & Assembly Optimizations | - ARM Assembly Optimizations (Aarch32/Aarch64/Arm32/Cortex-M/Neon) <br >- ARMv8 Cryptography Extensions<br >- RISC-V Assembly<br >- STM32 F2/F4/F7/L4/L5/U5/H5/H7 Hardware Crypto<br >- ATECC608B, ST-SAFEA110, SE050, IoT-Safe<br >- Single Precision Math (C and Assembly) | Some ARM optimizations |
| Command Line Utility | YES | NO |
| Certifications | YES (FIPS 140-3, DO-178 DAL-A)  | NO |
| Certificate Revocation Support | CRL, OCSP, OCSP Stapling | CRL |
| Crypto Library Abstraction Layer | YES | NO | 
| SSL Inspection (Sniffer) Support | YES | NO |
| Compression Support | zlib NO |
| OpenSSL Compatibility Layer | YES (Actively updated - over 1,600) | YES (Out of date) |
| Post Quantum Support | Kyber, LMS, XMSS and Dilithum/Falcon| NO |
| Supported Open Source Projects | OpenSSH, Stunnel, WPA Supplicant, lighttpd (lighty), cURL, mongoose, OpenVPN, NGINX and many others | | 
| Quality Assurance Testing | API Tests, Peer Review, Static Analysis, Product Specific Testing, Multiple Compilers, Benchmarks, Wrappers, Hardware Accelerated Testing, Security fuzzers (wolfSSL internal fuzzer, AFL, TLS Fuzzer, libFuzzer), known user configurations, external validation, big/little endian, multiple platforms (Embedded IOT Devices, Windows, Many Linux variants, MacOS, XCODE, Android) | Broken scripts |
| **SUPPORT    DOCUMENTATION    LICENSING**  | 
| Documentation | YES <br />(complete manual, API reference, build instructions, extensions reference, tutorials, source code, benchmarking, examples) | PARTIAL (build instructions, API reference, source code) | 
[... 66 lines cut ...]
- [wolfssl-staging/Arduino-wolfSSL](https://registry.platformio.org/libraries/wolfssl-staging/Arduino-wolfSSL)

See also the [wolfSSL now supported on PlatformIO](https://www.wolfssl.com/wolfssl-now-supported-on-platformio/) blog.

https://github.com/wolfSSL/wolfssl/tree/master/IDE/PlatformIO

## Arduino

See [Getting Started with wolfSSL on Arduino](https://www.wolfssl.com/getting-started-with-wolfssl-on-arduino/) blog.

https://www.arduino.cc/reference/en/libraries/wolfssl/

https://github.com/wolfSSL/wolfssl/tree/master/IDE/ARDUINO

## wolfSSL for the Apple HomeKit on the ESP32

See the https://github.com/AchimPieters/esp32-homekit-demo

https://github.com/AchimPieters/esp32-homekit-demo/pull/3

## Additional wolfSSL updates related to the Espressif environment

See [ESP32 Espressif Improvements - Roadmap Summary #6234](https://github.com/wolfSSL/wolfssl/issues/6234)

Have an idea for other improvements? Feel free to [open a new issue](https://github.com/wolfSSL/wolfssl/issues/new/choose) or send us an email support@wolfssl.com

igrr, 2024-06-13:
> Include wolfSSL as an ESP-IDF component. See preview [here](https://github.com/gojimmypi/esp-idf/tree/wolfssl/components/wolfssl).

Just wanted to mention that it's unlikely that we would accept such a PR due to the fact that wolfSSL is not compatible with IDF's Apache 2.0 license. So we have to distribute wolfSSL via the component registry, instead.

Outside of the scope of this ESP-IDF feature request, you might also want to include support for wolfSSL in https://github.com/project-chip/connectedhomeip, as that would allow using wolfSSL in Matter devices.

igrr, 2024-06-13:
> There would be no wolfSSL source code in the ESP-IDF components directory. I'm thinking the [Kconfig](https://github.com/gojimmypi/esp-idf/blob/834bd5749589ac83d7f48df02c90f778c7fcbc07/components/wolfssl/Kconfig#L283) options would allow something like this:

This (user having to specify paths in their system) is a pattern we are trying to avoid in ESP-IDF, because it results in projects which are not easily portable between different computers. Imagine a team of developers working on a project, and different developers have projects checked out in different locations. Options such as this one would be impossible to commit to a common sdkconfig file in the project repository.

I think the component registry already provides a convenient way to install wolfSSL. The trend we want to see is that in the future, more features are moved out of ESP-IDF into standalone components. I'd like to avoid taking a step in the reverse direction.

> The ESP-IDF environment does not see the local project managed component when compiling the IDF components. For example, I added this #include to [components/esp_wifi/include/esp_wifi.h](https://github.com/espressif/esp-idf/blob/master/components/esp_wifi/include/esp_wifi.h):

This is possible, please see for example how it is done in [esp-tls component](https://github.com/espressif/esp-idf/blob/8760e6d2a7e19913bc40675dd71f374bcd51b0ae/components/esp-tls/CMakeLists.txt#L25-L28). We possibly have to change that line a bit to also consider `wolfssl__wolfssl` component name in addition to the legacy `esp-wolfssl` one, but the idea would be the same.

gojimmypi, 2024-06-13:
> This (user having to specify paths in their system) is a pattern we are trying to avoid in ESP-IDF, because it results in projects which are not easily portable between different computers. 

Yes, I agree. That's a very valid point.

I've also considered an option where the `CMakeLists.txt` would automatically `git clone` a specific version. That too, has pros and cons.

> I think the component registry already provides a convenient way to install wolfSSL.  ...  see for example how it is done in [esp-tls component](https://github.com/espressif/esp-idf/blob/8760e6d2a7e19913bc40675dd71f374bcd51b0ae/components/esp-tls/CMakeLists.txt#L25-L28)

Great! That looks like a promising alternative. I don't see any `idf_component.yml` in that component, nor the parent directory. Is there a way for the end user to adjust the version, or do you envision a component version being fixed to a given ESP-IDF release? 

It might be nice to have the user be able to select from the `esp-wolfssl`, the managed component, and even the custom directory.  

Yes: although I completely agree with your portability concerns, I still find the ability to point to a specific component version source code location to be a useful feature (e.g. pointing to a different GitHub repo directory, allowing changes to be easily checked in, or selected from different branches or releases). I could limit this to wolfSSL examples if you'd prefer not to have this feature in the ESP-IDF.

Thanks very much being receptive to including wolfSSL in the ESP-IDF.

gojimmypi, 2024-06-13:
> The user would add idf_component.yml at the application level (e.g. in main component)

AHA! I didn't understand the part about using `idf_component_get_property` in the ESP-IDF, in particular the `wolfssl__wolfssl`  name of the managed component can be used _instead of_ the `esp-wolfssl` repository name. Thanks for the link to the docs, I had not seen that.

I've confirmed by adding this line:

```c
idf_component_get_property(wolfssl wolfssl__wolfssl COMPONENT_LIB)
```

... to the [esp-tls/CMakeLists.txt](https://github.com/espressif/esp-idf/blob/8760e6d2a7e19913bc40675dd71f374bcd51b0ae/components/esp-tls/CMakeLists.txt#L26) like this:

```
if(CONFIG_ESP_TLS_USING_WOLFSSL)
    idf_component_get_property(wolfssl wolfssl__wolfssl COMPONENT_LIB)
    target_link_libraries(${COMPONENT_LIB} PUBLIC ${wolfssl})
endif()
```
... that indeed the ESP-IDF components can find the local project [wolfssl Managed Component](https://components.espressif.com/components/wolfssl/wolfssl). That's a clever implementation! I like it!! I agree this is a considerably better way of implementing a 3rd party component in the ESP-IDF.

Reminder for future reference: The `idf_component_get_property` needs to be added *after* the [idf_component_register(...)](https://github.com/espressif/esp-idf/blob/8760e6d2a7e19913bc40675dd71f374bcd51b0ae/components/esp_wifi/CMakeLists.txt#L56). 

In my `esp-wifi-h` example [above](https://github.com/espressif/esp-idf/issues/13966#issuecomment-2165364284), I was also able to resolve the compiler error with the same section of code `idf_component_get_property()...` in the respective [CmakeLists.txt](https://github.com/espressif/esp-idf/blob/8760e6d2a7e19913bc40675dd71f374bcd51b0ae/components/esp_wifi/CMakeLists.txt#L65).

There's a bit more work required on my part. In particular the currently published managed component (the [5.7.0 release](https://components.espressif.com/components/wolfssl/wolfssl/versions/5.7.0?language=en)) unfortunately does not include the Kconfig file setting for  `TLS_STACK_WOLFSSL`. Thus even when selecting wolfssl in the `menuconfig`, the setting quietly falls back to _not_ using wolfSSL due to the [Kconfig dependency](https://github.com/espressif/esp-idf/blob/8760e6d2a7e19913bc40675dd71f374bcd51b0ae/components/esp-tls/Kconfig#L12).

> I still don't see a strong need to introduce a Kconfig option for source code path, or handling of this option in CMake.

For the typical end user, you are probably right. But from a component developer's perspective of modification of the component and contributing to the upstream source repository... how would you see that working? The Managed Component is static, disconnected source code, no? This is probably not an important issue for the ESP-IDF, but I personally find it quite appealing while developing the component. In the case of wolfSSL, there are continual, daily changes.

@igrr - Thank you so much for your help. I'm glad I asked in advance! Stay tuned while I put together a PR and publish a fresh wolfssl managed component. 

Best Regards,

Jim

igrr, 2024-06-13:
> But from a component developer's perspective of modification of the component and contributing to the upstream source repository... how would you see that working?

There is an additional property called `override_path` you can add in the manifest when specifying a dependency.

Here is an example: https://github.com/espressif/idf-extra-components/blob/46213e9adb3fd3e09829afdbcb287cd2e5e06ea7/catch2/examples/catch2-console/main/idf_component.yml#L4. This line tells the component manager that if it is able to find a `catch2` directory at the specified location (in this case, 3 directory levels up), it should use that as the component location, instead of downloading the component from the registry.

This approach works in cases when examples are somewhere inside a repository with the managed component, like with this `catch2` component. Developers can work on an example project, while making changes to the component outside of the example project. Nothing gets downloaded, you can simply change the component and recompile the example project.

When the component is uploaded to the registry, the registry automatically removes `override_path` from from all the manifests, so that if somebody downloads the same example/component from the registry, we don't end up looking at the non-existent override_path.

(I have tried to explain this and other things in my last year's talk (https://igrr.github.io/edc23/1, https://www.youtube.com/watch?v=D86gQ4knUnc), perhaps you could go through it again? I think it will answer some of your questions.)

gojimmypi, 2024-06-14:
> There is an additional property called override_path you can add in the manifest when specifying a dependency.

Ah, I see. That's a good idea. I'd certainly rather follow the standards. 

>  other things in my last year's talk

@igrr I had forgotten about the presentation! Thank you for the reminder. It is very helpful. 

btw - It is an awesome presentation, particularly the companion https://igrr.github.io/edc23/1.

Thanks again for all your help & feedback.