espressif/esp-idf #13967: Windows installer doesn't complete

Rob190, 2024-06-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

Universal Online Installer 2.26

### Operating System used.

Windows

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

The installer should complete with 'Finish' or 'Close' etc.

### What is the actual behavior?

'Finishing ESP-IDF' Installation', 'Updating fileMode in submodules' ' Running command: C:\Espressif\tools\idf-git\2.44.0\cmd\git.exe -C C:\Espressif\frameworks\esp-idf-v5.2.2 submodule foreach --recursive git config --local core.fileMode false'.

There is no detectable activity in windows task manager. I waited about 5 minutes before clicking on cancel.
[Setup Log 2024-06-13 #001.txt](https://github.com/user-attachments/files/15817030/Setup.Log.2024-06-13.001.txt)

### Steps to reproduce.

1. Go to: https://dl.espressif.com/dl/esp-idf/?idf=4.4
2. Click Universal Online installer

### Build or installation Logs.

```plain
2024-06-13 10:10:04.503   Log opened. (Time zone: UTC+01:00)
2024-06-13 10:10:04.503   Setup version: Inno Setup version 6.2.2
2024-06-13 10:10:04.503   Original Setup EXE: C:\Users\robert\Downloads\esp-idf-tools-setup-online-2.26.exe
2024-06-13 10:10:04.503   Setup command line: /SL5="$307BE,3328120,844288,C:\Users\robert\Downloads\esp-idf-tools-setup-online-2.26.exe" 
2024-06-13 10:10:04.504   Windows version: 10.0.19045  (NT platform: Yes)
2024-06-13 10:10:04.504   64-bit Windows: Yes
2024-06-13 10:10:04.504   Processor architecture: x64
2024-06-13 10:10:04.504   User privileges: None
2024-06-13 10:10:04.509   Administrative install mode: No
2024-06-13 10:10:04.510   Install mode root key: HKEY_CURRENT_USER
2024-06-13 10:10:04.510   64-bit install mode: Yes
2024-06-13 10:10:06.239   Created temporary directory: C:\Users\robert\AppData\Local\Temp\is-J99DK.tmp
2024-06-13 10:10:06.250   -- DLL function import --
2024-06-13 10:10:06.250   Function name: idpAddFile
2024-06-13 10:10:06.250   DLL name: files:idp.dll
2024-06-13 10:10:06.250   Extracting temporary file: C:\Users\robert\AppData\Local\Temp\is-J99DK.tmp\idp.dll
2024-06-13 10:10:06.261   Dest DLL name: C:\Users\robert\AppData\Local\Temp\is-J99DK.tmp\idp.dll
2024-06-13 10:10:06.261   Importing the DLL function.
2024-06-13 10:10:06.343   Successfully imported the DLL function. Delay loaded? No
2024-06-13 10:10:06.344   -- DLL function import --
2024-06-13 10:10:06.344   Function name: idpAddFileComp
2024-06-13 10:10:06.344   DLL name: files:idp.dll
2024-06-13 10:10:06.344   Dest DLL name: C:\Users\robert\AppData\Local\Temp\is-J99DK.tmp\idp.dll
2024-06-13 10:10:06.344   Importing the DLL function.
2024-06-13 10:10:06.344   Successfully imported the DLL function. Delay loaded? No
2024-06-13 10:10:06.344   -- DLL function import --
2024-06-13 10:10:06.344   Function name: idpAddMirror
2024-06-13 10:10:06.344   DLL name: files:idp.dll
2024-06-13 10:10:06.344   Dest DLL name: C:\Users\robert\AppData\Local\Temp\is-J99DK.tmp\idp.dll
[... 582 lines cut ...]
2024-06-13 10:23:12.763   Running command: C:\Espressif\tools\idf-git\2.44.0\cmd\git.exe -C C:\Espressif\frameworks\esp-idf-v5.2.2 submodule foreach --recursive git config --local core.fileMode false
                          
2024-06-13 10:23:12.763   --------
2024-06-13 10:23:12.763   Message box (OK):
                          Installation has been cancelled.
2024-06-13 10:23:14.774   User chose OK.
2024-06-13 10:23:14.778   Resetting the repository: C:\Espressif\tools\idf-git\2.44.0\cmd\git.exe -C C:\Espressif\frameworks\esp-idf-v5.2.2 reset --hard
2024-06-13 10:23:14.824   Done, exit code=-1
2024-06-13 10:23:14.824   --------
2024-06-13 10:23:14.825   Running command: C:\Espressif\tools\idf-git\2.44.0\cmd\git.exe -C C:\Espressif\frameworks\esp-idf-v5.2.2 reset --hard
                          
2024-06-13 10:23:14.825   --------
2024-06-13 10:23:14.825   Message box (OK):
                          Installation has been cancelled.
2024-06-13 10:23:15.791   User chose OK.
2024-06-13 10:23:15.804   Message box (Yes/No):
                          Installation log has been created, it may contain more information about the problem.
                          Display the installation log now?
2024-06-13 10:23:19.328   User chose Yes.
2024-06-13 10:23:19.474   CurStepChanged raised an exception.
```

### More Information.

No

jakub-kocka, 2024-06-19:
Hello @Rob190,

Thank you for reporting this issue. Unfortunately, we were unable to reproduce the problem on our end. However, your report helped us identify that when the Installer enters a blocking wait state, there is no fallback mechanism to exit this state if something, as in your case, does not work as expected.

A pull request proposing a fix for this issue has been submitted.

In the meantime, you might try rerunning the Installer (if you haven't already). The installation should be successful upon retry. It seems there may have been some issues with library calls during the installation process, which caused the Installer to hang indefinitely due to the bug you discovered.

dobairoland, 2024-07-02:
Closed by https://github.com/espressif/idf-installer/pull/250

closed 2024-07-02 by dobairoland