espressif/esp-idf #13968: Opening an I2C bus more than once

RobMeades, 2024-06-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

We provide a C library (for handling of our GNSS/cellular/Wifi/BLE modules) in which we have a "porting" API that is used on top of all our supported platforms (ESP-IDF, STM32Cube, Zephyr, MCUXpresso, Linux, etc.).  In the I2C case the port API includes a [uPortI2cOpen()](https://github.com/u-blox/ubxlib/blob/ba0a0f17cbd3028dfa2097decccd48458e08b021/port/api/u_port_i2c.h#L108) function, as you might expect, which opens one of our devices on a given I2C HW block, but it also has a [uPortI2cAdopt()](https://github.com/u-blox/ubxlib/blob/ba0a0f17cbd3028dfa2097decccd48458e08b021/port/api/u_port_i2c.h#L122) function, which an application will use if it has _already_ opened that I2C HW block itself, using whatever APIs the platform provides (e.g. because they are talking to other I2C devices on the same bus), and just wants our library to open our device on the same bus without touching anything; [uPortI2cAdopt()](https://github.com/u-blox/ubxlib/blob/ba0a0f17cbd3028dfa2097decccd48458e08b021/port/api/u_port_i2c.h#L122) is the more usual "way in", since most applications will have more than one thing hanging off the I2C bus.

[uPortI2cAdopt()](https://github.com/u-blox/ubxlib/blob/ba0a0f17cbd3028dfa2097decccd48458e08b021/port/api/u_port_i2c.h#L122) has worked fine on all platforms for years now and it worked fine with the "old" ESP-IDF I2C API, however I can't see a way to make it work on the new ESP-IDF I2C API introduced in ESP-IDF v5.2: the ESP-IDF layer now passes back a handle for an opened bus, which we have no access to (only the application does) and [i2c_master_bus_add_device()](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/peripherals/i2c.html#_CPPv425i2c_master_bus_add_device23i2c_master_bus_handle_tPK19i2c_device_config_tP23i2c_master_dev_handle_t), which we need to call to address our device, requires this handle.   The function [i2c_new_master_bus()](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/peripherals/i2c.html#_CPPv418i2c_new_master_busPK23i2c_master_bus_config_tP23i2c_master_bus_handle_t) _looks_ as though it will return an error if we try to open the same I2C HW block again in order to talk to our device.

Any ideas how we can make this work?

mythbuster5, 2024-06-18:
Maybe a function sounds like `i2c_master_get_handle_via_port` will be helpful..?

RobMeades, 2024-06-18:
Ooo, yes indeed, that would be _extremely_ helpful!

closed 2024-07-04 by espressif-bot

RobMeades, 2024-07-04:
Excellent, many thanks @mythbuster5!

RobMeades, 2024-08-28:
Strangely, even [5.4-dev](https://github.com/espressif/esp-idf/commits/v5.4-dev/components/esp_driver_i2c/i2c_master.c) doesn't bring in this commit - are we getting our code the right way by using a release tag?

synologic, 2024-08-28:
@mythbuster5 definitely ubxlib needs this commit so we can port our code to latest IDF

AxelLin, 2024-08-29:
> [uPortI2cAdopt()](https://github.com/u-blox/ubxlib/blob/ba0a0f17cbd3028dfa2097decccd48458e08b021/port/api/u_port_i2c.h#L122) has worked fine on all platforms for years now and it has always worked fine with the "old" ESP-IDF I2C API, however I can't see a way to make it work on the new ESP-IDF I2C API introduced in ESP-IDF v5.2: the ESP-IDF layer now passes back a handle for an opened bus, which we have no access to 

@mythbuster5  This fix should be back-ported to v5.2+.