espressif/esp-idf #13971: PSRAM and SD-Card don't work simultaneously

matti122, 2024-06-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4-dev-826-g8760e6d2a7

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

I expect that I can use PSRAM and also SD-Card on ESP32-S3, but SD-Card can't mounted if PSRAM is activated and initialized at startup.

### What is the actual behavior?

If I activate "Initialize SPI RAM during startup" and want to mount SD-Card connected via 4-Line emmc interface. The SD-Card is not mounted correct and also format of SD-Card is destroying the file System.

### Steps to reproduce.

1. Use emmc SD-Card demo and activate "Initialize SPI RAM during startup"
2. Try to mount SD-Card

### Debug Logs.

```plain
I (6037) gpio: GPIO[17]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (6047) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (6047) gpio: GPIO[2]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (6057) gpio: GPIO[4]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (6077) gpio: GPIO[12]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (6077) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (6447) gpio: GPIO[13]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
W (6457) vfs_fat_sdmmc: failed to mount card (13)
W (6457) vfs_fat_sdmmc: partitioning card
W (6457) vfs_fat_sdmmc: formatting card, allocation unit size=16384
W (8197) vfs_fat_sdmmc: mounting again
I (8197) SD-Card: Filesystem mounted
Name: APPSD
Type: SDHC/SDXC
Speed: 20.00 MHz (limit: 20.00 MHz)
Size: 3840MB
CSD: ver=2, sector_size=512, capacity=7864320 read_bl_len=9
SSR: bus_width=4
```

### More Information.

The issue is available also with 1-line emmc interface.
The issue is available with Quad- and Octa-Channel PSRAM and Flash.
I'm using the following pinning:
#define BSP_SD_HOST_CMD       (GPIO_NUM_14)
#define BSP_SD_HOST_CLK       (GPIO_NUM_17)
#define BSP_SD_HOST_D0        (GPIO_NUM_2)
#define BSP_SD_HOST_D1        (GPIO_NUM_4)
#define BSP_SD_HOST_D2        (GPIO_NUM_12)
#define BSP_SD_HOST_D3        (GPIO_NUM_13)

nopnop2002, 2024-06-15:
There may be a conflict between the GPIO used by PSRAM /CE and PSRAM SCLK and the GPIO used by SD-MMC.
Please check the circuit diagram of your board.

igrr, 2024-06-16:
We have indeed broken the SDMMC driver for ESP32-S3 in case PSRAM is enabled in this commit: https://github.com/espressif/esp-idf/commit/49b4bc175ec24b273d19f371450a60686bd7f83c. The fix has been already merged internally and should be available on GitHub soon.

igrr, 2024-06-17:
Should be fixed with https://github.com/espressif/esp-idf/commit/6ed7e93676f6dfe5933865c9c9e45fe8512230ee.

matti122, 2024-06-17:
@igrr The issue has been fixed. Thanks for your help!

closed 2024-06-17 by matti122