espressif/esp-idf #13972: ESP-IDF 5.2 CMD Errors

rannick1982, 2024-06-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello all, I'm new to all of this, and I might just be an idiot, or completely missed something in the documentation, however, this is what I get when I try to run ESP-IDF 5.2 CMD/Powershell

dobairoland, 2024-06-14:
This looks perfectly normal to me. As the output says, you can now...
```
Go to the project directory and run:

idf.py build
```

rannick1982, 2024-06-14:
Here is the result of 'idf.py build'

C:\Espressif\frameworks\esp-idf-v5.2.2>idf.py build
Executing action: all (aliases: build)
Running cmake in directory C:\Espressif\frameworks\esp-idf-v5.2.2\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=C:\Espressif\python_env\idf5.2_py3.11_env\Scripts\python.exe -DESP_PLATFORM=1 -DCCACHE_ENABLE=1 C:\Espressif\frameworks\esp-idf-v5.2.2"...
-- The C compiler identification is Clang 16.0.1
-- The CXX compiler identification is Clang 16.0.1
-- The ASM compiler identification is Clang with GNU-like command-line
-- Found assembler: C:/Espressif/tools/esp-clang/16.0.1-fe4f10a809/esp-clang/bin/clang.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: C:/Espressif/tools/esp-clang/16.0.1-fe4f10a809/esp-clang/bin/clang.exe
-- Check for working C compiler: C:/Espressif/tools/esp-clang/16.0.1-fe4f10a809/esp-clang/bin/clang.exe - broken
CMake Error at C:/Espressif/tools/cmake/3.24.0/share/cmake-3.24/Modules/CMakeTestCCompiler.cmake:69 (message):
  The C compiler

    "C:/Espressif/tools/esp-clang/16.0.1-fe4f10a809/esp-clang/bin/clang.exe"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: C:/Espressif/frameworks/esp-idf-v5.2.2/build/CMakeFiles/CMakeTmp

    Run Build Command(s):C:/Espressif/tools/ninja/1.11.1/ninja.exe cmTC_fb641 && [1/2] Building C object CMakeFiles/cmTC_fb641.dir/testCCompiler.c.obj
    [2/2] Linking C executable cmTC_fb641.exe
    FAILED: cmTC_fb641.exe
    cmd.exe /C "cd . && C:\Espressif\tools\esp-clang\16.0.1-fe4f10a809\esp-clang\bin\clang.exe   CMakeFiles/cmTC_fb641.dir/testCCompiler.c.obj -o cmTC_fb641.exe -Wl,--out-implib,libcmTC_fb641.dll.a -Wl,--major-image-version,0,--minor-image-version,0  -lkernel32 -luser32 -lgdi32 -lwinspool -lshell32 -lole32 -loleaut32 -luuid -lcomdlg32 -ladvapi32 && cd ."
    lld: error: unable to find library -lkernel32
    lld: error: unable to find library -luser32
    lld: error: unable to find library -lgdi32
    lld: error: unable to find library -lwinspool
    lld: error: unable to find library -lshell32
    lld: error: unable to find library -lole32
    lld: error: unable to find library -loleaut32
    lld: error: unable to find library -luuid
    lld: error: unable to find library -lcomdlg32
    lld: error: unable to find library -ladvapi32
    lld: error: unable to find library -lmingw32
    lld: error: unable to find library -lunwind
    lld: error: unable to find library -lmoldname
    lld: error: unable to find library -lmingwex
    lld: error: unable to find library -lmsvcrt
    lld: error: unable to find library -ladvapi32
    lld: error: unable to find library -lshell32
    lld: error: unable to find library -luser32
    lld: error: unable to find library -lkernel32
    lld: error: unable to find library -lmingw32
    lld: error: too many errors emitted, stopping now
    clang: error: linker command failed with exit code 1 (use -v to see invocation)
    ninja: build stopped: subcommand failed.

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:2 (project)

-- Configuring incomplete, errors occurred!
See also "C:/Espressif/frameworks/esp-idf-v5.2.2/build/CMakeFiles/CMakeOutput.log".
See also "C:/Espressif/frameworks/esp-idf-v5.2.2/build/CMakeFiles/CMakeError.log".
cmake failed with exit code 1, output of the command is in the C:\Espressif\frameworks\esp-idf-v5.2.2\build\log\idf_py_stderr_output_22340 and C:\Espressif\frameworks\esp-idf-v5.2.2\build\log\idf_py_stdout_output_22340

igrr, 2024-06-14:
It looks like you are running `idf.py build` from the root directory of esp-idf, not from a project directory. Could you please try changing to one of the example directories and running `idf.py build` there?

roc138, 2024-06-19:
安装Espressif-IDE后遇到同样的问题，install Espressif-IDE 5.2.2  ，and get same problem

dobairoland, 2024-06-19:
> 安装Espressif-IDE后遇到同样的问题，install Espressif-IDE 5.2.2 ，and get same problem

IDEs are failing because I think they parse the output and fail because they find an error message. It is just a warning and can be ignored. This will be fixed on the IDEs side.

The issue reported here is just a misunderstanding.

dobairoland, 2024-06-24:
I'm closing this for inactivity. Please let me know if this is still an issue for you.

closed 2024-06-24 by dobairoland

denizzzka, 2024-08-16:
Run into same issue

> It is just a warning and can be ignored

This isn't warning by default - build configuation fails

denizzzka, 2024-08-16:
I'm maybe not quite correctly checked, but somehow way it is presented on `v5.3` and `5.4-dev`, but fixed in current `~master` (d7ca8b94c852052e3bc33292287ef4dd62c9eeb1)

denizzzka, 2024-08-16:
CMake update from `3.29.6 to 3.30.2 also helped

denizzzka, 2024-08-16:
My issue was fixed by:
```
$ idf_tools.py install esp-clang
$ Espressif_IDE/esp-idf-v5.3/export.fish
```