espressif/esp-idf #13975: The examples/protocols/mqtt/ssl demo cannot use TLS 1.3 to connect with MQTT broker.

robbinlu-ayla, 2024-06-14:
I enabled TLS 1.3 via menuconfig for examples/protocols/mqtt/ssl demo in IDF 5.2.1 SDK, but the ssl demo still used TLS 1.2 to connect with MQTT broker. I attached 2 packet data screenshot, please help to find what is the issue? how to enable TLS1.3 in IDF 5.2.1 SDK?
[image]
[image]

euripedesrocha, 2024-06-18:
Hi @robbinlu-ayla thanks for reporting. 
The issue here is that the version should be set through the transport layer and mqtt client doesn't offer the option. I'm working on the solution to make it easier for users to set options to the transport in use. 

One possible work around is to use the custom transport in the mqtt client. 
The steps would be:
- create a ssl transport using `esp_transport_ssl_init`
- set all the required options using the esp_transport API
- Pass the transport configured as an option to `esp_mqtt_client_init` through the configuration field `network.transport`.

Other alternative would be to set only TLS 1.3 in menuconfig in the mbedTLS options, disabling TLS 1.2.

robbinlu-ayla, 2024-06-28:
Hi @euripedesrocha,
I tried enable TLS 1.3 and disable TLS 1.2 in menuconfig, then build the ssl example, but 
[ ] Support TLS 1.2 protocol
[*] Support TLS 1.3 protocol

robbinlu-ayla, 2024-07-11:
Hi @euripedesrocha,
Thanks your reply. We didn't use components/mqtt in our project.
We will update and test after get your fix code.

euripedesrocha, 2024-07-12:
@robbinlu-ayla didn't understand your comment about not using components/mqtt. 

About the compilation issue setting only TLS 1.3, below are the settings to disable TLS v1.2:
```
CONFIG_MBEDTLS_SSL_PROTO_TLS1_3=y
CONFIG_MBEDTLS_SSL_RENEGOTIATION=n
CONFIG_MBEDTLS_SSL_PROTO_TLS1_2=n
CONFIG_ESP_WIFI_MBEDTLS_TLS_CLIENT=n
```

robbinlu-ayla, 2024-09-03:
@euripedesrocha 
I set these config items in sdkconfig, rebuilt project, flashed app bin into ESP32 DevKit, the app cannot connected with our cloud server. The TLS1.3 still doesn't work.
CONFIG_MBEDTLS_SSL_PROTO_TLS1_3=y
CONFIG_MBEDTLS_SSL_RENEGOTIATION=n
CONFIG_MBEDTLS_SSL_PROTO_TLS1_2=n
CONFIG_ESP_WIFI_MBEDTLS_TLS_CLIENT=n

Which esp-idf version could support TLS1.3?

euripedesrocha, 2024-09-09:
Hi @robbinlu-ayla, 
using IDF 5.2.1 and the settings I shared you should be able to use TLS 1.3

If you follow the path of creating a transport and passing it to mqtt client after setting the required configuration, it should work. 

If you use the latest master from mqtt client, we still need to move it to IDF and backport to older versions, you will be able to use the code presented [here](https://github.com/espressif/esp-idf/issues/13975#issuecomment-2222350424)

Could you share the errors you are facing, so I can try to help you?

robbinlu-ayla, 2024-09-18:
We are using IDF 5.2.1, using the below config,
CONFIG_MBEDTLS_SSL_PROTO_TLS1_3=y
CONFIG_MBEDTLS_SSL_RENEGOTIATION=n
CONFIG_MBEDTLS_SSL_PROTO_TLS1_2=n
CONFIG_ESP_WIFI_MBEDTLS_TLS_CLIENT=n

And we didn't use esp mqtt client components, we wrote a mqtt client(based components/mbedtls), our mqtt client cannot connect with our cloud with TLS1.3.

euripedesrocha, 2024-09-18:
@robbinlu-ayla I'm a bit confused here, since the issue is regarding one of the examples from idf that uses the mqtt client. 

I need you to share the logs with the errors that you are facing, so I can try to identify potential misconfiguration from your side.

hawkhan, 2024-09-18:
We use socket to connect to the service on the port 443, after the connection is established, it uses mbedtls_ssl_setup to setup the context for the TLS handshaking, and then use mbedtls_ssl_handshake to handshake. According to your mentioned, I think that you want us to specified TLS 1.3 in the SSL context? I have two questions,
1. How to specified TLS version by mbedtls API?
2. If the service only supports TLS 1.2, does it will fall back to TLS 1.2 if I specified TLS 1.3?

euripedesrocha, 2024-09-20:
Hi @hawkhan / @robbinlu-ayla 
For usage of the mbedtls API directly you can refer to [this exmple](https://github.com/espressif/esp-idf/tree/master/examples/protocols/https_mbedtls).
The TLS version in use decision will be part of the TLS negotiation between client and server, so if you have both set and there is only TLS 1.2 in one of the sides that will be selected. 
If there is a decision to use only TLS 1.3, you can use `mbedtls_ssl_conf_min_tls_version` to set it as the minimum.

closed 2025-09-09 by euripedesrocha