espressif/esp-idf #13976: MQTT does not work on WiFi if OpenThread DNS64 client is compiled in.

Marttico, 2024-06-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

## Summary:
When enabling `CONFIG_OPENTHREAD_DNS64_CLIENT` in `sdkconfig`, WiFi fails to connect to the MQTT broker. Disabling this option resolves the WiFi connection issue but causes the Thread connection to the MQTT broker to fail, likely due to the absence of DNS64 functionality.

## Steps to Reproduce:

1. Configure the project to enable `CONFIG_OPENTHREAD_DNS64_CLIENT` in sdkconfig.
2. Attempt to establish an MQTT connection over WiFi.
3. Observe that WiFi fails to connect to the MQTT broker.
4. Disable `CONFIG_OPENTHREAD_DNS64_CLIENT` in sdkconfig.
5. Attempt to establish an MQTT connection over WiFi again.
6. Observe that WiFi successfully connects to the MQTT broker.
7. Attempt to connect to the MQTT broker through Thread.
8. Note the failure to connect, likely due to DNS64 being disabled.

## Expected Behavior:
Both WiFi and Thread should be able to connect to the MQTT broker when `CONFIG_OPENTHREAD_DNS64_CLIENT` is enabled.

## Actual Behavior:

* With `CONFIG_OPENTHREAD_DNS64_CLIENT` enabled:
  * WiFi cannot connect to the MQTT broker 
  * ```
    E (2332) OT_DNS64: Cannot find NAT64 prefix
    E (2342) esp-tls: couldn't get hostname for :x.x.x.x: getaddrinfo() returns 202, addrinfo=0x0
    E (2352) transport_base: Failed to open a new connection: 32769
    E (2352) mqtt_client: Error transport connect 
    ```
  * Thread can connect to the MQTT broker.
* With `CONFIG_OPENTHREAD_DNS64_CLIENT` disabled:
  * WiFi can connect to the MQTT broker.
  * Thread cannot connect to the MQTT broker (presumably due to lack of DNS64).

## Additional Observations:

* Sending UDP and TCP messages over both WiFi and OpenThread works correctly with `CONFIG_OPENTHREAD_DNS64_CLIENT` enabled.
* This suggests that MQTT may be specifically incompatible with WiFi when `CONFIG_OPENTHREAD_DNS64_CLIENT` is enabled.

## System Information:
* ESP-IDF Version: 5.3
* Module: ESP32-C6
* WiFi and Thread configurations are set up to establish MQTT connections.

## Suggested Fix:
Investigate the interaction between MQTT and DNS64 client settings to ensure compatibility across both WiFi and Thread connections.

zwx1995esp, 2024-06-19:
Hi, @Marttico For the thread running on esp32C6, what device role of openthread? Is it an border router or just a normal thread router device? Could you please share the topology of your test env? That might be helpful to us to understand what happens. Thanks.

Marttico, 2024-06-20:
Hey @zwx1995esp,

My setup in step 7 is specifically that my Thread device is a child. It's connected to an OpenThread border router hosted on a Raspberry Pi 4 connected to an nrf52840 as an RCP.

Do note that the issue I'm experiencing is specifically that an MQTT connection does not work while having compiled OpenThread DNS64 into my project but **not** initializing OpenThread. The way I'm expecting to connect to the MQTT broker is through a Wi-Fi connection, which usually works, but not when `CONFIG_OPENTHREAD_DNS64_CLIENT` is enabled in sdkconfig.

Connectivity through OpenThread seems to work fine with the `CONFIG_OPENTHREAD_DNS64_CLIENT` enabled in sdkconfig.

Thanks in advance for any help or suggestions you can offer!

zwx1995esp, 2024-08-07:
Hi, @Marttico Sorry for the delay. So your project is running Thread and Wi-Fi together in one APP and you want both Thread and Wi-Fi can connect MQTT server at same time, right? Or you just want to run them separated and you want to run Wi-Fi or Thread MQTT client with the same configuration?

zwx1995esp, 2024-09-26:
@Marttico Ok, for supporting OT device DNS feature, we add a hook to `lwip_hook_netconn_external_resolve` to send the resolve and get the result manually. When using wifi, this hook should not be processed. So you can define a global flag to
distinguish when wifi or thread are active. Return `0` in the function `lwip_hook_netconn_external_resolve`  when using wifi, and process this logic when using thread. Please have a try.

zwx1995esp, 2024-09-26:
FYI: [lwip_hook_netconn_external_resolve](https://github.com/espressif/esp-idf/blob/v5.3.1/components/openthread/src/esp_openthread_dns64.c#L58) and `CONFIG_OPENTHREAD_DNS64_CLIENT` should be enabled.

Marttico, 2024-09-30:
@zwx1995esp When replacing the logic in esp_openthread_dns64.c to just return 0, it seems to work with WiFi again! I haven't added any global flags yet though. This may be a little off topic for this issue, but would the best course of action currently just be to modify the OpenThread component and use a custom component for this?

```c++
int lwip_hook_netconn_external_resolve(const char *name, ip_addr_t *addr, u8_t addrtype, err_t *err)
{
    return 0;
}
```

Marttico, 2024-11-27:
An update on this issue, it seems like the current version of ESP-IDF v5.3 fixed this issue. Wi-Fi will now still work when Thread is compiled together.

Marttico, 2024-11-29:
I tested my prototype today, and it seems broken again.

```
I (3129) connected-service: Binding MQTT WiFi functions
E (3129) OT_DNS64: Cannot find NAT64 prefix
E (3139) esp-tls: couldn't get hostname for :x.x.x.x: getaddrinfo() returns 202, addrinfo=0x0
E (3149) transport_base: Failed to open a new connection: 32769
E (3149) mqtt_client: Error transport connect
I (3159) mqtt-service: MQTT_EVENT_DISCONNECTED
```

So for now I'm going to fix it using the fix above.

theijattomoto, 2025-02-17:
Hi Marttico, could you share how you solved the problem? I built the exact same thing as you and encountered the same error.

This is when i disable the openthread:

I (9458) main_task: Returned from app_main()
I (14458) ot_esp_cli: Starting Advanced OTA example
I (14458) ot_esp_cli: OTA started
E (14458) OPENTHREAD: esp_openthread_task_switching_lock_acquire(38): Failed to acquire the lock because the mutex is not ready
E (14458) OPENTHREAD: esp_openthread_task_switching_lock_release(46): Failed to release the lock because the mutex is not ready
E (14458) OT_DNS64: Cannot find NAT64 prefix
E (14458) esp-tls: couldn't get hostname for :192.168.207.114: getaddrinfo() returns 202, addrinfo=0x0
E (14458) transport_base: Failed to open a new connection: 32769
E (14458) HTTP_CLIENT: Connection failed, sock < 0
E (14458) esp_https_ota: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT
E (14458) esp_https_ota: Failed to establish HTTP connection
E (14458) ot_esp_cli: ESP HTTPS OTA Begin failed

and this is when i enable the openthread + wifi:

I (14675) ot_esp_cli: Starting Advanced OTA example
I (14675) ot_esp_cli: OTA started
E (19685) esp-tls: [sock=54] select() timeout
E (19685) transport_base: Failed to open a new connection: 32774
E (19685) HTTP_CLIENT: Connection failed, sock < 0
E (19685) esp_https_ota: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT
E (19685) esp_https_ota: Failed to establish HTTP connection
E (19685) ot_esp_cli: ESP HTTPS OTA Begin failed

adam-hotait, 2025-02-20:
Solved by #15069 and backported to v5.1

To use it while waiting for an official release:
```
cd $IDF_PATH
git fetch
git checkout release/v5.x
git submodule update --init --recursive
```

[See versionning documentation](https://docs.espressif.com/projects/esp-idf/en/v5.4/esp32/versions.html#updating-to-stable-release)