espressif/esp-idf #13977: ESP32 slave role Multiple SCO connection problem

AbdelRahman3li, 2024-06-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

**Describe the bug**

I have an ESP32 board and want to make it a Bluetooth relay between a phone and a headphone the problem occurs with phones that force the ESP32 to be in a slave role the phone that doesn't require role change works correctly forces the ESP32 to make only one SCO connection with the phone and when I make a second SCO connection I got error code 0x0A which is from bluetooth.h corresponds to (ERROR_CODE_SYNCHRONOUS_CONNECTION_LIMIT_TO_A_DEVICE_EXCEEDED)

HINT: the ESP32 is in a slave role with the phone but master with the headphone but when it is a slave role with the phone I get an error 
          code 0x0A with the headphone despite ESP32 being a master with the headphone.

**To Reproduce**

Steps to reproduce the behavior:
1. I have combined the HFP AG and UNIT examples from the btstack examples.
2. make an ACL connection to the headphone and the phone
3. try to initiate a call an SCO connection is successfully created with the phone while with the headphone error code 0x0A

**Expected behavior**

The ESP32 should route audio back and forth between the phone and headphone without any problem as it do when it is a master role

**HCI Packet Logs**
Unfortunately, I didn't take packet logs as the problem relies only on the ESP32 role.

**Environment:  (please complete the following information):**
 - Current BTstack branch: latest release.
 - esp-idf version: 5.1.4
 - Bluetooth Controller : ESP32.
 - Remote device: Samsung Galaxy A22 and an old headphone as I am faced with newer headphones and later I will raise an 
                             issue with this.

**Additional context**
To Avoid opening another Issue I also face a problem with connection to various headphones such as DL19 and Sound Core life Q30 if you can look at this problem also it will be good but I want to fix the problem of multiple SCO connection as this is the main problem.

BetterJincheng, 2024-06-17:
@AbdelRahman3li ESP32 now only support one SCO connection. Maybe you can try with multiple eSCO connections.

AbdelRahman3li, 2024-06-17:
> @AbdelRahman3li ESP32 now only support one SCO connection. Maybe you can try with multiple eSCO connections.

Hi [BetterJincheng](https://github.com/BetterJincheng),

Thanks for your reply, can you explain more I am setting CONFIG_BTDM_CTRL_BR_EDR_MAX_SYNC_CONN=3 and the problem happens only when the phone forces the ESP32 to be in a slave role.

AbdelRahman3li, 2024-06-19:
> @AbdelRahman3li We limited the number of (e)SCO connection in slave role to 1 in the Bluetooth Controller. `CONFIG_BTDM_CTRL_BR_EDR_MAX_SYNC_CONN=3` can support just one slave connection and no more than 2 master conncetions or just three master connections.

@BetterJincheng Thank you for your clarification, is there any way to increase the number of (e)SCO connection in slave role to 2?

BetterJincheng, 2024-06-20:
@AbdelRahman3li Sorry, I'm afraid not.

AbdelRahman3li, 2024-06-20:
@BetterJincheng Thank you very much, regarding the second Issue of the ESP32 is unable to connect to modern headphones like Sound Core life Q30 despite I am able to connect with them with my phone but with ESP32 using the HFP_AG example of btstack or the esp-idf examples even is a2dp_source I get an error code 0x04 as if the headphone is not exist.

BetterJincheng, 2024-06-21:
@AbdelRahman3li Error code 0x4 is `page timeout` that means failures occured in the process of device connection. The whole console log will do help. Please attach the log. Thanks

AbdelRahman3li, 2024-06-21:
@BetterJincheng The console log for when trying to connect to the DL19 Headset using HFP_AG example of esp-idf 5.1.4 is as follows my ESP32 Chip is ESP32-D0WD-V3 (revision v3.0)

```
hfp_ag> con
Connect.
hfp_ag> W (70527) BT_SDP: SDP - Rcvd conn cnf with error: 0x4  CID 0x40

W (70537) BT_HCI: hcif conn complete: hdl 0xfff, st 0x4
I (70537) BT_APP_HF: APP HFP event: CONNECTION_STATE_EVT
I (70547) BT_APP_HF: --connection state DISCONNECTED, peer feats 0x0, chld_feats 0x0
hfp_ag> con
Connect.
hfp_ag> W (138497) BT_SDP: SDP - Rcvd conn cnf with error: 0x4  CID 0x41

W (138497) BT_HCI: hcif conn complete: hdl 0xfff, st 0x4
I (138507) BT_APP_HF: APP HFP event: CONNECTION_STATE_EVT
I (138517) BT_APP_HF: --connection state DISCONNECTED, peer feats 0x0, chld_feats 0x0
```

AbdelRahman3li, 2024-06-21:
@BetterJincheng and this is the HCI logs when I attempt to connect to the DL19 headset with the HFP_AG example

```
[02:21:33.097] LOG -- hfp_ag_demo.c.198: USER:'a'
Establish HFP service level connection to 76:09:E4:A5:64:A1...
[02:21:33.100] LOG -- hfp.c.665: Create HFP context 0x3ffdeabc: role 1, addr 76:09:E4:A5:64:A1
[02:21:33.102] LOG -- l2cap.c.2544: create channel addr 76:09:E4:A5:64:A1 psm 0x1 mtu 1691 -> local mtu 1691, sec level 0
[02:21:33.109] LOG -- l2cap.c.2509: create channel 0x3ffdedd0, local_cid 0x0045
[02:21:33.119] LOG -- hci.c.7694: Create_connection to 76:09:E4:A5:64:A1
[02:21:33.128] LOG -- hci.c.326: create_connection_for_addr 76:09:E4:A5:64:A1, type fd
[02:21:33.129] LOG -- hci.c.7713: conn state 0
[02:21:33.138] CMD => 05 04 0D A1 64 A5 E4 09 76 18 CC 00 00 00 00 01 
[02:21:33.142] EVT <= 0F 04 00 05 05 04 
[02:21:48.504] EVT <= 03 0B 04 FF FF A1 64 A5 E4 09 76 01 00 
[02:21:48.506] LOG -- hci.c.3818: Connection_complete (status=4) 76:09:E4:A5:64:A1
[02:21:48.507] LOG -- hci.c.2632: Outgoing connection to 76:09:E4:A5:64:A1 failed
[02:21:48.512] LOG -- l2cap.c.1141: L2CAP_EVENT_CHANNEL_OPENED status 0x4 addr 76:09:E4:A5:64:A1 handle 0xffff psm 0x1 local_cid 0x45 remote_cid 0x0 local_mtu 1691, remote_mtu 672, flush_timeout 0
[02:21:48.523] LOG -- l2cap.c.1158: ERTM mode 0, fcs enabled 0
[02:21:48.533] LOG -- sdp_client.c.507: SDP Client Connection failed, status 0x04.
Connection failed, status 0x04
[02:21:48.543] LOG -- hfp.c.815: rfcomm service not found, status 0x04
[02:21:48.544] LOG -- l2cap.c.2515: free channel 0x3ffdedd0, local_cid 0x0045
[02:21:48.553] LOG -- l2cap.c.1419: l2cap_stop_rtx for local cid 0x45
[02:21:48.563] LOG -- l2cap.c.363: Stop Retransmission timer
[02:21:48.564] LOG -- l2cap.c.349: Stop Monitor timer
```

and this is when I used gap_inquiry example it rarely discover the device once every 20 attempts

```
Start scanning...
[02:47:43.883] CMD => 01 04 05 33 8B 9E 05 00 
[02:47:43.887] EVT <= 0F 04 00 05 01 04 
[02:47:50.288] EVT <= 01 01 00 
Inquiry scan complete.
Start scanning...
[02:47:51.672] CMD => 01 04 05 33 8B 9E 05 00 
[02:47:51.675] EVT <= 0F 04 00 05 01 04 
[02:47:58.077] EVT <= 01 01 00 
Inquiry scan complete.
Start scanning...
[02:48:14.748] CMD => 01 04 05 33 8B 9E 05 00 
[02:48:14.751] EVT <= 0F 04 00 05 01 04 
[02:48:21.140] EVT <= 01 01 00 
Inquiry scan complete.
Start scanning...
[02:48:23.120] CMD => 01 04 05 33 8B 9E 05 00 
[02:48:23.124] EVT <= 0F 04 00 05 01 04 
[02:48:29.526] EVT <= 01 01 00 
Inquiry scan complete.
Start scanning...
[02:48:30.785] CMD => 01 04 05 33 8B 9E 05 00 
[02:48:30.789] EVT <= 0F 04 00 05 01 04 
[02:48:37.191] EVT <= 01 01 00 
Inquiry scan complete.
Start scanning...
[02:48:40.339] CMD => 01 04 05 33 8B 9E 05 00 
[02:48:40.343] EVT <= 0F 04 00 05 01 04 
[02:48:40.855] EVT <= 02 0F 01 A1 64 A5 E4 09 76 01 00 00 04 04 24 79 57 
Device found: 76:09:E4:A5:64:A1 with COD: 0x240404, 
[02:48:47.745] EVT <= 01 01 00 
Inquiry scan complete.
```

BetterJincheng, 2024-06-24:
@AbdelRahman3li Thanks for the log files. `W (70527) BT_SDP: SDP - Rcvd conn cnf with error: 0x4  CID 0x40` means nothing but failure to find the target device. The example of inquiry can also support this point. I think you need to make sure that the peer device is in the state of "page scan" or ""inquiry scan".
There doesn't seem to be anything wrong with the logs. The gap_inquiry example can find other devices except DL19. It's hard to say that the problem is on our side.

AbdelRahman3li, 2024-06-25:
@BetterJincheng Thanks for the reply, But I can connect to the DL19 or SoundCore Life Q30 with my phones so why ESP32 couldn't I tried 2 ESP32 boards and both have the same behavior I thought of maybe I need to make the headset in a state to accept the paging request before sending a page request maybe the phone does this before sending the connection request.
and for The gap_inquiry example the last one it detected the DL19 with BD ADDR 76:09:E4:A5:64:A1 but this is out of 20 attempts and also when I tried to connect I got 0x04 error code.

BetterJincheng, 2024-06-26:
@AbdelRahman3li You need to make the headset in a state of "inquiry scan"  so ESP32 can find the device using gap_inquiry example. That is to say, you need to make the headset discoverable. Maybe you need to disconnect all other links to the headset.

AbdelRahman3li, 2024-07-04:
@BetterJincheng Thank you very much I will try your suggestion to pair the ESP32 with the Soundcore Life Q30 headset you can consider this Issue as solved.

closed 2024-07-08 by Alvin1Zhang