espressif/esp-idf #13978: Packet Error Rate / Bit Error Rate API

ftab, 2024-06-14:
### Describe the solution you'd like.

When conducting EMC testing, the test facility has asked us to provide Packet Error Rate statistics. This is in reference to EN 301-489-17 section 6.2.2, stating that the unit must measure Packet Error Rate of 10% or less during operation.

### Describe alternatives you've considered.

At this time, since I cannot find an API to retrieve this information (and incidentally, the current project has no user interface with which to convey the information anyway) we have to accept a test deviation.

### Additional context.

Currently using Bluetooth Classic on ESP32 with a slightly modified ESP-IDF 5.2.1

boblane1, 2024-06-18:
Hi @ftab,

I find the following description in EN 301-489-17 section 6.2.2:
> For equipment that does not support a PER or a FER, the minimum performance level shall be a maximum reduction in
the data rate of 10% during the test.

So, does EMC testing have to rely on Packet Error Rate / Bit Error Rate API?

ftab, 2024-06-18:
> Hi @ftab,
>
>
>
> I find the following description in EN 301-489-17 section 6.2.2:
>
>
> the data rate of 10% during the test.
>
>
>
> So, does EMC testing have to rely on Packet Error Rate / Bit Error Rate API?

The testing facility does not seem to believe us when we say there is no way to retrieve the packet error rate, as they asked several times and said it's a known chip so it "should" be possible. (We additionally have no user interface on this project with which to present such information anyway. As we mentioned to them. Multiple times.)

Apparently, since some Bluetooth chips can do it, so can everyone else?

If it is harder than it sounds, that is satisfactory enough for me to justify not reporting packet error rate.

boblane1, 2024-06-19:
In my opinion, the Packet Error Rate / Bit Error Rate shall be computed in upper layer after EMC test. BTW, Testing related HCI commands may be used for EMC test. If I misunderstand the procedure, please describe the detailed test procedure and functional requirements of the Packet Error Rate / Bit Error Rate API.

Alvin1Zhang, 2025-04-03:
@ftab Thanks for reporting, would you please help share if any further updates? Thanks.

ftab, 2025-04-03:
> @ftab Thanks for reporting, would you please help share if any further updates? Thanks.

@Alvin1Zhang no further updates. We accepted a test deviation as the testing facility understood we do not have access to an API for this information.

Should it become available in the future, it would be good. Our hardware engineers refused to believe we needed to have a cutout area below the antenna, so I feel we have a suboptimal design, I just could not prove it with the data I was able to gather with the current SDK.