espressif/esp-idf #13984: static inline void spi_ll_read_buffer(...) muzz support 9-bytes I/O sessions

powerbroker, 2024-06-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.2.1-dirty

### Espressif SoC revision.

ESP32 chip revision: v3.1

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32 WROOM

### Power Supply used.

USB

### What is the expected behavior?

spi_device_polling_transmit(handle, &t) populates ALL t.length (== 72) bits of t.rx_buffer with received data, not only first 64 bits.

### What is the actual behavior?

```
t.length = 8 * 9;
t.tx_buffer = (unsigned char*){ 0xF7 };
t.rxLength = 8 * 8;
t.rx_buffer = (unsigned char*){ 1, 2, 3, 4, 5, 6, 7, 8, 9 };

spi_device_polling_transmit(spiDevice, &t);
```
SPI I/O flows successfully and results in following activity on CLK / MISO lines:
[image]

note 72 total CLK pulses and '00 0x80 00 00 0x80 00 00 0x80 00' stream on MISO line. so, exactly MISO data is expected in t.rx_buffer after all.
but it has '00 0x80 00 00 0x80 00 00 0x80 9' instead - last byte is dropped.

hardware activity indicates: all 9 bytes passed through SPI.

there is a function in spi_ll.h, responsible for transferring received data from hardware FIFO to user t.rx_buffer
```
/**
 * Read from SPI buffer.
 *
 * @param hw Beginning address of the peripheral registers.
 * @param buffer_to_rcv Address to copy buffer data to.
 * @param bitlen Length to copy, in bits.
 */
static inline void spi_ll_read_buffer(spi_dev_t *hw, uint8_t *buffer_to_rcv, size_t bitlen)
{
    for (size_t x = 0; x < bitlen; x += 32) {
        //Do a memcpy to get around possible alignment issues in rx_buffer
        uint32_t word = hw->data_buf[x / 32];
        int len = bitlen - x;
        if (len > 32) {
            len = 32;
        }
        memcpy(&buffer_to_rcv[x / 8], &word, (len + 7) / 8);
    }
}
```
this logic definitely cannot handle properly cases like 9-bytes(5-, 13-, etc.) I/O since for(...) stops at nearest lower than 'bitlen' multiple of 32 value and ignores e.g. bits 65-72 and so on.

### Steps to reproduce.

Implement described logic with non-multiple of 32-bits(of 4-bytes) total I/O length.
enjoy only multiple-of-32 bits(multiple-of-4 bytes) of received data is transferred into t.rx_buffer

wanckl, 2024-06-17:
@powerbroker Hello, As you see, it not a bug, it driver design as this,,,   

- If you only set `t.length = 8 * 9;`, tx and rx will be same length    
- If you set `t.length = 8 * 9;` and `t.rxLength = 8 * 8;`, it will trans 8 * 9 in hardware but write only 8 * 8 of rx buffer
- `t.rxLength > t.length` is not allowed

powerbroker, 2024-06-17:
> @powerbroker Hello, As you see, it not a bug, it driver design as this,,,

could you please explain it a little more?
i need to transmit 1-byte command to SPI device and read *some* bytes after that - 1, 2, 3, sometimes 8, 23, 252, 1013, etc.

it's okay to have 1-st byte of t.rx_buffer blank, since SPI device starts responding on second one. so, how `t.length` and `t.rxLength` should be set up in case of(assuming Tx buffer is populated and Rx one - properly aligned and have enough room):
1 byte Tx + 1 byte Rx (i think `.length = 2*8`, `.rxLength = 1*8`)
1 byte Tx + 3 bytes Rx (i think `.length = 4*8`, `.rxLength = 3*8`)
1 byte Tx + 4 bytes Rx (i think `.length = 5*8`, `.rxLength = 4*8`)
1 byte Tx + 6 bytes Rx (i think `.length = 7*8`, `.rxLength = 6*8`)
1 byte Tx + 12 bytes Rx (i think `.length = 13*8`, `.rxLength = 12*8`)

thanks!

wanckl, 2024-06-18:
@powerbroker  A bit different,    

You can understand as rx buffer is align to start, so if rx length one smaller then tx length, it still not you want.    

you can consider configure devices as `devcfg.command_bits = 8` then using `spi_transaction_t::cmd` to send you command, this length don't belong to t.length, then simply using `t.length` as your rx length.

powerbroker, 2024-06-20:
> @powerbroker A bit different,
> 
> You can understand as rx buffer is align to start, so if rx length one smaller then tx length, it still not you want.

i'm aware of full-duplex SPI does Tx and Rx at the same time, so Rx buffer is of `t.length` size, not `t.rx_length`

moreover, e.g. 1 byte Tx + 6 bytes Rx (t.length = 7*8, t.rx_length = 6*8) and many cases like that work exactly the way i expect basing on MCU manual: driver populates `t.length` (in bytes) of `t.rx_buffer` with all the data appearing on MISO line.

well, in case of only 1-byte command required let us give it a chance to to show it's best in the way you described: with `devcfg.command_bits = 8` and `spi_transaction_t::cmd`

wanckl, 2024-07-08:
@powerbroker Hello, If no problem, please close this issue, thanks

powerbroker, 2024-07-09:
> @powerbroker Hello, If no problem, please close this issue, thanks

unfotunately, ESP32 hardware behavior is still the same and workaround is required

Maksons, 2024-07-31:
You can manual control CS and do multiple SPI transactions to form exactly what you need.