espressif/esp-idf #13985: BLE 128Bit UUID's not reflecting only service and not in characteristics

meakashrao, 2024-06-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.1.2-666-ga5e3e0b90e

### Espressif SoC revision.

ESP32-WROOM-32E-N8 (revision v3.0 and V3.1)

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP-WROOM-32

### Power Supply used.

USB

### What is the expected behavior?

All BLE Services and Characteristics to have 128 bit UUID's 

### What is the actual behavior?

Advertisement starts correctly and the services and characteristics are discoverable but only the service has the custom 128 bit UUID and all other 9 characteristics have 16 bit UUID

[image]

### Steps to reproduce.

`uint8_t delishup_service_uuid[16] = {0x9E, 0xCA, 0xDC, 0x24, 0x0E, 0xE5, 0xA9, 0xE0, 0x93, 0xF3, 0xA3, 0xB5, 0x01, 0x00, 0x40, 0x6E};

static esp_ble_adv_data_t adv_data = {
    .set_scan_rsp        = false,
    .include_name        = true,
    .include_txpower     = true,
    .min_interval        = 0x0006, //slave connection min interval, Time = min_interval * 1.25 msec
    .max_interval        = 0x0010, //slave connection max interval, Time = max_interval * 1.25 msec
    .appearance          = 0x00,
    .manufacturer_len    = 0,    //TEST_MANUFACTURER_DATA_LEN,
    .p_manufacturer_data = NULL, //test_manufacturer,
    .service_data_len    = 0,
    .p_service_data      = NULL,
    .service_uuid_len    = sizeof(delishup_service_uuid),
    .p_service_uuid      = delishup_service_uuid,
    .flag = (ESP_BLE_ADV_FLAG_GEN_DISC | ESP_BLE_ADV_FLAG_BREDR_NOT_SPT),
};

// scan response data
static esp_ble_adv_data_t scan_rsp_data = {
    .set_scan_rsp        = true,
    .include_name        = true,
    .include_txpower     = true,
    .min_interval        = 0x0006,
[... 406 lines cut ...]
I (8807) ESP_BLE: update connection params status = 0, min_int = 0, max_int = 0,conn_int = 6,latency = 0, timeout = 500
I (8997) ESP_BLE: GAP_EVT, event 20

I (8997) ESP_BLE: update connection params status = 0, min_int = 0, max_int = 0,conn_int = 36,latency = 0, timeout = 500
I (16588) ESP_BLE: EVT 15, gatts if 3

I (16589) ESP_BLE: gatts_delishup_profile_event_handler - event: 15
I (16604) ESP_BLE: GAP_EVT, event 6

I (16604) ESP_BLE: advertising start successfully
I (18844) ESP_BLE: EVT 14, gatts if 3

I (18845) ESP_BLE: gatts_delishup_profile_event_handler - event: 14
I (19557) ESP_BLE: GAP_EVT, event 20

I (19558) ESP_BLE: update connection params status = 0, min_int = 0, max_int = 0,conn_int = 6,latency = 0, timeout = 500
I (19751) ESP_BLE: GAP_EVT, event 20

I (19752) ESP_BLE: update connection params status = 0, min_int = 0, max_int = 0,conn_int = 36,latency = 0, timeout = 500
I (26218) ESP_BLE: EVT 15, gatts if 3

I (26219) ESP_BLE: gatts_delishup_profile_event_handler - event: 15
I (26233) ESP_BLE: GAP_EVT, event 6

I (26234) ESP_BLE: advertising start successfully

meakashrao, 2024-06-15:
I made some more changes 

static const uint8_t primary_service_uuid[2] = {ESP_GATT_UUID_PRI_SERVICE & 0xFF, (ESP_GATT_UUID_PRI_SERVICE >> 8) & 0xFF};
static const uint8_t character_declaration_uuid[2] = {ESP_GATT_UUID_CHAR_DECLARE & 0xFF, (ESP_GATT_UUID_CHAR_DECLARE >> 8) & 0xFF};
static const uint8_t character_client_config_uuid[2] = {ESP_GATT_UUID_CHAR_CLIENT_CONFIG & 0xFF, (ESP_GATT_UUID_CHAR_CLIENT_CONFIG >> 8) & 0xFF};

#define CHAR_DECLARATION_SIZE (sizeof(uint16_t))

also based on logs all 27 handles are successfully created 

I (3655) BTDM_INIT: BT controller compile version [a38dc5c]
I (3657) BTDM_INIT: Bluetooth MAC: c0:49:ef:a9:85:f2
I (3660) phy_init: phy_version 4791,2c4672b,Dec 20 2023,16:06:06
I (3981) ESP_BLE: esp_ble_init init bluetooth

I (4058) ESP_BLE: EVT 0, gatts if 3

I (4059) ESP_BLE: gatts_delishup_profile_event_handler - event: 0
I (4061) ESP_BLE: EVT 22, gatts if 3

I (4066) ESP_BLE: gatts_delishup_profile_event_handler - event: 22
I (4070) ESP_BLE: creating attribute table, the number handle = 27

I (4077) ESP_BLE: service started successfully
I (4082) ESP_BLE: GAP_EVT, event 0

I (4086) ESP_BLE: GAP_EVT, event 1

I (4092) main_task: Returned from app_main()
I (4105) ESP_BLE: GAP_EVT, event 6

I (4106) ESP_BLE: advertising start successfully

Please if someone could point where am I making the error

meakashrao, 2024-06-16:
https://github.com/espressif/esp-idf/issues/11916

Says a patch file needs to run 0001-fix-bt-bluedroid-Fix-uuid-length-error-when-uuid-is-.patch

when I run it 

C:\Users\Akash\esp\v5.1\esp-idf>git apply 0001-fix-bt-bluedroid-Fix-uuid-length-error-when-uuid-is-.patch
error: patch failed: components/bt/host/bluedroid/stack/gatt/gatt_utils.c:1229
error: components/bt/host/bluedroid/stack/gatt/gatt_utils.c: patch does not apply

Please help what is to be done next

meakashrao, 2024-06-16:
[image]

https://github.com/espressif/esp-idf/issues/13673

Even the recent thing is failing for me, am I doing something wrong?

meakashrao, 2024-06-17:
[image]

I updated to the stable release of 5.2.2 and still this is a trouble? I am applying the patch file to the location where my ESP Downloaded file is located, do I need to apply the patch file somewhere else?

zhp0406, 2024-06-27:
I will give you a patch base on ESP-IDF v5.1.2-666-ga5e3e0b90e to display usage of 128 bit uuid

meakashrao, 2024-06-27:
I am able to use 128 bit UUID's if I use this table definition, am I right? or i still need to use the patch?

static const esp_gatts_attr_db_t delishup_gatt_db[DELISHUP_IDX_NB] =
    {
        // Service Declaration
        [DELISHUP_SVC_IDX] =
            {{ESP_GATT_AUTO_RSP}, {ESP_UUID_LEN_16, (uint8_t *)&primary_service_uuid, ESP_GATT_PERM_READ, sizeof(DELISHUP_SERVICE_UUID), sizeof(DELISHUP_SERVICE_UUID), 
            (uint8_t *)&DELISHUP_SERVICE_UUID}},

        /* HEARTBEAT Characteristic Declaration */
        [HEARTBEAT_CHAR_IDX] =
            {{ESP_GATT_AUTO_RSP}, {ESP_UUID_LEN_16, (uint8_t *)&character_declaration_uuid, ESP_GATT_PERM_READ, CHAR_DECLARATION_SIZE, CHAR_DECLARATION_SIZE, (uint8_t *)&char_prop_notify}},

        /* HEARTBEAT Characteristic Value */
        [HEARTBEAT_CHAR_VAL_IDX] =
            {{ESP_GATT_AUTO_RSP}, {ESP_UUID_LEN_128, (uint8_t *)&HEARTBEAT_UUID, ESP_GATT_PERM_READ, sizeof(heartbeat_val), sizeof(heartbeat_val), (uint8_t *)heartbeat_val}},

        //HEARTBEAT data notify characteristic Declaration
        [HEARTBEAT_NTF_CFG] =
            {{ESP_GATT_AUTO_RSP}, {ESP_UUID_LEN_16, (uint8_t *)&character_client_config_uuid, ESP_GATT_PERM_READ | ESP_GATT_PERM_WRITE, sizeof(heartbeat_val_ccc), sizeof(heartbeat_val_ccc), (uint8_t *)heartbeat_val_ccc}},
        
        // data receive characteristic Declaration
        [WEIGHT_CHAR_IDX] =
            {{ESP_GATT_AUTO_RSP}, {ESP_UUID_LEN_16, (uint8_t *)&character_declaration_uuid, ESP_GATT_PERM_READ, CHAR_DECLARATION_SIZE, CHAR_DECLARATION_SIZE, (uint8_t *)&char_prop_notify}},

        // data receive characteristic Value
        [WEIGHT_CHAR_VAL_IDX] =
            {{ESP_GATT_AUTO_RSP}, {ESP_UUID_LEN_128, (uint8_t *)&WEIGHT_UUID, ESP_GATT_PERM_READ, sizeof(weight_val), sizeof(weight_val), (uint8_t *)weight_val}},

        // data notify characteristic Declaration
        [WEIGHT_NTF_CFG] =
            {{ESP_GATT_AUTO_RSP}, {ESP_UUID_LEN_16, (uint8_t *)&character_client_config_uuid, ESP_GATT_PERM_READ | ESP_GATT_PERM_WRITE, sizeof(weight_val_ccc), sizeof(weight_val_ccc), (uint8_t *)weight_val_ccc}},

        // data receive characteristic Declaration
        [TEMPERATURE_CHAR_IDX] =
            {{ESP_GATT_AUTO_RSP}, {ESP_UUID_LEN_16, (uint8_t *)&character_declaration_uuid, ESP_GATT_PERM_READ, CHAR_DECLARATION_SIZE, CHAR_DECLARATION_SIZE, (uint8_t *)&char_prop_notify}},

        // data receive characteristic Value
        [TEMPERATURE_CHAR_VAL_IDX] =
            {{ESP_GATT_AUTO_RSP}, {ESP_UUID_LEN_128, (uint8_t *)&TEMPERATURE_UUID, ESP_GATT_PERM_READ, sizeof(temperature_val), sizeof(temperature_val), (uint8_t *)temperature_val}},

        // data notify characteristic Declaration
        [TEMPERATURE_NTF_CFG] =
            {{ESP_GATT_AUTO_RSP}, {ESP_UUID_LEN_16, (uint8_t *)&character_client_config_uuid, ESP_GATT_PERM_READ | ESP_GATT_PERM_WRITE, sizeof(temperature_val_ccc), sizeof(temperature_val_ccc), (uint8_t *)temperature_val_ccc}},

        // data receive characteristic Declaration
        [JAR_LID_CHAR_IDX] =
            {{ESP_GATT_AUTO_RSP}, {ESP_UUID_LEN_16, (uint8_t *)&character_declaration_uuid, ESP_GATT_PERM_READ, CHAR_DECLARATION_SIZE, CHAR_DECLARATION_SIZE, (uint8_t *)&char_prop_notify}},

        // data receive characteristic Value
        [JAR_LID_CHAR_VAL_IDX] =
            {{ESP_GATT_AUTO_RSP}, {ESP_UUID_LEN_128, (uint8_t *)&JAR_LID_UUID, ESP_GATT_PERM_READ, sizeof(jar_lid_val), sizeof(jar_lid_val), (uint8_t *)jar_lid_val}},

        // data notify characteristic Declaration
        [JAR_LID_NTF_CFG] =
            {{ESP_GATT_AUTO_RSP}, {ESP_UUID_LEN_16, (uint8_t *)&character_client_config_uuid, ESP_GATT_PERM_READ | ESP_GATT_PERM_WRITE, sizeof(jar_lid_val_ccc), sizeof(jar_lid_val_ccc), (uint8_t *)jar_lid_val_ccc}},

        // data receive characteristic Declaration
        [MOTOR_CHAR_IDX] =
            {{ESP_GATT_AUTO_RSP}, {ESP_UUID_LEN_16, (uint8_t *)&character_declaration_uuid, ESP_GATT_PERM_READ, CHAR_DECLARATION_SIZE, CHAR_DECLARATION_SIZE, (uint8_t *)&char_prop_notify}},

        // data receive characteristic Value
        [MOTOR_CHAR_VAL_IDX] =
            {{ESP_GATT_AUTO_RSP}, {ESP_UUID_LEN_128, (uint8_t *)&MOTOR_UUID, ESP_GATT_PERM_READ, sizeof(motor_val), sizeof(motor_val), (uint8_t *)motor_val}},

        // data notify characteristic Declaration
        [MOTOR_NTF_CFG] =
            {{ESP_GATT_AUTO_RSP}, {ESP_UUID_LEN_16, (uint8_t *)&character_client_config_uuid, ESP_GATT_PERM_READ | ESP_GATT_PERM_WRITE, sizeof(motor_val_ccc), sizeof(motor_val_ccc), (uint8_t *)motor_val_ccc}},

        // data receive characteristic Declaration
[... 18 lines cut ...]

        // data notify characteristic Declaration
        [DIAL_NTF_CFG] =
            {{ESP_GATT_AUTO_RSP}, {ESP_UUID_LEN_16, (uint8_t *)&character_client_config_uuid, ESP_GATT_PERM_READ | ESP_GATT_PERM_WRITE, sizeof(dial_val_ccc), sizeof(dial_val_ccc), (uint8_t *)dial_val_ccc}},

        // data receive characteristic Declaration
        [TEMPERATURE_PID_CHAR_IDX] =
            {{ESP_GATT_AUTO_RSP}, {ESP_UUID_LEN_16, (uint8_t *)&character_declaration_uuid, ESP_GATT_PERM_READ, CHAR_DECLARATION_SIZE, CHAR_DECLARATION_SIZE, (uint8_t *)&char_prop_notify}},

        // data receive characteristic Value
        [TEMPERATURE_PID_CHAR_VAL_IDX] =
            {{ESP_GATT_AUTO_RSP}, {ESP_UUID_LEN_128, (uint8_t *)&TEMPERATUREPID_UUID, ESP_GATT_PERM_READ, sizeof(temperature_pid_val), sizeof(temperature_pid_val), (uint8_t *)temperature_pid_val}},

[... cut ...]
        /* Characteristic Declaration */
        [WRITE_CHAR_IDX] = {
            {ESP_GATT_AUTO_RSP}, {ESP_UUID_LEN_16, (uint8_t *)&character_declaration_uuid, ESP_GATT_PERM_READ, CHAR_DECLARATION_SIZE, CHAR_DECLARATION_SIZE, (uint8_t *)&char_prop_write}},

        /* Characteristic Value */
        [WRITE_CHAR_VAL_IDX] = {
            {ESP_GATT_AUTO_RSP}, {ESP_UUID_LEN_128, (uint8_t *)&WRITE_UUID, ESP_GATT_PERM_READ | ESP_GATT_PERM_WRITE, sizeof(tablet_write_val), sizeof(tablet_write_val), (uint8_t *)tablet_write_val}},
};

zhp0406, 2024-06-27:
My patch only demonstrates how to use a 128-bit UUID and does not provide any fixes. Therefore, if you are already able to use a 128-bit UUID, you do not need to apply our patch.

It is important to note that 128-bit UUIDs can be classified into base 128-bit UUIDs and non-base 128-bit UUIDs. Base 128-bit UUIDs can be converted to 16-bit UUIDs, and this conversion will be done automatically.

For example, if you define a base 128-bit UUID on a GATT server, the GATT client will automatically convert the discovered 128-bit UUID to the corresponding 16-bit UUID.

Alvin1Zhang, 2024-12-18:
Thanks for reporting, will close due to inactivity, feel free to reopen if the issue still happens. Thanks.

closed 2024-12-18 by Alvin1Zhang