espressif/esp-idf #13986: control transfer failed

ulao, 2024-06-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4-dev-826-g8760e6d2a7

### Espressif SoC revision.

esp32-s3

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

wroom32 devkitc

### Power Supply used.

USB

### What is the expected behavior?

No control transfer errors. 

### What is the actual behavior?

depends;
sometimes it works
other times:
E (21420) USBH: Dev 1 EP 0 STALL
E (21420) ENUM: [0:0] Control transfer failed, status=1

some device I always get
E (21420) USBH: Dev 1 EP 0 STALL
E (21420) ENUM: [0:0] Control transfer failed, status=4
 

### Steps to reproduce.

1. I ran the usb host lib exmaple
2. ran monitor
3. plugged in a device
...

### Debug Logs.

```plain
PS C:\Users\Administrator\Desktop\usb2llapi\usb_host_lib> set IDF_PATH=c:\Users\Administrator\Desktop\esp\esp-idf
PS C:\Users\Administrator\Desktop\usb2llapi\usb_host_lib> C:\Users\Administrator\.espressif\python_env\idf5.4_py3.11_env\Scripts\python.exe c:\Users\Administrator\Desktop\esp\esp-idf\tools\idf_monitor.py -p COM9 -b 115200 --toolchain-prefix xtensa-esp32s3-elf- --target esp32s3 c:\Users\Administrator\Desktop\usb2llapi\usb_host_lib\build\usb_host_lib_example.elf
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM9 instead...
--- esp-idf-monitor 1.4.0 on \\.\COM9 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x1790
load:0x403c8700,len:0x4
load:0x403c8704,len:0xcf0
load:0x403cb700,len:0x2d88
entry 0x403c8928
I (27) boot: ESP-IDF v5.4-dev-826-g8760e6d2a7 2nd stage bootloader
I (27) boot: compile time Jun 14 2024 17:48:55
I (27) boot: Multicore bootloader
I (31) boot: chip revision: v0.2
I (35) boot.esp32s3: Boot SPI Speed : 80MHz
I (40) boot.esp32s3: SPI Mode       : DIO
I (45) boot.esp32s3: SPI Flash Size : 2MB
I (49) boot: Enabling RNG early entropy source...
I (55) boot: Partition Table:
I (58) boot: ## Label            Usage          Type ST Offset   Length
I (66) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (73) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (81) boot:  2 factory          factory app      00 00 00010000 00100000
I (88) boot: End of partition table
I (92) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=0bbd4h ( 48084) map
I (109) esp_image: segment 1: paddr=0001bbfc vaddr=3fc92800 size=028f4h ( 10484) load
I (112) esp_image: segment 2: paddr=0001e4f8 vaddr=40374000 size=01b20h (  6944) load
I (119) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=1e160h (123232) map
I (148) esp_image: segment 4: paddr=0003e188 vaddr=40375b20 size=0cc1ch ( 52252) load
I (166) boot: Loaded app from partition at offset 0x10000
I (166) boot: Disabling RNG early entropy source...
I (178) cpu_start: Multicore app
I (187) cpu_start: Pro cpu start user code
I (187) cpu_start: cpu freq: 160000000 Hz
I (187) app_init: Application information:
I (190) app_init: Project name:     usb_host_lib_example
I (196) app_init: App version:      1
I (200) app_init: Compile time:     Jun 14 2024 17:48:23
I (206) app_init: ELF file SHA256:  febbb507c...
I (212) app_init: ESP-IDF:          v5.4-dev-826-g8760e6d2a7
I (218) efuse_init: Min chip rev:     v0.0
I (223) efuse_init: Max chip rev:     v0.99
I (228) efuse_init: Chip rev:         v0.2
I (233) heap_init: Initializing. RAM available for dynamic allocation:
I (240) heap_init: At 3FC95958 len 00053DB8 (335 KiB): RAM
I (246) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (252) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (258) heap_init: At 600FE100 len 00001EE8 (7 KiB): RTCRAM
I (265) spi_flash: detected chip: generic
I (269) spi_flash: flash io: dio
W (273) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (286) sleep: Configure to isolate all GPIO pins in sleep state
I (293) sleep: Enable automatic switching of GPIO sleep configuration
I (300) main_task: Started on CPU0
I (310) main_task: Calling app_main()
I (310) gpio: GPIO[48]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (320) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2
I (330) USB host lib: Installing USB Host Library
I (360) CLASS: Registering Client
(((plugged in device here )))
E (21420) USBH: Dev 1 EP 0 STALL
E (21420) ENUM: [0:0] Control transfer failed, status=4
I (21800) USB host lib: No more devices
```

### More Information.

I have many USB devices so it maybe common for this issue to occur. but I do not know the code well enough to know where to look. If I knew where to add some debug I may be able to see what is causing this issue.  With one device I always get the status 4, so that maybe a device problem but I will need to know what initialed this request. Where did it fail, on what control transfer?

My device board had a otg jumper I have to use to get power to the OTG port. I am running  the jtag via the other USB com port. Since I see debug messages I assume that is all done correctly. I'm new to platform.io so I will not know much about how its set up. Let em know what else I need here.

roma-jam, 2024-06-15:
Hi @ulao,

Thanks for reporting the problem. 
You are right, this is a regression which has been added recently. Sorry for any inconvenience. 

We have already fix it internally, so the next iteration of updating the esp-idf here will solve the problem. 

When device doesn‘t have a descriptor (for example String Descriptor) it could STALL the EP during control transfer. 
This is not the issue for string descriptors (probably one of your devices or bootloaders don‘t have some string descriptors, that is why the control request is STALLed) . 

Please, stay updated and try again when the fix will be available. 

Regarding the extra symbol in the end of the string descriptor - I will take a look ASAP.

ulao, 2024-06-15:
wow thx for the update! I was debug the heck out of this but glad to know I can stop. yes I do not have a serial or version
	//case 0x03: no serial
	//case 0x04: no version
in my devices.

So I'm new to this repo, when should I expect updates? Are they on a weekly/nightly or other rotation?

tore-espressif, 2024-06-17:
Hello @ulao , the `master` branch is always considered to be work in progress, we advise users to start working with latest stable version, which is [v5.2.2](https://github.com/espressif/esp-idf/tree/v5.2.2) at the time of writing.

The fix is already in master branch, so just update your local repo with `git pull`. See fix commit https://github.com/espressif/esp-idf/commit/d214c482f64c90deb7e55576adcf1a4a4cfe4bb3

More details about esp-idf release plan are in [official docs](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/versions.html#which-version-should-i-start-with)

ulao, 2024-06-18:
yes the latest code does work nice work, also the last charter changed now, so something altered that bug, but  I know you have not looked in to to yet. 

One other slightly off topic question. For my genera ESP question I find esp32.com  useless, for platform.io I find community.platformio.org exceptional but for this repo I'm lost when I need help I do not want to open bugs. Is there no way to chat about how the overall culture works? This repo changes the way things work and confuses anyone trying to help me.  I have so many small but much needed question I find that are specific to this repo. Is there any means of help out there?

ulao, 2024-06-19:
well its adding up too much for me, so I guess I will just make more bug reports but I personalty feel that incredible inefficient. Please consider a chat group maybe a discord?

tore-espressif, 2024-06-20:
> For my general ESP questions I find esp32.com useless, for platform.io I find community.platformio.org exceptional but for this repo I'm lost when I need help. I do not want to open bugs. Is there no way to chat about how the overall culture works?

The way we want it to be is:
1. Use esp32.com forum for general questions, ideas and so on
2. Use GitHub issues for bug reports

But I see your confusion. The developers are overloaded with similar requests and we often prioritize GH issues...

closed 2024-06-20 by tore-espressif