espressif/esp-idf #13989: Using RS485 from M5 Stack

diplfranzhoepfinger, 2024-06-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi. 

M5 Stack used quite different pinout:

nopnop2002, 2024-06-15:
There are no restrictions on GPIO used with ESP-IDF's UART.
You can use any pin you like on M5STACK.

diplfranzhoepfinger, 2024-06-16:
> There are no restrictions on GPIO used with ESP-IDF's UART. You can use any pin you like on M5STACK.

set RTS to -1 then ?

diplfranzhoepfinger, 2024-06-16:
i made a Test Setup: 

use the 2 PCBs from https://static-cdn.m5stack.com/resource/docs/products/base/lan_poe_v12/lan_poe_v12_sch_03.webp

with this code: 
https://github.com/DiplomInformatikerFranzHoepfinger/m5stack-modbus/tree/rs-232-ok

and it works. 
Log: https://gist.github.com/diplfranzhoepfinger/6a896cf1cc1dba1e8d0bf36eb6e328e4

diplfranzhoepfinger, 2024-06-16:
same Software, using RS-485 PCB: 

https://gist.github.com/diplfranzhoepfinger/ff6ec398b999edde3b6ee178f1f6be2c

does not work.

diplfranzhoepfinger, 2024-06-18:
https://community.m5stack.com/topic/306/plc-module/12?_=1718738548748

diplfranzhoepfinger, 2024-06-18:
I now made a own PCB for M5 Stack: 

with a correct wired RS-485, 
also Chip which has Dual Volt: 

[image]

alisitsyn, 2024-07-25:
Hi @diplfranzhoepfinger,

Could you update with the results of investigation? The schematic above should work just fine with esp32 driver in half duplex mode. Please let me know if you still have any issues. Thanks.

diplfranzhoepfinger, 2024-07-25:
I am just ordering PCBs this day.

alisitsyn, 2024-07-25:
Ok, thank you for update. Please let me know once you have results. Thanks.

diplfranzhoepfinger, 2024-08-13:
Hi, 

i made EAGLE Files public here: 
https://github.com/Fliegl-Agrartechnik-GmbH/M5_COMMS

the PCBs should arrive tomorrow, i will post you the Results then. 

Thanks,
Franz

franzhoepfinger, 2024-08-14:
got the Boards today: 

1. MAX3485, is a Fully 3.3V RS-485 Transceiver
[image]

2. LTC2877IMSE, is a 5V RS-485 Transceiver with a VIO Pin, so can interface to a 3.3V and even 1.8 V MCU
[image]

3. MAX485 is a 5V only Transceiver, using the Schottky Diode to not damage the ESP32 at RX.
[image]

4. CAN-Bus with a Infineon Transceiver
[image]

alisitsyn, 2024-08-14:
@franzhoepfinger,

The boards look nice. However, I would recommend galvanic isolation on the RS485 interface. This would allow to solve many of possible issues in future. The chips like TI ISO1500, Analog Devices ADM2485, work good. Probably this suggestion is not for your project.
Looking forward to get some test results with ESP32 for these chips. Just because some users reported issues with the SP3485 chip for example connected to ESP32. I personally could not check it on my side and do not expect any issues with the selected chips.

Thanks.

franzhoepfinger, 2024-08-16:
> Hi @franzhoepfinger,
> 
> The boards look nice. However, I would recommend galvanic isolation on the RS485 interface.

i did share this meaning a long time, say years, unless i realized that most people do not understand it. 
if the Device provide galvanic isolation and the Electrican connects the GND to the common GND, all isolation is fully gone. 

so, yes, isolation is cool, but all circumstances must support it wisely. 

> This would allow to solve many of possible issues in future. The chips like TI ISO1500, Analog Devices ADM2485, work good. Probably this suggestion is not for your project. Looking forward to get some test results with ESP32 for these chips. 

> Just because some users reported issues with the SP3485 chip for example connected to ESP32. 

> I personally could not check it on my side and do not expect any issues with the selected chips.

this here is manly about the Connection 3 Wire vs. 2wire from Chip to ESP32, 
because the Boards from M5Stack @m5stack seem really not working. 

> 
> Thanks.

alisitsyn, 2024-08-16:
@franzhoepfinger,

> so, yes, isolation is cool, but all circumstances must support it wisely.

Agree.

> this here is manly about the Connection 3 Wire vs. 2wire from Chip to ESP32,
because the Boards from M5Stack https://github.com/m5stack seem really not working.

Sorry, I did not understand why the RS485 board from M5 Stack with 2 wire connection (auto switching of direction) does not work with ESP32 on your side. If it is, the reason for this should be found.

[image]

It should work just fine and it is [recommended](https://docs.espressif.com/projects/esp-idf/en/v4.3.2/esp32/api-reference/peripherals/uart.html#circuit-c-auto-switching-transmitter-receiver) and tested variant except the chip is different.
It does not matter which connection type is used for ESP32 3 or 2 wire, both are supported and tested. With the 3-wire variant the difference is that driver is controlled from the TX line and differential high level on output is set by the pull-up(pull_down) resisters (RTS is not used for driver control). I remember that long time ago I was able to verify your M5 adapter board with the schematic as above and it was working just fine with ESP32.

Please let me know if I miss something.

diplfranzhoepfinger, 2024-08-17:
also visible in IDF 5.3
https://docs.espressif.com/projects/esp-idf/en/release-v5.3/esp32s3/api-reference/peripherals/uart.html

must make more Tests. 
for sure i did not set this Bits.

franzhoepfinger, 2024-08-17:
next bad thing: 
how to set these Bits.

franzhoepfinger, 2024-08-17:
it is working as well, Circuit C: 
https://github.com/Fliegl-Agrartechnik-GmbH/uart_echo_rs485/tree/feature/Circuit-C

the Signals using the M5 Board are not as clean as using the other 3 Boards, especcially the Mid-Voltage is offset a lot. (2V Mid, while Max Voltage of B Channel is about 4.6V, and Min of A Channel 0.35V

but other than this the Echo App is working fine so far ...

franzhoepfinger, 2024-08-17:
Next Problem: 
https://github.com/espressif/esp-idf/tree/master/examples/protocols/modbus/serial/mb_master

cannot set RTS to -1 by SDKCONFIG

diplfranzhoepfinger, 2024-08-17:
> Hi @franzhoepfinger,
> 
> The boards look nice. However, I would recommend galvanic isolation on the RS485 interface. 

https://docs.m5stack.com/en/module/Module13.2-PwrCAN

M5 seems to have a isolated one now.

alisitsyn, 2024-08-19:
@franzhoepfinger,

> should have hints about this. they only show Circuit B.

The example shows just the recommended type of connection, which is `Circuit B`. All possible variants are supported by UART driver and explained on its page and responsibility of user. However, I agree it worth to add at least the link to the documentation related to UART driver options where all these variants and options are explained. 

> cannot set RTS to -1 by SDKCONFIG

Yes, this is because the original example designed to use the recommended type of connection which is `Circuit A` or `Circuit B` as showed on the connection diagram. The "-1" can be changed in the source code in this case or just keep it as is and do not connect the RTS pin at all. I will try to address this later.

> (2V Mid, while Max Voltage of B Channel is about 4.6V, and Min of A Channel 0.35V 
but other than this the Echo App is working fine so far ...

Is not excellent but can work. Could you send the picture from oscilloscope for more details? If it is possible. 

> next bad thing: how to set these Bits.

If you are talking about the UART register bits, you do not need to set it manually (the information there is just for reference). The UART driver will take care of it. What you need is to install driver and set communication options for RS485 HALF DUPLEX. The driver will do the rest. For variant C of connection it does not matter which mode is used (PURE UART communication or HALF DUPLEX). However, it is better to set the RTS to "-1" to free the GPIO for other purposes.

```c
ESP_ERROR_CHECK(uart_driver_install(uart_num, BUF_SIZE * 2, 0, 0, NULL, 0));

    // Configure UART parameters
    ESP_ERROR_CHECK(uart_param_config(uart_num, &uart_config));

    ESP_LOGI(TAG, "UART set pins, mode and install driver.");

    // Set UART pins as per KConfig settings (The ECHO_TEST_RTS needs to be set manually to -1, to be fixed.)
    ESP_ERROR_CHECK(uart_set_pin(uart_num, ECHO_TEST_TXD, ECHO_TEST_RXD,  -1, ECHO_TEST_CTS));

    // Set RS485 half duplex mode (Note: is not required for variant C of RS485 connection. This line can be kept or removed.)
    // ESP_ERROR_CHECK(uart_set_mode(uart_num, UART_MODE_RS485_HALF_DUPLEX));

    // Set read timeout of UART TOUT feature
    ESP_ERROR_CHECK(uart_set_rx_timeout(uart_num, ECHO_READ_TOUT));
```

> M5 seems to have a isolated one now.

Thanks for the link. I see it now. So, the ISO_5V should come from isolated power supply `DC 050 connector` on user side, otherwise this is useless.

Thank you for your comments and will try to address them a bit later.

diplfranzhoepfinger, 2024-08-19:
> @franzhoepfinger,
> 
> > should have hints about this. they only show Circuit B.
> 

true. for Circuit C i used the M5 Stack PCBs. 

> The example shows just the recommended type of connection, which is `Circuit B`. All possible variants are supported by UART driver and explained on its page and responsibility of user. However, I [...]

i did edit it, and it is working. 

> 
> > (2V Mid, while Max Voltage of B Channel is about 4.6V, and Min of A Channel 0.35V
> > but other than this the Echo App is working fine so far ...
> 
> Is not excellent but can work. Could you send the picture from oscilloscope for more details? If it is possible.

YES, i will share it, already captured all these pics.

diplfranzhoepfinger, 2024-08-19:
true. 

so you can set Circuit B this way: 

```c
    // Set RS485 half duplex mode
    ESP_ERROR_CHECK(uart_set_mode(UART_NUM_1, UART_MODE_RS485_HALF_DUPLEX));
```

and Circuit A this way: 

```c
    // Set RS485 collision detection UART mode
    ESP_ERROR_CHECK(uart_set_mode(UART_NUM_1, UART_MODE_RS485_COLLISION_DETECT));
```

diplfranzhoepfinger, 2024-08-19:
> > M5 seems to have a isolated one now.
> 
> Thanks for the link. I see it now. So, the ISO\_5V should come from isolated power supply `DC 050 connector` on user side, otherwise this is useless. Ah, I see the isolated power over U4 F0505S now.

true, but again: if you sell the Devices, to my Experience many Customers just Bridge the 2 GNDs, and all isolation is useless. 

so, need to sell traingins as well, otherwise no benefit.

alisitsyn, 2024-08-30:
@diplfranzhoepfinger,

Is your issue addressed? Can this ticket be closed? 
Your other comments about stack are addressed separately and merged.

alisitsyn, 2024-09-04:
The issue is closed. Feel free to reopen.

closed 2024-09-04 by alisitsyn

diplfranzhoepfinger, 2024-09-04:
> @diplfranzhoepfinger,
> 
> Is your issue addressed? Can this ticket be closed? Your other comments about stack are addressed separately and merged.

YES, can be closed for now. 
i have all RS-485 Stuff running. 

Problems which are left are in the BSP Repo, mainly Display and Ethernet.

alisitsyn, 2024-09-04:
@diplfranzhoepfinger,

Thank you for update. 

> YES, can be closed for now.
i have all RS-485 Stuff running.

If it is possible please share the communication diagrams for your boards. They might be useful to determine some aspects related to different chips.

>  Problems which are left are in the BSP Repo, mainly Display and Ethernet.

Please submit the separate tickets for these components.

diplfranzhoepfinger, 2024-09-04:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

alisitsyn, 2024-09-04:
> Descriptions missing, also the Version with the M5 Original PCB missing yet.

This is ok, I can guess  the missing aspects)
What I was interested in is to see the diagram for that original board with TX - RX connection (5V) and the  LTC2877 one with RTS (3.3V + 5V). No problem, the diagrams are still can be useful. 

Thanks again for sharing.